openwrt/openwrt #11104: NanoPi R1: Unstable booting

AiyionPrime, 2022-10-29:
As @goligo and me were looking at the NanoPi R1 a few days ago, we found it does not boot with recent snapshots, but does with older releases, such the latest 21.x ones.

I found the commit to introduce the problem was abd67de9f0d6470ba48f6df635db29ce91855397 after bisecting it.
The boot fails in about five out of six times for every build afterwords.
This is a screenlog of a failing boot: http://aiyionpri.me/shared/sunxi_boot/screenlog.0

After mentioning it oftc/#openwrt-devel @hauke suggested to remove the clockfrequency from the uboot-config:

> 00:16 <hauke> aiyion: this looks strange: cpufreq: cpufreq_online: CPU0: Running at unlisted initial frequency: 480000 KHz, changing to: 648000 KHz
00:17 <hauke> I think the U-Boot configuration is broken, see package/boot/uboot-sunxi/patches/252-sunxi-h3-add-support-for-nanopi-r1.patch
00:17 <hauke> Could you remove this line please: CONFIG_SYS_CLK_FREQ=480000000
00:19 <hauke> aiyion: please create a ticket for this problem with your logs attached

Which appears to solve the problem or get rid of the symptom.

The device does still not boot reliably, but that's due to it waiting for a blockdevice it might mix up, an issue I'll open in a moment.

For completeness these are the two possible boot-outcomes after @hauke's suggestion.

http://aiyionpri.me/shared/sunxi_boot/screenlog_0.0 (sd-card mixed up)
http://aiyionpri.me/shared/sunxi_boot/screenlog_1.0 (working)

They appear about half of the time each, though the sample quantity is fairly limited.

linxcow, 2022-12-11:
I think i have the same problem , new nanopi R1 => older friendlywrt works ( giving out ip on wired device )
openwrt 22.03.2 images dont give out ip's tested with multiple cards.
Is there a fix ?

AiyionPrime, 2022-12-11:
Maybe I'll just start with a patch for the first issue so the problems get less...

linxcow, 2022-12-11:
I can confirm that 21.03.5 hands out ip over lan wired . 
So probably only 22.x.x affected

AiyionPrime, 2022-12-11:
Yep, that's true. I think I found the kernel upgrade to 5.10 to be the introducing commit.

linxcow, 2022-12-11:
will your patch be included in the builds or do I have to do this myself ( no experience )

AiyionPrime, 2022-12-11:
@linxcow I submitted the patch, you could help get it merged by verifying it resolves the first issue [1].
On the right, there is a is a button called "mbox" which will download the patch. You can then apply it in your local OpenWrt environment and build an image with it.

[1] https://patchwork.ozlabs.org/project/openwrt/patch/20221211130446.23492-1-git@aiyionpri.me/

AiyionPrime, 2022-12-11:
Indeed.
There are several pages around this in the wiki, but this should give you an idea, what tools you need to build OpenWrt on eg. Ubuntu or Debian:
https://openwrt.org/docs/guide-developer/toolchain/install-buildsystem#debianubuntu

Afterwords you'd clone the OpenWrt Repo, download our patch, apply it using git and build it, which might take a while depending on your system. (Takes about an hour and a half on my laptop...)

I hope this list of steps is complete, but I have not tested it:
If you encounter **any** errors, stop immediately, do **not** continue with the next step.
Maybe that's obvious to you but I've dealt with far to many people that ignored problems during builds an wondered why it failed afterwords :)

```bash
# install ubuntu/debian dependencies
sudo apt update
sudo apt install build-essential clang flex g++ gawk gcc-multilib gettext \
git libncurses5-dev libssl-dev python3-distutils rsync unzip zlib1g-dev \
file wget

# clone the repo
git clone https://git.openwrt.org/openwrt/openwrt.git

# change into that new directory
cd openwrt

# download the patch
wget https://patchwork.ozlabs.org/project/openwrt/patch/20221211130446.23492-1-git@aiyionpri.me/mbox/ -o nanopi_frequency_fix.patch

# apply the patch
git am nanopi_frequency_fix.patch

# start build preparations
./scripts/feeds update -a
./scripts/feeds install -a

# choose the correct device, this looks harder than it is. It will open an interactive interface in which you select what device you want to build an image for.

# simply choose the following from the dropdown menus:
* Target System: `Allwinner A1x/A20/A3x/H3/H5/R4`
* Subtarget: `Allwinner A20/A3x/H3/R40`
* Target Profile: `FriendlyARM NanoPi R1`
* choose `save` afterwords and after saving `exit`

make menuconfig

# you just prepared the buildconfig, good job.
# find out how many cores you have to efficiently speed up the following build:

cat /proc/cpuinfo | grep processor | wc -l 

# this shout have returned a number that tells how many cpu cores are there in your system.
# for me its four. Add one to the number (that gives 5 for my laptop.)

# call `make -j` with your calculated number. This will take a while.
make -j5

# done? your images are in the directory `bin/targets/sunxi/cortexa7`.
```

AiyionPrime, 2022-12-15:
@linxcow The second part of the issue is this patch, I think.

https://patchwork.ozlabs.org/project/openwrt/patch/20221215141833.100172-1-git@aiyionpri.me/

Testing it would be highly appreciated, at least for me this resolves all above issues.

linxcow, 2022-12-15:
Thx , Ill' try to find some time to spin up a virtual machine and prepare a build environment this weekend.

stintel, 2022-12-22:
Should be solved by bf06a7c865531cf20735cbdfac46d4947fcc18a6.

closed 2022-12-22 by stintel

n-pat, 2023-01-02:
I can confirm that it boots without issues regarding the "cpu frequency". Also "boot drive alias" seems to work (if not related to following problems).

But the uhttpd seems to be missing, HTTP UI is not available. I can login using SSH, but cannot issue `/etc/init.d/uhttpd start` because the file does not exist (/etc/init.d/uhttpd).

I tried using the snapshot from today (2023-01-02) and also a self build image as described in the postings above (patches were not neccessary anymore as it seems everything is available in master in the meantime). I will try to inspect this in the next days, but I think it is not related.