openwrt/openwrt #11109: Meraki MR16 won't boot from openwrt-22.03.2-ath79-generic-meraki_mr16-initramfs-kernel.bin

ajuniper, 2022-10-29:
Trying to fresh install Meraki MR16, following instructions on MR16 hardware page.
When I try to boot the device using `openwrt-22.03.2-ath79-generic-meraki_mr16-initramfs-kernel.bin`, I see this:
```
ar7100> setenv bootcmd bootm 0xbf0a0000; saveenv;
Saving Environment to Flash...
Protect off BF040000 ... BF04FFFF
Un-Protecting sectors 4..4 in bank 1
Un-Protected 1 sectors
Erasing Flash...Erase Flash from 0xbf040000 to 0xbf04ffff in Bank # 1 
First 0x4 last 0x4 sector size 0x10000
   4
Erased 1 sectors
Writing to Flash... write addr: bf040000
done
Protecting sectors 4..4 in bank 1
Protected 1 sectors
ar7100> tftpboot 0c00000 openwrt-22.03.2-ath79-generic-meraki_mr16-initramfs-kernel.bin
Trying eth0
pll reg 0x18050010: 0x110000  Using eth0 device
TFTP from server 192.168.1.101; our IP address is 192.168.1.2
Filename 'openwrt-22.03.2-ath79-generic-meraki_mr16-initramfs-kernel.bin'.
Load address: 0xc00000
Loading: #################################################################
<snip>
done
Bytes transferred = 5610218 (559aea hex)
ar7100> bootm 0c00000;
## Booting image at 00c00000 ...
   Image Name:   MIPS OpenWrt Linux-5.10.146
   Created:      2022-10-14  22:44:41 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    5610154 Bytes =  5.4 MB
   Load Address: 80060000
   Entry Point:  80060000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... ERROR: LzmaDecode.c, 543

Decoding error = 1
LZMA ERROR 1 - must RESET board to recover

Resetting...
```
And then it re-enters the bootloader with an unbootable system (uboot operates ok).  The problem is repeatable, the sha256sum verifies correctly, I tried downloading the image again and no change.
I overcame the problem by booting with `openwrt-21.02.5-ath79-generic-meraki_mr16-initramfs-kernel.bin` at which point I flashed `openwrt-22.03.2-ath79-generic-meraki_mr16-squashfs-sysupgrade.bin` and the system installed and booted fine.

The uboot identifies itself as:
```
U-Boot 1.1.4-g5416eb09-dirty (Mar  3 2011 - 16:28:15)

AP96 (ar7100) U-boot 0.0.1 MERAKI 
```

Hurricos, 2022-11-01:
I ran into this issue elsewhere too.

The problem is that 0xc00000 is 12MB; the decompressed kernel (plus rootfs?) is now slightly larger than that, so in the process of decompressing, you end up overwriting your LZMA dictionary at 0xc00000(ish)!

Would you please try `tftpboot 2000000 openwrt-22.03.2-ath79-generic-meraki_mr16-initramfs-kernel.bin` for me (or `tftpboot 1000000`)? Basically, look for the highest memory value (note the board has 64MB of physical memory) that still works to copy to.

Also, for a quick hex calculator, `bash` and `ash` let you just `echo $((0xc00000))` for the decimal value.

VikingGer, 2022-11-06:
There's likely the same issue for the Netgear DGN 3500(B). I ran into the same problem as these guys: https://forum.openwrt.org/t/dgn3500-brick-after-22-03-update/137332/13

Unfortunately my USB2Serial seems somehow broken (or something I do not see is misconfigured in minicom) cause the serial output is some utter ASCII rubbish so I couldn't look into the boot process. I did some flashings from the console/Luci with always the same bricked result. I'm available for further testings if there are any ideas.

cygeus, 2022-11-26:
> The problem is that 0xc00000 is 12MB; the decompressed kernel (plus rootfs?) is now slightly larger than that, so in the process of decompressing, you end up overwriting your LZMA dictionary at [...]

Thanks. That was indeed the issue. And the solution.
I've updated the [Installation process of the Meraki MR16](https://openwrt.org/toh/meraki/mr16#installation_process) on the Wiki.

Hurricos, 2022-12-11:
Sounds like this can be closed?

ajuniper, 2022-12-11:
Certainly does sound like it can be closed.

All my MR16s are in use so I can't easily confirm what others have reported.

closed 2022-12-11 by ajuniper