openwrt/openwrt #11116: Improve github automated checks

M95D, 2022-10-30:
I have some suggestions to improve github automated checks:

- Start checks for one platform/target only - arm or another platform that is common in routers. Auto-start the other targets only if the first one builds ok. Why: Most errors are the same on all platforms. It's not necessary to run checks for all platforms if they all fail with the same error. It's a waste of resources. Allow the pull request author to start the other checks manually if he/she thinks that the error is platform-dependent and the other checks won't fail in the same way as the first.

- Allow the user who made the pull request to manually stop the running checks. Why: If there is something to be fixed, there is no need to finish the checks. They will run again anyway when the commit is updated with the fix.

Thanks.

DragonBluep, 2022-10-30:
> * Start checks for one platform/target only - arm or another platform that is common in routers. Auto-start the other targets only if the first one builds ok.

good idea.

> * Allow the pull request author to start the other checks manually if he/she thinks that the error is platform-dependent and the other checks won't fail in the same way as the first.

At present, we only check generic kernel patches and tools, and we cannot ignore and skip any errors. Most importantly, I doubt whether github really supports such operations?

M95D, 2022-11-20:
@hauke
I see that you are proposing some changes here: https://github.com/openwrt/openwrt/pull/11171
Please consider these improvements. The first one would reduce a lot of build bots load.

closed 2024-12-04 by M95D