openwrt/openwrt #11117: realtek: RTL839x lock contention in phy_read()

plappermaul, 2022-10-30:
@musashino205 @janh @svanheule this might be no easy one.

In the last weeks 2 commits found there way into the realtek tree.

https://github.com/openwrt/openwrt/commit/4657a5301eb5512b335de696d6a719c280826d46
https://github.com/openwrt/openwrt/pull/10631/commits/07063d6fca9746552b3af58a9a9ed2639717f675

Afterwards we got reports about slow startups on RTL839x targets 

https://forum.openwrt.org/t/support-for-rtl838x-based-managed-switches/57875/2150

Taking each commit for itself they are both valid and make sense. In combination they show a design flaw of the current PHY implementation. Without long explanation the facts are:

- phy_state_machine() polls the 50+ RTL839x PHYs every second
- the state is read with multiple phy_read() calls under a bus lock
- This finally translates to sequential rtl839x_read_phy() calls
- Avoiding busy waiting the polling inside the function will at least sleep 1 timer interval
- MIPS timer interval was ~1ms while otto timer interval is 10ms (timer default)
- So as of today we can only call a maximum of 100 phy reads per second 

To mitigate that behaviour I can offer to increase the otto timer frequency. But this will only fight the symptoms.

The discussion is open.

svanheule, 2022-10-30:
Thanks for digging into this regression.

Switching to `CONFIG_HZ_1000=y` does seem to improve performance, but that still has a limit of one poll per ms.

plappermaul, 2022-10-31:
> Edit: We were using the R4K timer with `CONFIG_HZ=100` too, so I suspect something else is at play here, rather than just the periodic timer interval.

Nope. cevt-r4k.c does not care about CONFIG_HZ at all.

svanheule, 2022-10-31:
Right, missed that part. But even disabling the .set_periodic and .set_oneshot members, which reduces the timer feature set to that of cevt-r4k, doesn't help.

What does seem to help, is flagging the clock source as `CLOCK_SOURCE_IS_CONTINUOUS`. I'll prepare a patch.

LGA1150, 2022-11-01:
Why not use `cond_resched()` instead of `readx_poll_timeout`?
torvalds/linux@3630d519d7c3eab92567658690e44ffe0517d109

closed 2022-11-02 by svanheule