openwrt/openwrt #11149: Mesh issues with backports-5.15.74-1/net/mac80211/mesh_hwmp.c

Noki, 2022-11-03:
I have a Zyxel NWA55AXE running on OpenWrt 22.03-SNAPSHOT, r19835-0ff3adfa11 which produces the following trace over and over again. Mesh is unstable.

@hauke maybe you can have a look as it seems to be introduced by your commit https://github.com/openwrt/openwrt/commit/00d7702796d922e4258b7acb6e6b88a93071eebe

Please let me know if I should provide any additional information.

```
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.475157] ------------[ cut here ]------------
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.475243] WARNING: CPU: 1 PID: 3911 at backports-5.15.74-1/net/mac80211/mesh_hwmp.c:348 0x82c6c444 [mac80211@3dc80c66+0x81260]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.475254] Modules linked in: nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_compat nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack_bridge nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY nfnetlink nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c iptable_mangle iptable_filter ipt_ECN ip_tables hwmon compat ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 sha256_generic libsha256 seqiv jitterentropy_rng drbg hmac cmac leds_gpio gpio_button_hotplug
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476021]  crc32c_generic
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476068] CPU: 1 PID: 3911 Comm: reader#0 Tainted: G        W         5.10.152 #0
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476081] Stack : 00000000 80726368 809b0000 00000580 00000000 00000000 00000000 00000000
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476160]         00000000 00000000 00000000 00000000 00000000 00000001 82763818 545c9e09
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476236]         827638b0 00000000 00000000 827636c0 00000038 80399d64 ffffffea 00000000
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476315]         827636cc 0000025a 807ca868 20202020 827637f8 806f513c 80000000 806f0000
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476385]         00000009 82b6cda0 816801f0 00000580 00000018 00000020 00000004 80990004
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476460]         ...
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476478] Call Trace:
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476591] [<80399d64>] 0x80399d64
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476643] [<80007b08>] 0x80007b08
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476657] [<80007b10>] 0x80007b10
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476670] [<8037ee68>] 0x8037ee68
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476703] [<82c6c444>] 0x82c6c444 [mac80211@3dc80c66+0x81260]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476714] [<8007fb60>] 0x8007fb60
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476738] [<82c6c444>] 0x82c6c444 [mac80211@3dc80c66+0x81260]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476750] [<8003007c>] 0x8003007c
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476778] [<82c6c444>] 0x82c6c444 [mac80211@3dc80c66+0x81260]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476794] [<80030144>] 0x80030144
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476838] [<82c6c444>] 0x82c6c444 [mac80211@3dc80c66+0x81260]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476863] [<82b05ed0>] 0x82b05ed0 [cfg80211@18be209c+0x45fe0]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476904] [<82c08ff4>] 0x82c08ff4 [mac80211@3dc80c66+0x81260]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.476960] [<80379b68>] 0x80379b68
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477024] [<82c24010>] 0x82c24010 [mac80211@3dc80c66+0x81260]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477066] [<82b2dc20>] 0x82b2dc20 [cfg80211@18be209c+0x45fe0]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477079] [<800c154c>] 0x800c154c
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477092] [<8037896c>] 0x8037896c
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477112] [<803784cc>] 0x803784cc
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477129] [<800101d0>] 0x800101d0
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477152] [<80026237>] 0x80026237
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477196] [<80574ae8>] 0x80574ae8
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477237] [<80575e04>] 0x80575e04
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477252] [<8050e6c4>] 0x8050e6c4
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477284] [<80579790>] 0x80579790
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477307] [<804e3108>] 0x804e3108
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477354] [<82b2dcb0>] 0x82b2dcb0 [cfg80211@18be209c+0x45fe0]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477378] [<82b2db2c>] 0x82b2db2c [cfg80211@18be209c+0x45fe0]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477394] [<805797c0>] 0x805797c0
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477419] [<82b2db2c>] 0x82b2db2c [cfg80211@18be209c+0x45fe0]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477432] [<80578d2c>] 0x80578d2c
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477468] [<8057946c>] 0x8057946c
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477484] [<805783ac>] 0x805783ac
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477519] [<80578d14>] 0x80578d14
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477539] [<80577978>] 0x80577978
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477559] [<801a1410>] 0x801a1410
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477609] [<80577e10>] 0x80577e10
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477629] [<8017f9fc>] 0x8017f9fc
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477664] [<80577aac>] 0x80577aac
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477677] [<804d94c4>] 0x804d94c4
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477695] [<803233c0>] 0x803233c0
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477738] [<804db354>] 0x804db354
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477759] [<800101d0>] 0x800101d0
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477798] [<80174798>] 0x80174798
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477837] [<80174798>] 0x80174798
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477873] [<801cc3b8>] 0x801cc3b8
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477900] [<804db564>] 0x804db564
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477937] [<80014578>] 0x80014578
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.477965]
Wed Nov  2 16:24:53 2022 kern.warn kernel: [   36.478014] ---[ end trace a5af265f4d7ee6f7 ]---
```

dirkvanderwalt, 2022-11-08:
Hi,
I posted the forum entry and are also available to help with feedback and testing.
I have Atheros and IPQ (and also Mediatek) hardware I can test on.

john-tho, 2022-11-08:
The stack dump is only a warn, not an oops, so things do proceed from there:
https://elixir.bootlin.com/linux/v5.15.77/source/net/mac80211/mesh_hwmp.c#L348

If you can build an image, you can see when this breakage was introduced like https://github.com/openwrt/openwrt/issues/11095: try older versions of backports: https://github.com/openwrt/openwrt/commits/master/package/kernel/mac80211/Makefile
Then see what changed in backports: https://git.kernel.org/pub/scm/linux/kernel/git/backports/backports.git/log/, and in the stable kernel backports used to generate a package: 
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/log/net/mac80211?h=linux-5.15.y
Looks like there is a new patch for mesh since that backports package was generated: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/net/mac80211?h=linux-5.15.y&id=6e8eadfa9bb1eb3b6a2eda3a69a3713ab8a55a45, you could try adding that to `package/kernel/mac80211/patches/subsys/`

Noki, 2022-11-11:
Thanks @john-tho for pointing out the fix for the warning. It's fixed now with the following commits:

https://github.com/openwrt/openwrt/commit/161b22d103decd82868e8e3c3fe09b88cf64724c
https://github.com/openwrt/openwrt/commit/615dbec2238c2ea513e72a6928272d7fdf4be3b4

closed 2022-11-11 by Noki