openwrt/openwrt #11162: Zyxel Armor Z2: Adding a new VLAN doesn't create interface

vlolteanu, 2022-11-04:
First things first: I'm using OpenWrt 22.03.2 r19803-9a599fee93 / LuCI openwrt-22.03 branch git-22.288.45147-96ec0cd on a Zyxel Armor Z2.

My VLAN config is as follows:
[image]
The first two are part of the vanilla config. I created VLAN 3 and moved port 4 from 1 to 3.

The immediate effect was that eth1.3 was immediately available in the web UI for bridging etc. However, no network connectivity was available for any device inserted into port 4. A WiFi network was (supposedly) bridged with eth1.3, and it had DHCP, Internet connectivity etc.

Upon closer inspection via ssh, `ip link` reported that eth1.3 does not exist, so there was nobody there to strip the .1q tags, nor was the interface part of any bridge.

Running the following commands in bash fixed the issue (until the next reboot, of course):
```
ip link add link eth1 name eth1.3 type vlan id 3
ip link set dev eth1.3 up
brctl addif br-dmz eth1.3
```

The behaviour seems to be the same when adding a 4th VLAN: eth1.4 isn't actually created and manual commands are needed to get it working.

Here's the output of `uci show`:
```
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].nonegcache='0'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].ednspacket_max='1232'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.dhcpv4='server'
dhcp.lan.dhcpv6='server'
dhcp.lan.ra='server'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.dmz=dhcp
dhcp.dmz.interface='dmz'
dhcp.dmz.start='100'
dhcp.dmz.limit='150'
dhcp.dmz.leasetime='12h'
DROPBEAR STUFF REDACTED
FIREWALL STUFF REDACTED
luci.main=core
luci.main.lang='auto'
luci.main.mediaurlbase='/luci-static/bootstrap'
luci.main.resourcebase='/luci-static/resources'
luci.main.ubuspath='/ubus/'
luci.flash_keep=extern
luci.flash_keep.uci='/etc/config/'
[... 186 lines cut ...]
wireless.radio1.htmode='HT20'
wireless.radio1.channel='auto'
wireless.radio1.cell_density='0'
wireless.wifinet2=wifi-iface
wireless.wifinet2.device='radio0'
wireless.wifinet2.mode='ap'
wireless.wifinet2.ssid='REDACTED'
wireless.wifinet2.encryption='psk2'
wireless.wifinet2.key='REDACTED'
wireless.wifinet2.network='lan'
wireless.wifinet1=wifi-iface
wireless.wifinet1.device='radio1'
wireless.wifinet1.mode='ap'
wireless.wifinet1.ssid='REDACTED'
wireless.wifinet1.encryption='psk2'
wireless.wifinet1.key='REDACTED'
wireless.wifinet1.network='dmz'
wireless.wifinet3=wifi-iface
wireless.wifinet3.device='radio1'
wireless.wifinet3.mode='ap'
wireless.wifinet3.ssid='REDACTED'
wireless.wifinet3.encryption='psk2'
wireless.wifinet3.key='REDACTED'
wireless.wifinet3.network='lan'
```

john-tho, 2022-11-06:
[snip]'ed config down to the relevant parts

> Here's the output of `uci show`:
> ```
> network.dmz=interface
> network.dmz.proto='static'
> network.dmz.ipaddr='10.0.2.1/24'
> network.dmz.type='bridge'
> network.dmz.device='br-dmz'
> network.@device[3]=device
> network.@device[3].type='bridge'
> network.@device[3].name='br-dmz'
> network.@switch_vlan[2]=switch_vlan
> network.@switch_vlan[2].device='switch0'
> network.@switch_vlan[2].vlan='3'
> network.@switch_vlan[2].vid='3'
> network.@switch_vlan[2].ports='6t 4'
> ```

It looks like you need to add your new swconfig vlan to your bridge. Guessing `eth1.3` as a port in `br-dmz`. swconfig only configures the switch hardware. netifd (Openwrt's network managment daemon) does not know about your swconfig vlan until you reference it somewhere.

vlolteanu, 2022-11-06:
I did try adding it to br-dmz but it never worked.

Aster a fresh reboot (to clear the side-effects of the manual interface creation) and adding it to br-dmz, this is what the devices tab looks like:

[image]

eth1.3 shows up and is selectable for bridging, but `ip link` and `brctl show` still report it as missing. The other VLAN interfaces, eth1.1 and eth0.2, show up as "Network devices", rather than "VLAN (802.1q)"; I'm not sure why that is.

These are the relevant bits from uci:
```
root@OpenWrt:~# uci show | grep dmz
dhcp.dmz=dhcp
dhcp.dmz.interface='dmz'
dhcp.dmz.start='100'
dhcp.dmz.limit='150'
dhcp.dmz.leasetime='12h'
firewall.@zone[2].name='dmz'
firewall.@zone[2].network='dmz'
firewall.@forwarding[1].dest='dmz'
firewall.@forwarding[2].src='dmz'
firewall.@rule[9].src='dmz'
firewall.@rule[10].src='dmz'
network.dmz=interface
network.dmz.proto='static'
network.dmz.ipaddr='10.0.2.1/24'
network.dmz.type='bridge'
network.dmz.device='br-dmz'
network.@device[3].name='br-dmz'
wireless.wifinet1.network='dmz'

root@OpenWrt:~# uci show | grep eth1.3
network.@device[3].ports='eth1.3'
```

john-tho, 2022-11-06:
Okay, check the logs with `logread`
You might be confusing netifd by defining both the dmz inferface with bridge, and the device as bridge.
```
network.dmz=interface
network.dmz.proto='static'
network.dmz.ipaddr='10.0.2.1/24'
network.dmz.type='bridge'
network.dmz.device='br-dmz'
network.@device[3]=device
network.@device[3].type='bridge'
network.@device[3].name='br-dmz'
```

Take the `type` line out of the dmz network definition `network.dmz.type='bridge'`, then restart the network `/etc/init.d/network restart`.
If that does not work, check the log via `logread`. If you cannot see anything relevant, you can make `netifd` more verbose by adjusting it's `/etc/init.d/network`. In `start_service`, change the `command` line to
```
procd_set_param command /sbin/netifd -d 15 -l 7
```
and add these two lines
```
    procd_set_param stdout 1
    procd_set_param stderr 1
```
then restart network and check log again. Change it back after you fix things.

vlolteanu, 2022-11-09:
I've dug even further and uncovered this weird behavior:

(For clarity, `eth1.3` always shows up in LuCI. The actual Linux virtual device, that can be interacted with in bash with rudimentary tools, e.g. `ip`, `brctl` etc. is the issue.)

If I enable VLAN filtering `br-dmz` and set 3 as untagged, `eth1.3` shows up (as reported by `ip link`). `tcpdump` on `eth1.3` shows the wired device's DHCP requests. 

However, bridges with VLAN filtering don't deliver anything locally, so accessing DHCP won't work.

Disabling VLAN filtering on `br-dmz` also deletes `eth1.3`, i.e. `ip link` no longer shows it. LuCI still pretends that `eth1.3` is there, and lets me manipulate it etc.

vlolteanu, 2022-11-09:
All in all, it seems like LuCI is working with an abstraction of `eth1.3`; unless certain configuration steps are made, it decides that the abstraction doesn't need to be backed by an actual device.