openwrt/openwrt #11164: Xiaomi Redmi AX6000 - Slow(er) LAN Transfer Speeds

shauno100, 2022-11-04:
Hi, Not sure if this has been reported already or not, i have noticed an issue even under stock MiWiFi firmware where if i copy a large file from my NAS connected to router at 1Gbps via ethernet, the file transfer rate (MB/s) according to Windows maxs out at approx 90MB/s. I have a spare Xiaomi AX3200 router and the transfer rate maxes out at around 114MB/s. Not a huge difference but i am curious why the transfer rate would be 25-30MB/s less then an older less powerful router. Please let me know what would be required to look into this further. 

I should mention i have the same issue with software/hardware NAT offloading turned on or off and as mentioned the slow(er) transfer rate also occurs with stock firmware.

[image]

cryobry, 2022-11-04:
Try using iperf for benchmarks, Windows-reported file transfer speeds are not a reliable measurement tool.

graysky2, 2022-11-04:
Agreed, you can also use `rsync` to move files and get some speed stats in addition to `time`

shauno100, 2022-11-05:
> Try using iperf for benchmarks, Windows-reported file transfer speeds are not a reliable measurement tool.

Thats fine, I will test further. What I am saying is that when i use any of my other routers in exact same setup file transfer rate is around 115MB/s to my computer from NAS but when i using AX6000 it's never more then 90MB/s so something is clearly different here.

zk00790, 2024-11-12:
Encountered the same problem
[image]
[image]

shauno100, 2024-11-21:
I've moved on and have a different router, i hope you guys have success in getting this looked at by the devs but considering 2 years have passed since i reported the issue i doubt it will ever be resolved.