openwrt/openwrt #11169: iptables display wrong rules

ptpt52, 2022-11-05:
setup rules by:
```
iptables -A PREROUTING -s 0.0.0.0/32 -m mark --mark 0x0/0xff000000 -m conntrack --ctstate NEW -m comment --comment natcap-rule -j MARK --set-xmark 0xd000000/0xff000000
```

iptables-save: display wrong output:
```
# Generated by iptables-save v1.8.8 (nf_tables) on Sat Nov  5 16:36:58 2022
*mangle
:PREROUTING ACCEPT [0:0]
:INPUT ACCEPT [0:0]
:FORWARD ACCEPT [0:0]
:OUTPUT ACCEPT [0:0]
:POSTROUTING ACCEPT [0:0]
-A PREROUTING -s 0.0.0.0/32 -m mark --mark 0x0/0xff000000 -m conntrack --ctstate NEW -m comment --comment natcap-rule -j MARK --set-xmark 0xd000000/0xff000000
COMMIT
# Completed on Sat Nov  5 16:36:58 2022
```

ptpt52, 2022-11-05:
installed ipks:
```
iptables-mod-conntrack-extra - 1.8.8-1
iptables-mod-ipopt - 1.8.8-1
iptables-nft - 1.8.8-1
```

no  iptables-legacy

ptpt52, 2022-11-05:
this ping @hauke @champtar

champtar, 2022-11-06:
I can reproduce it with Fedora 37
-> reported upstream https://bugzilla.netfilter.org/show_bug.cgi?id=1642

ptpt52, 2022-11-28:
> I can reproduce it with Fedora 37 -> reported upstream https://bugzilla.netfilter.org/show_bug.cgi?id=1642

looks like fixed by commit f315af1cf8871 ("nft: track each register individually")
could you help backport it? @champtar

Spudz76, 2022-12-02:
I modded the package to pull the latest upstream git master from last night (`39589b3edb`) and it has this and a bunch of other fixes (but could have even more, different bugs of course).  But, it builds...

[iptables-upstream-git-master-39589b3edb.patch.txt](https://github.com/openwrt/openwrt/files/10144361/iptables-upstream-git-master-39589b3edb.patch.txt)

Neustradamus, 2022-12-27:
To follow this ticket

rodsmar, 2024-09-25:
I submitted an PR to backport this commit.

> > I can reproduce it with Fedora 37 -> reported upstream https://bugzilla.netfilter.org/show_bug.cgi?id=1642
> 
> looks like fixed by commit f315af1cf8871 ("nft: track each register individually") could you help backport it? @champtar

PR #16504 (2024-10-19): iptables: backport "nft: track each register individually" from 1.9
From the upstream repo:

`Instead of assuming only one register is used, track all 16 regs individually.

This avoids need for the 'PREV_PAYLOAD' hack and also avoids the need to clear out old flags:

When we see that register 'x' will be written to, that register state is reset automatically.

Existing dissector decodes
ip saddr 1.2.3.4 meta l4proto tcp
... as
-s 6.0.0.0 -p tcp

iptables-nft -s 1.2.3.4 -p tcp is decoded correctly because the expressions are ordered like:

meta l4proto tcp ip saddr 1.2.3.4
                     [... cut]
commit 44b1993f (2024-10-19): iptables: backport "nft: track each register individually" from 1.9

From the upstream repo:

Instead of assuming only one register is used, track all 16 regs
individually.

This avoids need for the 'PREV_PAYLOAD' hack and also avoids the need to
clear out old flags:

When we see that register 'x' will be written to, that register state is
reset automatically.

Existing dissector decodes
ip saddr 1.2.3.4 meta l4proto tcp
... as
-s 6.0.0.0 -p tcp

iptables-nft -s 1.2.3.4 -p tcp is decoded correctly because the expressions
are ordered like:

meta l4proto tcp ip saddr 1.2.3.4
                     [... cut]
commit 088bb1e1 (2024-10-23): iptables: backport "nft: track each register individually" from 1.9

From the upstream repo:

Instead of assuming only one register is used, track all 16 regs
individually.

This avoids need for the 'PREV_PAYLOAD' hack and also avoids the need to
clear out old flags:

When we see that register 'x' will be written to, that register state is
reset automatically.

Existing dissector decodes
ip saddr 1.2.3.4 meta l4proto tcp
... as
-s 6.0.0.0 -p tcp

iptables-nft -s 1.2.3.4 -p tcp is decoded correctly because the expressions
are ordered like:

meta l4proto tcp ip saddr 1.2.3.4
                     [... cut]
closed 2024-10-19 by openwrt-bot