openwrt/openwrt #11170: ath79: add support for TP-Link CPE605-v1

ruildias, 2022-11-05:
TP-Link CPE605-v1 is an outdoor wireless CPE for 5 GHz with one Ethernet port based on Atheros AR9344

Specifications:

Based on the same underlying hardware as the TP-Link CPE610 v2 (reference issue [#3120](https://github.com/openwrt/openwrt/pull/3120))

Power, LAN0, WLAN green LEDs	
23dBi high-gain directional antenna and a dedicated metal reflector
Flashing instructions:
Flash factory image through stock firmware WEB UI
or through TFTP
To get to TFTP recovery just hold reset button while powering on for
around 4-5 seconds and release.
Rename factory image to recovery.bin
Stock TFTP server IP:192.168.0.100
Stock device TFTP adress:192.168.0.254

PS: I am starting this request for new device support with the expertise of @apcameron that build an image that is running stable on the device since 2 days ago, based on:

Firmware Version | OpenWrt 21.02.5 r16688-fa9a932fdb / LuCI openwrt-21.02 branch git-22.288.45155-afd0012

Attached the modified files, by him in order to make it work.
Pls. I am trying to compile with the version 22.03.1 and to do that, I had to change a line of code (against the original from @apcameron):  [.soft_ver = SOFT_VER_DEFAULT instead of NULL]
[01_leds.pdf](https://github.com/openwrt/openwrt/files/9943443/01_leds.pdf)
[02_network.pdf](https://github.com/openwrt/openwrt/files/9943444/02_network.pdf)
[ar9344_tplink_cpe605-v1.dts.pdf](https://github.com/openwrt/openwrt/files/9943445/ar9344_tplink_cpe605-v1.dts.pdf)
[generic-tp-link.mk.pdf](https://github.com/openwrt/openwrt/files/9943446/generic-tp-link.mk.pdf)
[tplink-safeloader.c.pdf](https://github.com/openwrt/openwrt/files/9943447/tplink-safeloader.c.pdf)

`/** Firmware layout for the CPE210/220 V1 */
	{
		.id     = "CPE210",
		.vendor = "CPE510(TP-LINK|UN|N300-5):1.0\r\n",
		.support_list =
			"SupportList:\r\n"
			"CPE210(TP-LINK|UN|N300-2):1.0\r\n"
			"CPE210(TP-LINK|UN|N300-2):1.1\r\n"
			"CPE210(TP-LINK|US|N300-2):1.1\r\n"
			"CPE210(TP-LINK|EU|N300-2):1.1\r\n"
			"CPE220(TP-LINK|UN|N300-2):1.1\r\n"
			"CPE220(TP-LINK|US|N300-2):1.1\r\n"
			"CPE220(TP-LINK|EU|N300-2):1.1\r\n",
		.part_trail = 0xff,
		.soft_ver = SOFT_VER_DEFAULT,

		.partitions = {
			{"fs-uboot", 0x00000, 0x20000},
			{"partition-table", 0x20000, 0x02000},
			{"default-mac", 0x30000, 0x00020},
			{"product-info", 0x31100, 0x00100},
			{"signature", 0x32000, 0x00400},
			{"firmware", 0x40000, 0x770000},
			{"soft-version", 0x7b0000, 0x00100},
			{"support-list", 0x7b1000, 0x00400},
			{"user-config", 0x7c0000, 0x10000},
			{"default-config", 0x7d0000, 0x10000},
			{"log", 0x7e0000, 0x10000},
			{"radio", 0x7f0000, 0x10000},
			{NULL, 0, 0}
		},

		.first_sysupgrade_partition = "os-image",
		.last_sysupgrade_partition = "support-list",
	},`

apcameron, 2022-11-05:
For anyone wanting more details you can refer to the post at [https://forum.openwrt.org/t/openwrt-support-for-tp-link-cpe605/141192](url)

ruildias, 2022-11-06:
Update: built a successful image with ver 22.03.1, stable.

apcameron, 2022-12-17:
This was added to Master via [This Pull Request](https://github.com/openwrt/openwrt/pull/11454) which has been merged.

ruildias, 2022-12-18:
On its way!! Thank you @apcameron

closed 2022-12-18 by ruildias

Fboriongan5, 2024-08-10:
Good day everyone any one can help after i upgrade my tplink cpe605 to openwrt it successfully install but one day i tried to back it to stock firmware and i download the original firmware and i uploaded there in webpage but when it uploaded it never work and function it just stop blinking and it just the light blue is steady so anyone can help me fix this i always trying to press the reset button but nothing change😭😭
[image]

Maralexbar, 2024-08-31:
@Fboriongan5 Reverting to Stock Firmware on most TP-Link models is via tftp.

Download the Stock firmware from their official website and use the same installation method via tftp (Check the first message), NEVER DO IT FROM LUCI it is only for sysupgrade images.

Fboriongan5, 2024-08-31:
But after i flash the firmware of stock it seem the range signal it gets
short only and also when reinstall again the openwrt firmware the wireless
ssid not in the option even brige or client mode nothing the wireless
option not in the list why it only ahow is the interface, firewall, routing
but the wireless is not there?