openwrt/openwrt #11177: Add XFS support to block extroot

crass, 2022-11-06:
I'm attempting to setup extroot and one of several problems I've encountered following the official wiki is that `block extroot` [does not support XFS](https://git.openwrt.org/?p=project/fstools.git;a=blob;f=block.c;hb=HEAD#l1559) filesystems. Is there any real reason for this or has the whitelist just not been updated? I find it hard to believe that BTRFS is supported while XFS not.

Mijzelf, 2022-11-15:
Not all types of an xfs filesystem work together nicely with overlayfs
https://medium.com/@khushalbisht/docker-on-centos-7-with-xfs-filesystem-can-cause-trouble-when-d-type-is-not-supported-64cee61b39ab

crass, 2022-11-16:
Hmm, good to know. Looks like `mkfs.xfs` has had `ftype=1` as default since at [least 2015](https://groups.google.com/g/linux.debian.bugs.dist/c/OfWfXf3l2KU/m/cDgo7_CIEwAJ). If you're creating an XFS volume today it will be on by default on pretty much every operating system. I expect that 99% of users wanting to use extroot will be creating a new filesystem on a USB. Not using a 7 year old filesystem. Nor are they likely using Centos 7 and haven't updated to more recent xfsprogs. I think support can be enabled and if anything add some documentation to the wiki.