openwrt/openwrt #11186: Dramatic decrease of performance after upgrading TP-Link Archer C7 v2 to 22.03

svillar, 2022-11-06:
Hi,

after upgrading my TP-Link Archer C7 v2 to `22.03.0 r19685-512e76967f` I noticed a severe reduction of transfer rate (WAN interface) in both download and upload speeds, but specially in the download.

I have a symmetric 1Gb fiber connection, and the TP-Link used to be able to provide 700Mbps download and 500Mbps upload. However after upgrading to 22.03 using the same configuration they got down to 210Mbps down and 350 Mbps up. Initially I thought the problem was some change in the configuration but after trying many different configurations (I keep backups of all the changes) I safely concluded that was not the issue. Then I decided to downgrade back to `OpenWrt 21.02.5 r16688-fa9a932fdb` and suddenly just after doing that (with no changes in the config) I got the previous figures in terms of transfer rate.

I don't know the reason (maybe nftables performance?) but the router performance is much worst in 22.03. I'll keep using 21.02.5 in the meantime. I'll be happy to help debugging the issue, just let me know what is needed.

xabolcs, 2022-11-07:
Duplicate of #10224.
If you really want to use 22.03 then you could try 22.03.0-rc1 which isn't affected!

svillar, 2022-11-07:
> Duplicate of #10224.
> If you really want to use 22.03 then you could try 22.03.0-rc1 which isn't affected!

Right, it's indeed a software offload issue with PPoE connections. Thanks for the pointers.

closed 2022-11-07 by svillar