openwrt/openwrt #11187: OpenWrt 22.03, USB port not working properly on Linksys MR8300 V1.1

duc5800, 2022-11-06:
USB drives that rely on the power supplied by the USB port (e.g. Flash disks, SD Card readers) don't really work at all.
According to the logs the devices are detected by the router, but a short moment later disconnect again.

USB drives that have their own external power supply appear to work fine for periods of time, but may also experience connection issues after a while.

There is a thread with details about this issue in the OpenWrt forum.
https://forum.openwrt.org/t/linksys-mr8300-openwrt-22-03-0-rc4-usb-port-powered-storage-devices-not-working/131391

sscheib, 2023-01-25:
I can definitively confirm this behavior, although I am unsure whether it's actually the power draw that's an issue.
I have tried a vast amount of different USB sticks (everything from USB 1 to latest USB 3.2 Gen 1), both with and without active USB hub and all showed the very same behavior.
The USB device was recognized and immediatly afterwards resetted. This already happens during the boot time multiple times (can be confirmed via logread or serial access).
What I have found that did also *not* work, was generally managing a USB device from within the original Linksys firmware - the page simply never loaded. This happened with both preformatted (as FAT, EXFAT, NFTS, EXT4) as well as raw (unformatted) USB devices - I am actually unsure if this USB port is even properly working.

Further, I have measured the power draw of all of these devices and found that enough power was in fact drawn by the device (compared the values of the USB devices inserted to a default Fedora installation) when it is inserted into the router.
I also tried to mess around with kernel USB quirks to see if there is some flag that could help to resolve that issue or at least make it less often happen, but I couldn't make any meaningful progress there either.

Unfortunately, I am not out of ideas. Maybe somebody has some hints on how to debug this further, as the MR8300 seems to be the only available replacement for my beloved EA8300 devices.

I really appreciate any hints.

All the best,
Steffen

sscheib, 2023-02-12:
I found a working solution (cross posting from https://forum.openwrt.org/t/linksys-mr8300-openwrt-22-03-0-rc4-usb-port-powered-storage-devices-not-working/131391/64?u=sscheib). I have done more thorough testing [here](https://forum.openwrt.org/t/linksys-mr8300-openwrt-22-03-0-rc4-usb-port-powered-storage-devices-not-working/131391/69?u=sscheib)

Solution (for an already booted system):
```
# install tools
opkg install gpiod-tools gpioctl-sysfs
# figure out the base of the GPIO chip
cat /sys/class/gpio/gpiochip*/base | head -n1
412

# as the 68th PIN is referenced in Linksys' firmware with the power reset, add that to the base to determine the exact PIN: 412 + 68 = 480
# export the PIN 480
echo "480" > /sys/class/gpio/export

# set PIN direction
echo "out" > /sys/class/gpio/gpio480/direction
 
# set PIN to active_low
echo "1" > /sys/class/gpio/gpio480/active_low

# disable USB port (you should receive a disconnect of USB device of the kernel with that command - see logread or dmesg)
echo "0" > /sys/class/gpio/gpio480/value

# enable USB port (you should receive a connect of a connected USB device from the kernel - see logread or dmesg)
echo "1" > /sys/class/gpio/gpio480/value

# profit
mount /dev/sda1 /mnt/
[ 1162.319823] EXT4-fs (sda1): mounted filesystem with ordered data mode. Opts: (null). Quota mode: disabled.
root@OpenWrt:/# dd if=/dev/zero of=/mnt/test.file.10G bs=1M count=10000
10000+0 records in
10000+0 records out
root@OpenWrt:/# du -hs /mnt/test.file*
1000.0M /mnt/test.file
9.8G    /mnt/test.file.10G
root@OpenWrt:/#
```

This works obviously only temporarily until the system is rebooted. I guess one needs to change the DTS-definition so that the kernel is applying an active_low to pin # 480 from the get-go. I have honestly know idea how to achieve that. It would be great if somebody who has some experience in working with DTS-definitions would be so glad to help me out here.
I am more than happy to try to changes that need to be applied, but I would need a helping hand here.

Could somebody give me a helping hand on modifying the DTS definition?

Thanks, and all the best,
Steffen

hauke, 2023-02-25:
The problem was fixed in ed64c3323590e3c9fa8b423bf37689023a7a101f (#12024)

closed 2023-02-25 by hauke