openwrt/openwrt #11208: SSH multiplexing unstable since OpenWrt 22.03.2 / dropbear 2022.82

sk2449, 2022-11-09:
**Background**: I have a small monitoring script, which pulls data from devices via SSH to check their health. Since several threaded checks are run against each machine every minute, i use SSH multiplexing to save system resources.

**Observation**: Since upgrading to 22.03.2 (from 21.02.?) Dropbear closes the master connection unexpectedly (according to "ssh -vvv") as soon as two checks happen to use the master connection at the same time. This causes one of the two simultaneous checks to fail, because the SSH client does not return any data in that case.

**What i have tested so far**:
OpenSSH server on the same OpenWrt host -> stable
The same version of Dropbear (2022.82) under Debian -> also stable
Different SSH client versions (OpenSSH_9.0p1 and OpenSSH_8.4p1) -> no impact on the result

**How to reproduce**:
This issue is reproducible for the standard image for TP-Link TL-WR1043N/ND v4, v3, v2 and x86-64 combined-EFI (squashFS) using the following steps:

Open the master connection:
`ssh root@<openwrt-host> -vvv -S .ssh/mastersocket/<openwrt-host> -o BatchMode=yes -o ControlMaster=auto -o ControlPersist=2m "date"`

Open a new (virtual) terminal (appears to be necessary to prevent the SSH client processes from controlling/hijacking each other's stdout) and run an infinite loop of ssh commands utilizing the master connection until something goes wrong based on the exit code:
`while true; do ssh root@<openwrt-host> -vvv -S .ssh/mastersocket/<openwrt-host> -o BatchMode=yes -o ControlMaster=auto -o ControlPersist=2m "date"; if [ $? -ne 0 ]; then break; fi; done`

Run another loop in another new (virtual) terminal and observe one of the two loops exiting after a few seconds with:
`"debug2: Control master terminated unexpectedly"`

**Additional notes**:
"logread" doesn't show any entries from Dropbear when this happens. I tried to increase verbosity as described at https://openwrt.org/docs/guide-user/base-system/dropbear (verbose=1), but didn't see any additional output (does the mentioned "start script" log anywhere else?).
I have also analyzed the TCP flows using Wireshark to be sure, that no packets are lost while testing -> all fine; no RST; no resends

Any hints would be appreciated, because i'm not familiar with more in-depth debugging of such behaviour.
Thank you.

rockdrilla, 2022-11-14:
@sk2449 please take a look at #11223.

sk2449, 2023-10-22:
Issue is still reproducible on OpenWrt 23.05.0 / dropbear 2022.82-5.

rockdrilla, 2023-10-22:
@sk2449, I'd like to ask you to build firmware with MR noted above and write your feedback. Thanks in advance.

sk2449, 2024-04-07:
@rockdrilla Sorry for my delayed reply.
It looks like multiplexing is more stable now with OpenWRT 23.05.3 / dropbear 2022.82-6 .
I have to run the loops for several minutes to cause an unexpectedly closed connection compared to a few seconds before.
Will observe it in real-world usage for a few weeks and report back again afterwards.

rockdrilla, 2024-04-07:
AFAIK "openwrt-23.05" branch doesn't contain changes that reflect your issue. However, I can backport and build packages for you if it's applicable to you.

sk2449, 2024-04-13:
You are right. The build scripts are still pulling v2022.82-6.
Unfortunately i have already rolled back to OpenSSH after experiencing the same disconnects again after a few hours with no spare device left for easy testing at the moment.
I guess i will wait until the patches found their way into the main branch.