openwrt/openwrt #11210: Feature request: web GUI-based creating and editing CNAME records (without hand-editing /etc/dnsmasq.conf)

esbeeb, 2022-11-10:
Hello,
I was able to configure everything I wanted in OpenWRT's web gui, _except_ creating cname records.  That sort of felt like an enbarrasment, holding me back from recommending OpwnWRT to my less-technical friends and peers.  

N_ote: Please see [here](https://forum.openwrt.org/t/getting-dnsmasq-to-create-and-report-cname-records/20768/13?u=esbeeb), as to how/why I created CNAME records, for context._

PS: In order to edit /etc/dnsmasq.conf, I had to install "nano", which was not installed by default.  If I had my druthers, the "[ne](https://ne.di.unimi.it)" text editor would have been even better (than nano), just as easily available for gui-based software installation.  It's more user friendly than nano (tap Escape to see a menubar along the top, with familiar menus like "File", and several familiar keyboard shortcuts work, for example).

esbeeb, 2022-11-10:
Actually, on second thought, creating a second A record (in "Network" -> "DHCP and DNS" -> "Hostnames" -> "Add" button), _which points to the same IP address_ is acceptable, and works just as well as creating a CNAME record, IMHO.

So please disregard this feature request.  Additional A records re-using IP addresses is allowed.  That functionality is GUI-based, which is what I was wanting: to stay within the GUI, and not drop down to the CLI.

pkgadd, 2022-11-11:
While not exposed by luci, you can store `CNAME` definitions in uci (ending up in `/etc/config/dhcp`)just fine, without resorting to `/etc/dnsmasq.conf` editing.

ErwanMAS, 2023-07-30:
a pr is ready to be commited since april 2023 , https://github.com/openwrt/luci/pull/6335

the pr need some support .