openwrt/openwrt #11218: ARP-requests not being answered on MediaTek-chipsets

Starhowl, 2022-11-11:
Affected device with corresponding switch chip:
- ASUS RT-AC58U: Qualcomm Atheros QCA8075
- ASUS RT-AC85: MediaTek MT7621
- Linksys AX3200: MediaTek MT7531BE
- Xiaomi Mi Router 3 Pro: MediaTek MT7621AT

Threads about the issue:

- [ARP requests not answered](https://forum.openwrt.org/t/arp-requests-not-answered/138418)
- [ARP resolution failing in LAN](https://forum.openwrt.org/t/arp-resolution-failing-in-lan/104550)

schuettecarsten, 2022-11-11:
Take a look at https://github.com/openwrt/openwrt/issues/11029, did you try to use a SNAPSHOT build?

Starhowl, 2022-11-11:
> Take a look at #11029, did you try to use a SNAPSHOT build?

[Snapshot builds](https://downloads.openwrt.org/snapshots/targets/mediatek/) appear to only be available for MediaTek MT7622 whereas my device is using a MediaTek MT7621 for switching.

mrkiko, 2022-11-11:
MT7621 is in ramips:
https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/

Starhowl, 2022-11-11:
> MT7621 is in ramips: https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/

I tried a sysupgrade to latest snapshot while retaining settings, but the issue persisted.

So I tried to install [asus_rt-ac85p-squashfs-factory.bin](https://downloads.openwrt.org/snapshots/targets/ramips/mt7621/openwrt-ramips-mt7621-asus_rt-ac85p-squashfs-factory.bin) but I'm getting an error message about image metadata being missing, despite the hashes of the one from the error dialogue and file being the same.

I fear bricking my device in case of a forced upgrade despite the warning. What do you think - is it safe?

mrkiko, 2022-11-11:
>     MediaTek-chipsets (Issue #11218)

No, and you should not do it.
Maybe try resetting your config to default (maybe after having exported it, so you can re-import it later), and see if the problem still exists.

Starhowl, 2022-11-11:
> No, and you should not do it. Maybe try resetting your config to default (maybe after having exported it, so you can re-import it later), and see if the problem still exists.
> […](#)

I just reset the upgraded firmware image to defaults and the issue persists.

schuettecarsten, 2022-11-21:
Maybe, this is related to #11277 and #11029

kvuorine, 2023-03-12:
@schuettecarsten I don't think this is related to those, I think I am experiencing this on r21808+457-b7dfbc2aa4 but this problem doesn't timeout away, network restart solves this for a while.

I have done some sniffing on interfaces found on RT-AX53U, because vlans with heavier traffic seem to stop working quite fast.

After failure, openwrt ping requests move out br-lan.6 -> br-lan tagged -> lan1 tagged and responses are received on lan1 tagged but after that, nothing. Same thing with ARP traffic.

schuettecarsten, 2023-03-13:
@kvuorine If you build your images yourself, take a look at the patch that's attached to https://github.com/openwrt/openwrt/issues/9706. This patch solved my problems.

kvuorine, 2023-03-14:
@schuettecarsten I built a system with that patch applied, it did not solve my problem. Was your problem occuring only with WLAN clients or did your AP also lose connection?

kvuorine, 2023-03-14:
AP I am having problems with also is supposed to be my new gateway at some point.

GS1900 Switch tags untagged packets going into AP with ID 1 what it should not do with current settings, which is also PVID on the AP.

AP  receives those packets on device br-lan.1 , and responds with untagged packets as supposed.

When this error situation occurs, packets destined to device br-lan.6 coming from switch tagged with ID 6 get dropped.

But packets destined to IP-address assigned to br-lan.6 coming in with ID 1 are getting a response and not from device br-lan.6 , this shouldn't happen.

Ditti4, 2023-09-10:
Any news on this issue? I'm running into this same issue using two D-Link COVR-X1860, with and without the patch attached to #9706. This happens on different devices and more so on devices that fast-roam a lot between the two APs.

While the ARP replies are still generated, they don't make it to the target machine anymore. A bridge restart on the affected AP fixes the issue and so does re-connecting the WiFi client (which then defeats the purpose of fast roaming). At first I assumed it was related to my unmanaged switch as restarts of this seemed to fix it, but that seems to have been a coincidence - I just ran into the issue again and restarting the unmanaged switch did nothing.

Are there any more details I can provide?

corvin84, 2024-03-11:
+1 - Having this issue on 23.05.2 as well. Did not have any problem running 22 branch. Any news on the fix?

scMarkus, 2024-05-20:
Hi,
I first posted in the OpenWrt forum threads (linked at the top) and only afterwards found this issue. +1 for me since issue seams to persist for me as well. I detailed my observations over [here](https://forum.openwrt.org/t/arp-requests-not-answered/138418/14?u=nocte)

Neustradamus, 2024-12-27:
How is the current situation?