openwrt/openwrt #11245: Deprecated nginx ssl protocols TLSv1.0 and TLSv1.1 should be disabled by default

ArtelMike, 2022-11-14:
The current default for nginx ssl is to have TLSv1.0 and TLSv1.1 enabled. This causes security scans to fail. The following should be added to /etc/nginx/uci.conf as the default.

` uci set nginx._lan.ssl_protocols='TLSv1.2 TLSv1.3'
`

which results in

```
server {
    ...
    ssl_protocols TLSv1.2 TLSv1.3;
    ...
}

```
The following is a summary of a security scan

**Summary**
It was possible to detect the usage of the deprecated TLSv1.0 and/or TLSv1.1 protocol on this system.
**Vulnerability Detection Result**
In addition to TLSv1.2+ the service is also providing the deprecated TLSv1.0 and
,→ TLSv1.1 protocols and supports one or more ciphers. Those supported ciphers c
,→an be found in the 'SSL/TLS: Report Supported Cipher Suites' (OID: 1.3.6.1.4.1
,→.25623.1.0.802067) VT.
**Impact**
An attacker might be able to use the known cryptographic flaws to eavesdrop the connection between clients and the service to get access to sensitive data transferred within the secured connection.
Furthermore newly uncovered vulnerabilities in this protocols won't receive security updates anymore.
**Solution:**
**Solution type:** Mitigation
It is recommended to disable the deprecated TLSv1.0 and/or TLSv1.1 protocols in favor of the TLSv1.2+ protocols. Please see the references for more information.

@ArtelMike

Spudz76, 2022-11-20:
Yeah but what if you need to get in to the router and all you have is IE8?

ynezz, 2022-11-21:
nginx package is being maintained in packages feed, so https://github.com/openwrt/packages/issues is appropriate place for such issues.

closed 2022-11-21 by ynezz