openwrt/openwrt #11246: netifd improperly running ubus command

crass, 2022-11-14:
In debugging why one of my wireless interfaces is failing, I'm seeing this in the logs:

```
Mon Nov 14 17:51:57 2022 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0)
Mon Nov 14 17:51:57 2022 daemon.err hostapd: Interface name wlan0 already in use
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566): Command failed: ubus call hostapd config_add {"iface":"wlan0", "config":"/var/run/hostapd-phy0.conf"} (Invalid argument)
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566): Usage: ubus [<options>] <command> [arguments...]
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566): Options:
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566):  -s <socket>:        Set the unix domain socket to connect to
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566):  -t <timeout>:        Set the timeout (in seconds) for a command to complete
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566):  -S:                     Use simplified output (for scripts)
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566):  -v:                     More verbose output
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566):  -m <type>:              (for monitor): include a specific message type
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566):                  (can be used more than once)
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566):  -M <r|t>                (for monitor): only capture received or transmitted traffic
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566):
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566): Commands:
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566):  - list [<path>]                 List objects
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566):  - call <path> <method> [<message>]      Call an object method
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566):  - subscribe <path> [<path>...]  Subscribe to object(s) notifications
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566):  - listen [<path>...]                    Listen for events
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566):  - send <type> [<message>]               Send an event
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566):  - wait_for <object> [<object>...]       Wait for multiple objects to appear on ubus
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566):  - monitor                               Monitor ubus traffic
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566):
Mon Nov 14 17:51:57 2022 daemon.notice netifd: radio0 (15566): Device setup failed: HOSTAPD_START_FAILED
Mon Nov 14 17:51:57 2022 daemon.notice netifd: Wireless device 'radio0' set retry=0
Mon Nov 14 17:51:57 2022 daemon.crit netifd: Wireless device 'radio0' setup failed, retry=0
Mon Nov 14 17:51:57 2022 kern.debug kernel: [29745.591851] ieee80211 phy0: change: 0x40
Mon Nov 14 17:51:57 2022 daemon.notice hostapd: Remove interface 'wlan0'
Mon Nov 14 17:51:57 2022 daemon.notice hostapd: wlan0: interface state ACS->DISABLED
Mon Nov 14 17:51:57 2022 daemon.notice hostapd: wlan0: AP-DISABLED
Mon Nov 14 17:51:57 2022 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Mon Nov 14 17:51:57 2022 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Mon Nov 14 17:51:57 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Mon Nov 14 17:51:57 2022 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Mon Nov 14 17:51:57 2022 kern.info kernel: [29745.618599] device wlan0 left promiscuous mode
```

This is being called from `/lib/netifd/wireless/mac80211.sh` line 1153. I've strace'd netifd and got the [ubus process execution trace](https://github.com/openwrt/openwrt/files/10006356/netifd.ubus.strace.txt). The arguments to ubus look fine and it appears as though ubus is sending the request to ubusd and that the call is returning. Looking in the ubus source, I think the usage is getting printed out [here](https://git.openwrt.org/?p=project/ubus.git;a=blob;f=cli.c;h=5d2cd4c9cfbcec341028f41dfb480ab4487ca29e;hb=HEAD#l668) at line 668. If this is true, then `ret` == `-2`. Since the command was run, this must mean that`ubus_cli_call` is returning `-2`.

 the return status of `hostapd config_add` is `-2`. I haven't looked into whether or when this happens.

Assuming that the above it true, then the printing of usage is a redherring (one expects the usage to be printed when it would be helpful, such as failure to parse arguments). The issue is that `ret` represents two different sets of numbers that overlap and is checking for `-2` from one set, and not considering that it can come from another set.

Perhaps the function `ubus_cli_call` (along with other commands) should be changed from `int ubus_cli_call(struct ubus_context *ctx, int argc, char **argv)` to `int ubus_cli_call(struct ubus_context *ctx, int argc, char **argv, int *err)`. If `ubus_cli_call` returns `0`, then the value in `err` is valid and can be processed.

This doesn't fix the same issue for the parent process trying to figure out what a return code of `-2` means. I judge that to be less of an issue though.

ptpt52, 2022-11-15:
https://github.com/openwrt/openwrt/pull/11227

hnyman, 2023-02-11:
I ran into these symptoms today.
Somehow my router's wifi got locked so that it did not gracefully turn off, which then caused errors at the subsequent wifi up/wifi down.  

Possibly the original reason is that I had a spelling mistake in the wireless uci config file ("options" vs. "option" keyword), which I then corrected.  The strange part was that the radio was up all the time, and fixing the wireless config file did not change change things at all.

More detailed log, how this error can manifest itself:

First both radios are up, and should be taken down with 'wifi down':

```
root@router1:/etc/config# iwinfo
phy0-ap0  ESSID: "HN5WLAN"
          Access Point: DC:EF:09:EF:F3:E6
          Mode: Master  Channel: 100 (5.500 GHz)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 0 dBm  Link Quality: unknown/70
          Signal: unknown  Noise: -104 dBm
          Bit Rate: unknown
          Encryption: WPA3 SAE (CCMP)
          Type: nl80211  HW Mode(s): 802.11ac/n
          Hardware: 168C:0046 168C:CAFE [Qualcomm Atheros QCA9984]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy0

phy1-ap0  ESSID: "HNWLAN"
          Access Point: DC:EF:09:EF:F3:E7
          Mode: Master  Channel: 3 (2.422 GHz)  HT Mode: HT20
          Center Channel 1: 3 2: unknown
          Tx-Power: 20 dBm  Link Quality: 70/70
          Signal: -35 dBm  Noise: -95 dBm
          Bit Rate: 65.0 MBit/s
          Encryption: WPA2 PSK (CCMP)
          Type: nl80211  HW Mode(s): 802.11b/g/n
          Hardware: 168C:0046 168C:CAFE [Qualcomm Atheros QCA9984]
          TX power offset: none
          Frequency offset: none
          Supports VAPs: yes  PHY name: phy1

root@router1:/etc/config# wifi down
root@router1:/etc/config# logread | tail -n 16
Sat Feb 11 12:15:56 2023 user.notice WIFI: **** break ****
Sat Feb 11 12:16:13 2023 daemon.notice hostapd: Remove interface 'phy1-ap0'
Sat Feb 11 12:16:13 2023 daemon.notice hostapd: phy1-ap0: interface state ENABLED->DISABLED
Sat Feb 11 12:16:13 2023 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED 30:cd:a7:b3:33:5d
Sat Feb 11 12:16:13 2023 daemon.notice hostapd: phy1-ap0: AP-DISABLED
Sat Feb 11 12:16:13 2023 daemon.notice hostapd: phy1-ap0: CTRL-EVENT-TERMINATING
Sat Feb 11 12:16:13 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sat Feb 11 12:16:13 2023 daemon.notice hostapd: nl80211: deinit ifname=phy1-ap0 disabled_11b_rates=0
Sat Feb 11 12:16:13 2023 kern.info kernel: [ 1249.146623] device phy1-ap0 left promiscuous mode
Sat Feb 11 12:16:13 2023 kern.info kernel: [ 1249.146851] br-lan: port 2(phy1-ap0) entered disabled state
Sat Feb 11 12:16:13 2023 daemon.notice netifd: Network device 'phy1-ap0' link is down
Sat Feb 11 12:16:13 2023 kern.warn kernel: [ 1249.191222] ath10k_pci 0001:01:00.0: could not get mac80211 beacon, vdev_id: 0 addr: dc:ef:09:ef:f3:e7
Sat Feb 11 12:16:13 2023 kern.info kernel: [ 1249.278248] ath10k_pci 0001:01:00.0: mac flush null vif, drop 0 queues 0xffff
Sat Feb 11 12:16:13 2023 kern.warn kernel: [ 1249.317327] ath10k_pci 0001:01:00.0: peer-unmap-event: unknown peer id 0
Sat Feb 11 12:16:13 2023 kern.warn kernel: [ 1249.317474] ath10k_pci 0001:01:00.0: peer-unmap-event: unknown peer id 0
Sat Feb 11 12:16:14 2023 daemon.notice netifd: Wireless device 'radio1' is now down
```

phy0-ap0 is not visible in the log at all, but there is error about removing a control dir. 

One radio is still up:
```
root@router1:/etc/config#
root@router1:/etc/config# iwinfo
phy0-ap0  ESSID: "HN5WLAN"
          Access Point: DC:EF:09:EF:F3:E6
          Mode: Master  Channel: 100 (5.500 GHz)  HT Mode: NOHT
          Center Channel 1: unknown 2: unknown
          Tx-Power: 0 dBm  Link Quality: unknown/70
[... 64 lines cut ...]
Sat Feb 11 12:25:22 2023 daemon.notice netifd: radio0 (8607):  -v:                      More verbose output
Sat Feb 11 12:25:22 2023 daemon.notice netifd: radio0 (8607):  -m <type>:               (for monitor): include a specific message type
Sat Feb 11 12:25:22 2023 daemon.notice netifd: radio0 (8607):                   (can be used more than once)
Sat Feb 11 12:25:22 2023 daemon.notice netifd: radio0 (8607):  -M <r|t>         (for monitor): only capture received or transmitted traffic
Sat Feb 11 12:25:22 2023 daemon.notice netifd: radio0 (8607):
Sat Feb 11 12:25:22 2023 daemon.notice netifd: radio0 (8607): Commands:
Sat Feb 11 12:25:22 2023 daemon.notice netifd: radio0 (8607):  - list [<path>]                  List objects
Sat Feb 11 12:25:22 2023 daemon.notice netifd: radio0 (8607):  - call <path> <method> [<message>]       Call an object method
Sat Feb 11 12:25:22 2023 daemon.notice netifd: radio0 (8607):  - subscribe <path> [<path>...]   Subscribe to object(s) notifications
Sat Feb 11 12:25:22 2023 daemon.notice netifd: radio0 (8607):  - listen [<path>...]                     Listen for events
Sat Feb 11 12:25:22 2023 daemon.notice netifd: radio0 (8607):  - send <type> [<message>]                Send an event
Sat Feb 11 12:25:22 2023 daemon.notice netifd: radio0 (8607):  - wait_for <object> [<object>...]        Wait for multiple objects to appear on ubus
Sat Feb 11 12:25:22 2023 daemon.notice netifd: radio0 (8607):  - monitor                                Monitor ubus traffic
Sat Feb 11 12:25:22 2023 daemon.notice netifd: radio0 (8607):
[... cut ...]
Sat Feb 11 12:25:27 2023 daemon.info hostapd: phy1-ap0: STA 30:cd:a7:b3:33:5d IEEE 802.11: associated (aid 1)
Sat Feb 11 12:25:27 2023 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 30:cd:a7:b3:33:5d auth_alg=open
Sat Feb 11 12:25:27 2023 daemon.info hostapd: phy1-ap0: STA 30:cd:a7:b3:33:5d RADIUS: starting accounting session D6F5A6B9773A4422
Sat Feb 11 12:25:27 2023 daemon.info hostapd: phy1-ap0: STA 30:cd:a7:b3:33:5d WPA: pairwise key handshake completed (RSN)
Sat Feb 11 12:25:27 2023 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 30:cd:a7:b3:33:5d
```