openwrt/openwrt #11247: AP fails to start on 5GHz radio when set with channel with radar detection

crass, 2022-11-15:
I'm running 22.03.2 on a WRT1900ACS. I have one AP configured on the 5GHz radio. When I have the channel manually set to one which is listed by `iw` with `(radar detection)` the AP fails to come up. The other channels work fine. Here's the list o channels output from `iw`:

```
                Frequencies:
                        * 5180 MHz [36] (23.0 dBm)
                        * 5200 MHz [40] (23.0 dBm)
                        * 5220 MHz [44] (23.0 dBm)
                        * 5240 MHz [48] (23.0 dBm)
                        * 5260 MHz [52] (24.0 dBm) (radar detection)
                        * 5280 MHz [56] (24.0 dBm) (radar detection)
                        * 5300 MHz [60] (24.0 dBm) (radar detection)
                        * 5320 MHz [64] (24.0 dBm) (radar detection)
                        * 5500 MHz [100] (24.0 dBm) (radar detection)
                        * 5520 MHz [104] (24.0 dBm) (radar detection)
                        * 5540 MHz [108] (24.0 dBm) (radar detection)
                        * 5560 MHz [112] (24.0 dBm) (radar detection)
                        * 5580 MHz [116] (24.0 dBm) (radar detection)
                        * 5600 MHz [120] (24.0 dBm) (radar detection)
                        * 5620 MHz [124] (24.0 dBm) (radar detection)
                        * 5640 MHz [128] (24.0 dBm) (radar detection)
                        * 5660 MHz [132] (24.0 dBm) (radar detection)
                        * 5680 MHz [136] (24.0 dBm) (radar detection)
                        * 5700 MHz [140] (24.0 dBm) (radar detection)
                        * 5720 MHz [144] (24.0 dBm) (radar detection)
                        * 5745 MHz [149] (30.0 dBm)
                        * 5765 MHz [153] (30.0 dBm)
                        * 5785 MHz [157] (30.0 dBm)
                        * 5805 MHz [161] (30.0 dBm)
```

Its unclear to me if `(radar detection)` means that the device does radar detection on those channels or if it needs to be specially setup to do radar detection on those channels. In any case, those do not work. Any ideas on how I can get this to work?

worstperson, 2022-11-15:
Failing from the hostapd configuration, looks just like the multi SSID bug #9343. Maybe check the contents /var/run/hostapd-phy0.conf to see what's invalid.

crass, 2022-11-15:
> Failing from the hostapd configuration, looks just like the multi SSID bug #9343. Maybe check the contents /var/run/hostapd-phy0.conf to see what's invalid.

No I don't think this is the multi SSID bug because this is being triggered when there is only one AP configured for the radio. 

Here's a diff of the contents of `/var/run/hostapd-phy0.conf` from a working config (one not on a radar detection channel) to a non-working config (only changing to a channel with radar detection):

```
--- /var/run/hostapd-phy0.conf  2022-11-15 06:02:36.378231982 +0000
+++ hostapd-phy0.noproblem.conf 2022-11-15 04:03:12.338793005 +0000
@@ -9,19 +9,19 @@
 hw_mode=a
 beacon_int=100
 dtim_period=2
-channel=56
-chanlist=56
+channel=36
+chanlist=36
 
 tx_queue_data2_burst=2.0
 ieee80211n=1
 ht_coex=0
-ht_capab=[HT40-][LDPC][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]
+ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]
 ieee80211ac=1
 vht_oper_chwidth=0
-vht_oper_centr_freq_seg0_idx=54
+vht_oper_centr_freq_seg0_idx=38
 vht_capab=[RXLDPC][SHORT-GI-80][SU-BEAMFORMER][SU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][SOUNDING-DIMENSION-4][BF-ANTENNA-4][MAX-A-MPDU-LEN-EXP7]
 
-radio_config_id=134ebe295558f3b57d32b937f9a6abc2
+radio_config_id=6fe58dc6bb6431adf9f6ee3f9e98e973
 interface=wlan0
 ctrl_interface=/var/run/hostapd
 ap_isolate=1
```

I tried to bring up the AP with `[HT40+]` instead of `[HT40-]` in `ht_capab`, and it still fails. Could it be that the channel I setting it to is being seen as occupied by the radar and hostapd is failing because of that?

worstperson, 2022-11-15:
Suppose it's this error I overlooked that could be triggering the invalid argument
`Tue Nov 15 05:09:19 2022 daemon.err hostapd: Interface name wlan0 already in use`
But I don't see how you could be getting vht_oper_centr_freq_seg0_idx=-2 in your other report.

`Tue Nov 15 05:09:07 2022 daemon.notice netifd: radio0 (11482): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process 2050 path (/rom/usr/sbin/wpad)`
Not sure what's going on here, like you have two different wpad binaries for some reason?

Shine-, 2022-11-15:
Known issue with the mwlwifi driver and its region handling. Also see #9956.
Very unlikely to ever be fixed, since the driver has been abandoned.

crass, 2022-11-16:
Thank you @Shine-, I've learn a lot from that and related issues.

I recently moved this router to another country from the US. Am I correct in thinking that these radar detection channels should have worked in the US? And they do not work now because the router is detecting the non-US regulatory domain of nearby routers and then disabling the radar detection channels? Or didn't they never work? (I don't remember trying when I was in the US).

crass, 2022-11-16:
> Suppose it's this error I overlooked that could be triggering the invalid argument `Tue Nov 15 05:09:19 2022 daemon.err hostapd: Interface name wlan0 already in use` But I don't see how you could be getting vht_oper_centr_freq_seg0_idx=-2 in your other report.

Given this issue with the region handling, is that `-2` still a bug?

> `Tue Nov 15 05:09:07 2022 daemon.notice netifd: radio0 (11482): WARNING (wireless_add_process): executable path /usr/sbin/wpad does not match process 2050 path (/rom/usr/sbin/wpad)` Not sure what's going on here, like you have two different wpad binaries (edit: called from different paths) for some reason?

This should be unrelated. They are the same binary. It has to do with a second mount_root I'm doing and wpad is not restarted.

Shine-, 2022-11-16:
I don't know how this is handled on the WRT1900ACS. I only have experience with the WRT32X that I owned in the past.

Apparently the mwlwifi driver compares the EEPROM-preset region against all other WiFi chips' regions, against the region selected in OpenWrt. If there's a mismatch, it either refuses to start at all, or it refuses to use DFS channels.

Rule of thumb: if `iw reg get` shows any mismatch between "global", "phy#0" and "phy#1" (and "phy#2" in case you have three radios like the WRT32X/WRT32000ACM), you can expect the mwlwifi driver to fail on you.

My WRT32X was set by EEPROM to "IT" region (despite being a UK model). The other WiFi chip (mwifiex) was unlocked. This apparently creates an unsolvable mismatch for the mwlwifi driver. People (with US model?) apparently report that DFS channels do work when removing the mwifiex driver and firmware (thus disabling one of the radios for good) and leaving the OpenWrt region at "driver default". Didn't work for me, iirc.

The only way I could make DFS channels work was using the patch that I attached in the other issue. Whether this causes the mwlwifi driver to use the EEPROM-locked region or the one I set in OpenWrt, I don't know. UK, IT, and the rest of EU are all ETSI regdomain, thus no difference in allowed channels.

I can't test anymore, since I sold my WRT32X and don't have any other mvebu based device either.

rongten, 2023-07-29:
Same problem on 23.05.0-rc2 on a TP-Link Archer C5 v1 (ath79/generic).

iw reg get  | grep -B1 country
global
country BE: DFS-ETSI
--
phy#1
country US: DFS-FCC
--
phy#0
country US: DFS-FCC

Setting channel 149 (so no DFS conflict) fixes it.

Shine-, 2023-07-31:
The `ath10k` driver doesn't verify your selected region against the EEPROM region (resp. hw default region). This issue is about the `mwlwifi` driver doing exactly that.

For your devices (C7v2/C5v1), this has nothing to with DFS or not, but rather with the bandwidth or TX power that you selected:

E.g. if you have 80 MHz bandwidth configured, channel 140 isn't a valid channel. Try channel 132 instead. Depending on your region and/or hardware, you might also need to adjust TX power - channel 144 (which is the upper channel for 140 @ 40 MHz) supports only 13dbm on your devices and ETSI regdomain, while channels 100-140 support 26dbm.

When setting fixed channels, you will have to make sure _on your own_ to choose a supported channel/bw/txpower combination for your region and hardware.

Try `iw phy phy0 info` (for your device - for others, it might be `phy1`) to see what's supported.

ahmeaqas, 2024-09-28:
There is a script to restart the radio automatically in case of DFS Radar Event. Have a look onto [Automatic Restart of 5GHz Radio Using DFS Channel on DFS Radar Detection](https://hobbytronics.pk/automatic-restart-of-5ghz-radio-using-dfs-channel-on-dfs-radar-detection/)