openwrt/openwrt #11251: Bug: vlan on mediatek

jasle, 2022-11-15:
I am trying to use vlan on a bpi r3 with MT7986 with OpenWrt SNAPSHOT, r21292-6fdeb48c1e, but the router gets completely unreachable via network as soon, as vlan tagged frames arrives. 
The following stack trace appears again and again on the serial console:

```
[  286.224640] ------------[ cut here ]------------
[  286.229258] invalid VLAN protocol: 0x4404
[  286.233273] WARNING: CPU: 0 PID: 395 at vlan_do_receive+0x128/0x35c
[  286.239526] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 cfg80211 slhc sfp nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 mdio_i2c libcrc32c crc_ccitt compat crypto_safexcel pwm_fan hwmon i2c_gpio i2c_algo_bit sha1_generic seqiv md5 des_generic libdes authencesn authenc leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd gpio_button_hotplug usbcore usb_common
[  286.308536] CPU: 0 PID: 395 Comm: napi/mtk_eth-6 Tainted: G        W         5.15.78 #0
[  286.316518] Hardware name: Bananapi BPI-R3 (DT)
[  286.321029] pstate: 80000005 (Nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  286.327971] pc : vlan_do_receive+0x128/0x35c
[  286.332223] lr : vlan_do_receive+0x128/0x35c
[  286.336475] sp : ffffffc009943b70
[  286.339773] x29: ffffffc009943b70 x28: 000000000000a888 x27: 0000000000000000
[  286.346889] x26: ffffffc008ac9370 x25: ffffffc009943cd0 x24: ffffff8000923000
[  286.354005] x23: ffffffc008ab1268 x22: 0000000000000001 x21: 0000000000000000
[  286.361120] x20: ffffffc009943c38 x19: ffffff80003d1600 x18: 0000000000000e62
[  286.368236] x17: 0000000000000000 x16: 0000000000000004 x15: ffffffc008ada1a0
[  286.375351] x14: 0000000000002b26 x13: 0000000000000e62 x12: ffffffc009943898
[  286.382466] x11: ffffffc008b321a0 x10: 00000000fffff000 x9 : ffffffc008b321a0
[  286.389582] x8 : 0000000000000000 x7 : ffffffc008ada1a0 x6 : 0000000000000001
[  286.396697] x5 : ffffff807fb876f8 x4 : 0000000000000000 x3 : 0000000000000027
[  286.403812] x2 : 0000000000000027 x1 : 0000000000000023 x0 : 000000000000001d
[  286.410928] Call trace:
[  286.413360]  vlan_do_receive+0x128/0x35c
[  286.417266]  __netif_receive_skb_core.constprop.0+0x22c/0xd00
[  286.422995]  __netif_receive_skb_list_core+0xd4/0x1e0
[  286.428028]  netif_receive_skb_list_internal+0x164/0x260
[  286.433321]  napi_complete_done+0x64/0x1c0
[  286.437401]  mtk_napi_rx+0x108/0x184
[  286.440961]  __napi_poll+0x34/0x174
[  286.444434]  napi_threaded_poll+0x154/0x1a0
[  286.448602]  kthread+0x11c/0x130
[  286.451817]  ret_from_fork+0x10/0x20
[  286.455377] ---[ end trace 14d05052df858367 ]---
```

@nbd168

ptpt52, 2022-11-15:
apply this patch https://github.com/x-wrt/x-wrt/commit/579774bf7d0c8b5bd9e0512f614a939bafa62780.patch
to see if the issue fixed?

nbd168, 2022-11-21:
Fixed in latest master

closed 2022-11-21 by nbd168