openwrt/openwrt #11259: Wireless interface name can not be longer than 15 characters

crass, 2022-11-16:
Currently, if I set a wireless interface to a name with more than 16 characters via `option ifname` of a `wifi-iface` config block in `/etc/config/wireless` the interface will fail to come up. Looking in the logs, I can see that the interface name, "iface678901234567890", is being truncated:

```
Tue Nov 15 06:45:25 2022 daemon.err hostapd: nl80211: kernel reports: Attribute failed policy validation
Tue Nov 15 06:45:25 2022 daemon.err hostapd: Failed to create interface iface67890123456: -22 (Invalid argument)
Tue Nov 15 06:45:25 2022 daemon.err hostapd: Failed to add BSS (BSSID=00:23:5c:fc:4d:c0)
Tue Nov 15 06:45:25 2022 daemon.err hostapd: Interface initialization failed
```

This indicated that somewhere (in netifd?) some code is truncating the interface name to 16 bytes. I suspect in an attempt to satisfy the maximum interface length requirement, but is off by 1. The maximum length of an interface name is 15 characters:

```
root@OpenWRT:~# iw phy phy0 interface add iface67890123456 type monitor
kernel reports: Attribute failed policy validation
command failed: Invalid argument (-22)
root@OpenWRT:~# iw phy phy0 interface add iface6789012345 type monitor
root@OpenWRT:~# iw dev iface6789012345 del
root@OpenWRT:~# 
```

crass, 2022-11-16:
@pkgadd  I don't think you fully comprehended the bug report. I can see why you responded as you did just by reading the title. So I'm giving it a better title.

The bug is not in the kernel, its in OpenWRT not truncating enough characters to satisfy the 15 character interface name limit set by the kernel.

pkgadd, 2022-11-17:
It's difficult to say how long the ifname is going to get, with `br-` or `pppoe-` prepended, so it's best to keep it short and free of funny characters (or upper case).

crass, 2022-11-17:
> It's difficult to say how long the ifname is going to get, with `br-` or `pppoe-` prepended, so it's best to keep it short and free of funny characters (or upper case).

The option `option ifname` allows completely specifying the interface name. There is no `br-` or `pppoe-` prepended. That only happens for auto generated interface names. The name will be truncated, and that's not the problem. The problem is that the truncating that is already happening isn't enough, ie. it is incorrect. The advice on interface naming is neither her nor there wrt to the bug report.

crass, 2022-12-13:
I'm not setup to test this at the moment (I could test a build with that change, but I'm not setup to build openwrt). But for someone that can easily build OpenWRT, it should be trivial to test x86 build in a VM.

stintel, 2023-01-05:
Should be fixed by 2e61469a6c200c958bde0b564e435647dfc14c75.

closed 2023-01-05 by stintel