openwrt/openwrt #11262: Wrong HOME in hotplug scripts

eric-j-ason, 2022-11-16:
In the environment presented to hotplug scripts, the user is `root`, but the home directory is set to `/` instead of to `/root`. A test script, saved as `/etc/hotplug.d/iface/env`, containing the line `env | logger -p user.debug -t hotplug-env` prints the full environment thus:

```
Wed Nov 16 12:46:45 2022 user.debug hotplug-env: DEVICE=lo
Wed Nov 16 12:46:45 2022 user.debug hotplug-env: USER=root
Wed Nov 16 12:46:45 2022 user.debug hotplug-env: ACTION=ifdown
Wed Nov 16 12:46:45 2022 user.debug hotplug-env: SHLVL=1
Wed Nov 16 12:46:45 2022 user.debug hotplug-env: HOME=/
Wed Nov 16 12:46:45 2022 user.debug hotplug-env: HOTPLUG_TYPE=iface
Wed Nov 16 12:46:45 2022 user.debug hotplug-env: LOGNAME=root
Wed Nov 16 12:46:45 2022 user.debug hotplug-env: DEVICENAME=
Wed Nov 16 12:46:45 2022 user.debug hotplug-env: TERM=linux
Wed Nov 16 12:46:45 2022 user.debug hotplug-env: PATH=/usr/sbin:/usr/bin:/sbin:/bin
Wed Nov 16 12:46:45 2022 user.debug hotplug-env: INTERFACE=wg_mobile
Wed Nov 16 12:46:45 2022 user.debug hotplug-env: PWD=/
```

brada4, 2022-11-19:
Does any script use it?

eric-j-ason, 2022-11-19:
> Does any script use it?

I'm guessing none of the default scripts use it, or the issue would have been detected already.

I discovered it when I used SSH in one script, and it couldn't find its `$HOME/.ssh/known_hosts`.

brada4, 2022-11-19:
You cannot rely on HOME, IFS and TERM or PS? to be set to anything in init scripts e.g. early at boot. I think same propagates at hotplug, just use options with full path

eric-j-ason, 2022-11-19:
In a different issue (#6862) about system init, it was argued that, at that stage, no user is considered logged in, so there is no `USER` and no `HOME`. In hotplug, however, `USER` is set (and why wouldn't it be?), so it would seem natural to have `HOME` go along with it.

brada4, 2022-11-19:
It is historical, you really need to use full paths everywhere. First hotplug events are triggered in harsh conditions early at boot, in other linux even /usr /var /tmp are not guaranteed to be up as an example. Even worse - if you use sysvinit you actually get TERM PATH IFS from your current session, so it is wildly confusing that bootscripts choke failing to make terminal-wide output etc.