openwrt/openwrt #11265: ENS202EXT v1 OpenWRT 22.03.2 Image Bricks Device

stickman89, 2022-11-16:
Device: ENS202EXT version 1
Device page: https://openwrt.org/toh/engenius/ens202ext_1.0.0
------------------------------------

Upgrade file: https://downloads.openwrt.org/releases/22.03.2/targets/ath79/generic/openwrt-22.03.2-ath79-generic-engenius_ens202ext-v1-squashfs-sysupgrade.bin

Performed a sysupgrade from 21.02 to 22.03.02 via OpenWRT Luci, but the device never came back up after the reboot. Continuous (slow) blinking power button. No interface/Luci comes up. Connected UART Serial cable to discover it's stuck on 'No Initrd' and 'Starting Kernel', with no further logs from this point.

[image]

------------------------------------

I had to recover using the stock Engenius Firmware via UART Serial by running the 'run failsafe_boot' command. Now, back on stock firmware; I try to perform the firmware update again from the Engenius admin panel (which is supported), but this time using the file 'https://downloads.openwrt.org/releases/22.03.2/targets/ath79/generic/openwrt-22.03.2-ath79-generic-engenius_ens202ext-v1-squashfs-factory.bin', as is for installations under OEM FW (as the guide states). Same issue at reboot... The Device never comes back online. Only a blinking power btn and the obvious need to use the 'run failsafe_boot' command again. 

Back to stock firmware yet again... At this point I thought to try version: 'https://downloads.openwrt.org/releases/22.03.0/targets/ath79/generic/openwrt-22.03.2-ath79-generic-engenius_ens202ext-v1-squashfs-factory.bin' in the same manner, but this one boots.

Once it successfully booted I sysupgraded to version 22.03.01, and am happy to report this one boots to. The latest version however leads to the above scenario for ENS202EXTv1 though... A (soft?) brick!! Please investigate.

Best Regards,

stickman89

mpratt14, 2022-12-08:
@stickman89 you can follow progress on this issue at #11319 

and yes you're correct that 22.03.1 is not affected, 21.02 versions are also not affected

recovery info is also available in the wiki if anyone needs it

https://openwrt.org/toh/senao/failsafe

mpratt14, 2023-01-05:
@lynxis @hauke with release of 22.03.3 this can be closed

mpratt14, 2023-01-06:
@ynezz also this one, just like #11319

mpratt14, 2023-01-16:
@stickman89 22.03.3 is released and the issue is fixed there

closed 2023-01-16 by ynezz

stickman89, 2023-03-17:
> @stickman89 22.03.3 is released and the issue is fixed there

I've just upgraded to openwrt-22.03.3-ath79-generic-engenius_ens202ext-v1-squashfs-sysupgrade.bin for the ENS202EXT v1, and the device again hasn't come up after the upgrade. Not convinced this is fixed. I've verified I have the correct 22.03.3 stable sysupgrade file.

mpratt14, 2023-03-17:
can you please show the serial log like you did in the first post?

stickman89, 2023-03-17:
I'm really sorry. I was at work when I performed the upgrade remotely. I found out when I came home that the router never came back up due to an internet connection issue. I was wrong to assume it was the same issue. Sorry to misinform you.

stickman89, 2023-03-17:
Ahhh. I see what happened now... The upgrade removed some software packages I rely on to establish an internet connection. No wonder I failed to see it come back up. Thank you for your quick response though. I appreciate it.

mpratt14, 2023-03-17:
glad I could help

you may be interested in this

https://github.com/richb-hanover/OpenWrtScripts#opkgscriptsh

and hopefully in the future we will have a sysupgrade option for automatically doing lazy install of packages from the list of installed before upgrading

mpratt14, 2023-03-17:
seems like we already have it but it's an extra Luci package instead of being part of Luci by default