openwrt/openwrt #11276: current build system problems?

trippleflux, 2022-11-17:
I am getting pretty weird behavior with the OpenWrt buildsystem, currently building OpenWrt Snapshot for Mediatek 7622 or target-aarch64_cortex-a53_glibc.

Normally my steps for fixing OpenWrt packages compilation errors is by : 
```
1. Build stopped
2. Check and fixes the mentioned OpenWrt packages error
3. and re-run compilation with "make -j5 V=s 2>&1 | tee output.log"
```

Above 'loops' is what is working for me all this time, but somehow with the recent snapshot I am getting the following after invoking "make -j5 V=s 2>&1 | tee output.log" : 
- All folders/directories Inside the ./OpenWrt/build_dir/target-aarch64_cortex-a53_glibc getting deleted all the sudden, and there is 2 folder exist "toolchain" and "linux-mediatek"
- Also the ./OpenWrt/build_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_glibc directory is getting deleted

I have tried to do for starting over : 
- make clean
- deleting build_dir, staging_dir, ccache, tmp before invoking "make -j5 V=s 2>&1 | tee output.log"

neheb, 2022-11-20:
sounds like something related to @Ansuel 's AUTOREMOVE work

trippleflux, 2022-11-20:
Could it be because I am not using quilt for patching the OpenWrt packages?

I am only using manually replacing it with my collection of OpenWrt packages 'mod'.

Spudz76, 2022-11-20:
snapshot packages could have moved versions

are you also using feeds?  usually after some time I have to:
```
cp -a .config conf-good
./scripts/feeds uninstall -a
./scripts/feeds update -a
./scripts/feeds install -a
cp -a conf-good .config
make oldconfig
```

Which seems to smack anything strange back to proper.  Not sure about your drop-in method of package replacement especially as they diverge from the current packages (aka yes using drop-in patch files is better then fully replacing sections of the tree whenever possible)

trippleflux, 2022-11-21:
@Spudz76 
Thanks for sharing your workflow, haven't tried try it but will keep it in mind.

The issue that I have reported is not always happening, sometime I manage to rebuild without loosing toolchain and target folder. Will try to use @Spudz76 's workflow now. Perhaps my 'workflow' is not proper.

trippleflux, 2022-11-22:
Disabling CONFIG_AUTOREMOVE seems fixed it, closing.

Ansuel, 2022-11-22:
Can you reopen this? Is this your first time using AUTOREMOVE  or it's a
changed behaviour from the past?

trippleflux, 2022-11-22:
Normally I am building for x86_64 glibc and just checking it in my .config file, the CONFIG_AUTOREMOVE is disabled.

Later on it seem by default when using make defconfig, building for Redmi AX6S it was enabled by default, after disabled CONFIG_AUTOREMOVE for musl, it seems the target and toolchain deletion doesn't happened anymore but weird thing is when compiling for glibc it happened again. Been building along about a month a go OpenWrt image for R7800 but cannot recall this behavior of this self-deletion of toolchain and target folder contents. 

Checked my R7800 .config file, yup it is disabled on it.

Does autoremove using or storing any 'metadata' / 'config' file in ./OpenWrt/tmp ?

Ansuel, 2022-11-22:
AUTOREMOVE should never be set by default on any circumstance unless I guess for buildbot I guess but I will read this in a better way

trippleflux, 2022-11-23:
@Ansuel 
Now i have the following options in my .config : 
```
CONFIG_AUTOREBUILD=y
# CONFIG_AUTOREMOVE is not set
```
But somehow there is left over of CONFIG_AUTOREMOVE when it was being on, because when I am trying complete rebuild with 
```
make clean
rm -rf build_dir
rm -rf staging_dir
rm -rf ccache
```
And doing "make -j5 V=s", somehow after it finished building the musl toolchain, and going into part in which building kernel, it failed because somehow when entering the building kernel AUTOREMOVE still detected as being ON and removing the toolchain folder partly. That's why I am asking if either any AUTOREMOVE settings being saved anywhere else.

Thanks a lot for taking the time and effort for looking into these, really appreciate it.

trippleflux, 2022-11-25:
If you don't mind I can send the log file to you but it's quite big (~80 MB uncompressed), perhaps around 12 MB if compressed.

Ansuel, 2022-12-16:
@trippleflux any news for this? I still need to find time for repro but wonder if you notice other stuff in it.

trippleflux, 2022-12-18:
@Ansuel 
Now working with glibc for x86_64, haven't experienced anymore, when I was with musl for R7800 it's more producible.

If I am not mistaken when I was with musl, it happened after around 4-5 times of force stopped built, and when I run make "make -j5 V=s" suddenly it lost the toolchain folder and target-aarch64_cortex-a53_glibc.

This is really weird problems, I don't know how to reproduce it constantly.

When at glibc, I never experiences it. Or I just don't meet the 4-5 times force stopped built.

neheb, 2022-12-18:
musl...

Doesn'r tools/elfutils fail to compile?

trippleflux, 2022-12-19:
@neheb 
It was around 1 month ago I was tried to compile with musl

Neustradamus, 2024-11-04:
@trippleflux: Have you tested with master code?