openwrt/openwrt #11282: Linksys EA7300 V2 - usb-storage won't work

LinuxfarmerHH, 2022-11-18:
Architecture | MediaTek MT7621 ver:1 eco:3
Target Platform | ramips/mt7621
Firmware Version | OpenWrt 22.03.2 r19803-9a599fee93 / LuCI openwrt-22.03 branch git-22.288.45147-96ec0cd

Followed the usb-storage help and installed the kmod drivers for various usb controllers, so the system shows.

These are the running drivers now
nls_base                5136  1 usbcore
scsi_mod               88944  3 uas,usb_storage,sd_mod
usb_common              2736  5 ehci_platform,ohci_hcd,xhci_plat_hcd,xhci_hcd,usbcore
usb_storage            43472  1 uas
usbcore               147456 14 ehci_pci,ohci_pci,ehci_platform,ehci_fsl,uhci_hcd,ohci_platform,ohci_hcd,ehci_hcd,uas,usb_storage,xhci_plat_hcd,xhci_pci,xhci_mtk,xhci_hcd

View from kernel log looks good
[    4.532204] usbcore: registered new interface driver usbfs
[    4.537806] usbcore: registered new interface driver hub
[    4.543269] usbcore: registered new device driver usb
[    4.558498] xhci-mtk 1e1c0000.xhci: supply vbus not found, using dummy regulator
[    4.566176] xhci-mtk 1e1c0000.xhci: supply vusb33 not found, using dummy regulator
[    4.574021] xhci-mtk 1e1c0000.xhci: xHCI Host Controller
[    4.579354] xhci-mtk 1e1c0000.xhci: new USB bus registered, assigned bus number 1
[    4.599823] xhci-mtk 1e1c0000.xhci: hcc params 0x01401198 hci version 0x96 quirks 0x0000000000290010
[    4.608998] xhci-mtk 1e1c0000.xhci: irq 20, io mem 0x1e1c0000
[    4.616121] hub 1-0:1.0: USB hub found
[    4.620139] hub 1-0:1.0: 2 ports detected
[    4.624863] xhci-mtk 1e1c0000.xhci: xHCI Host Controller
[    4.630295] xhci-mtk 1e1c0000.xhci: new USB bus registered, assigned bus number 2
[    4.637779] xhci-mtk 1e1c0000.xhci: Host supports USB 3.0 SuperSpeed
[    4.644356] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[    4.653439] hub 2-0:1.0: USB hub found
[    4.657316] hub 2-0:1.0: 1 port detected

Pluggin various usb keys results to nothing new in the logfiles

lsusb-t says this

/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/1p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/2p, 480M

So there is an usb hub but nothing else to see. Have tested the usb port in linksys firmware earlier and there it was functional.

Ideas are welcome

brada4, 2022-11-19:
It should look like:
```
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-platform/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=, Driver=usb-storage, 480M
```
Seems the plugging of device does not get detected. It should.

Spudz76, 2022-11-19:
Not really related but I'm successfully using an old USB1.x webcam on my EA7300v1 with the same release, so the hub and usb basics work okay (I have never tried storage devices).

```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/1p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=, Driver=gspca_zc3xx, 12M
```

Spudz76, 2022-11-19:
I swapped to a USB key in and it shows up (I don't have any storage kmods installed though, so of course hotplug didn't load them).

```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/1p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/2p, 480M
    |__ Port 1: Dev 3, If 0, Class=, Driver=, 480M
...
Bus 001 Device 003: ID 05dc:c75c Lexar USB Flash Drive
```

Spudz76, 2022-11-19:
Left the key installed, did `opkg install kmod-usb-storage`, and it just works?

```
kmodloader: loading kernel modules from /etc/modules.d/*
SCSI subsystem initialized
usb-storage 1-1:1.0: USB Mass Storage device detected
scsi host0: usb-storage 1-1:1.0
usbcore: registered new interface driver usb-storage
kmodloader: done loading kernel modules from /etc/modules.d/*
kmodloader: loading kernel modules from /etc/modules.d/*
kmodloader: done loading kernel modules from /etc/modules.d/*
scsi 0:0:0:0: Direct-Access     Lexar    USB Flash Drive  8.07 PQ: 0 ANSI: 4
sd 0:0:0:0: [sda] 30662656 512-byte logical blocks: (15.7 GB/14.6 GiB)
sd 0:0:0:0: [sda] Write Protect is off
sd 0:0:0:0: [sda] Mode Sense: 23 00 00 00
sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
 sda: sda1 sda2 sda3
sd 0:0:0:0: [sda] Attached SCSI removable disk
```

brada4, 2022-11-19:
Looks like USB port does not detect device connected. I would suggest to plug mouse or something else as cheap, or having multimeter check voltage on pins inside USB port.

LinuxfarmerHH, 2022-11-20:
Okay my first test was to plug an usb mouse to it, no power, no light for the mouse. Pluggin a usb-lan 2.0 does also not come to energy on the usb, same with a illuminated keyboard. The hardware should have some kind of usb power management, where the power is disabled.

root@Medusa:~# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/1p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/2p, 480M

Does this look ok?

root@Medusa:~# cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 2
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 5.10
S:  Manufacturer=Linux 5.10.146 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=1e1c0000.xhci
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=02 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=5000 MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 3.00 Cls=09(hub  ) Sub=00 Prot=03 MxPS= 9 #Cfgs=  1
P:  Vendor=1d6b ProdID=0003 Rev= 5.10
S:  Manufacturer=Linux 5.10.146 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=1e1c0000.xhci
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

Spudz76, 2022-11-20:
Yes that also looks same as mine.

I did not previously confirm that your dmesg stuff also looks identical to mine (dummy 
supply regulator and no LPM, etc):
```
# dmesg | grep -iE '(hci|hub|usb)'
[    5.655971] usbcore: registered new interface driver usbfs
[    5.661613] usbcore: registered new interface driver hub
[    5.667040] usbcore: registered new device driver usb
[    5.683522] xhci-mtk 1e1c0000.xhci: supply vbus not found, using dummy regulator
[    5.691242] xhci-mtk 1e1c0000.xhci: supply vusb33 not found, using dummy regulator
[    5.699072] xhci-mtk 1e1c0000.xhci: xHCI Host Controller
[    5.704409] xhci-mtk 1e1c0000.xhci: new USB bus registered, assigned bus number 1
[    5.718911] xhci-mtk 1e1c0000.xhci: hcc params 0x01401198 hci version 0x96 quirks 0x0000000000290010
[    5.728090] xhci-mtk 1e1c0000.xhci: irq 20, io mem 0x1e1c0000
[    5.735127] hub 1-0:1.0: USB hub found
[    5.739052] hub 1-0:1.0: 2 ports detected
[    5.743704] xhci-mtk 1e1c0000.xhci: xHCI Host Controller
[    5.749093] xhci-mtk 1e1c0000.xhci: new USB bus registered, assigned bus number 2
[    5.756577] xhci-mtk 1e1c0000.xhci: Host supports USB 3.0 SuperSpeed
[    5.763140] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[    5.772197] hub 2-0:1.0: USB hub found
[    5.776085] hub 2-0:1.0: 1 port detected
[    6.278860] usb 1-1: new full-speed USB device number 2 using xhci-mtk
[   14.069512] input: gspca_zc3xx as /devices/platform/1e1c0000.xhci/usb1/1-1/input/input0
[   14.078062] usbcore: registered new interface driver gspca_zc3xx
```

Except of course that my port works and finds that webcam pretty quickly after boot.  I am also now on snapshot but I don't think that has much to do with your physically broken hardware.  Does it still work in factory firmware?  Could be ESD killed it since then...

LinuxfarmerHH, 2022-12-12:
This took some time, found a hint, that shutting down the WAN interface in Luci do start the usb port. Here is the result with my device. So the port isn't damaged, there is an ugly flaw in openwrt. 

root@Medusa:~# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/1p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=, Driver=usb-storage, 480M

root@Medusa:~# dmesg | grep usb
[    4.504517] usbcore: registered new interface driver usbfs
[    4.510220] usbcore: registered new interface driver hub
[    4.515664] usbcore: registered new device driver usb
[    4.538828] xhci-mtk 1e1c0000.xhci: supply vusb33 not found, using dummy regulator
[    4.613536] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[   11.400920] usbcore: registered new interface driver usb-storage
[   11.412330] usbcore: registered new interface driver uas
root@Medusa:~# dmesg | grep usb
[    4.504517] usbcore: registered new interface driver usbfs
[    4.510220] usbcore: registered new interface driver hub
[    4.515664] usbcore: registered new device driver usb
[    4.538828] xhci-mtk 1e1c0000.xhci: supply vusb33 not found, using dummy regulator
[    4.613536] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[   11.400920] usbcore: registered new interface driver usb-storage
[   11.412330] usbcore: registered new interface driver uas
[1905047.746589] usb 1-1: new high-speed USB device number 2 using xhci-mtk
[1905047.936902] usb-storage 1-1:1.0: USB Mass Storage device detected

Re-enabling the WAN and WAN6 interface cause

root@Medusa:~# lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/1p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci-mtk/2p, 480M

The usb is shutted down again.

Spudz76, 2022-12-12:
Yeah, very unsure how that could be related via software.  I still do not see anything like that on the v1 hardware, USB port just always works.  The DTS are essentially the same between the revisions maybe there is something strange in v2 that needs something different.

Did it do the same thing prior to #10238 ?  That's the only change that has something to do with wan port logistics.  I guess only parts of that PR were merged.  Commit c77858aa79265a3fb57800bf147e2c6efacdb7f7 has something, the rest seems in f1c9afd801380a05a91d979b475c76cc0a67caae or so, maybe a build from just before those patches acts normal?

brada4, 2022-12-12:
Depends what you need - 2x1Gbit CPU port or surplus power to USB port.
Best exit is powered USB hub, the router itself draws 12V and there is inverter that passes 5V to USB and 3.3V and 1V to SoC. You cannot re-wire around that.

LinuxfarmerHH, 2022-12-12:
Have tested USB1-1 and USB2 keys, but also ordered some USB3 keys now for testing. Maybe they use other voltage that workaround that. Most of the keys become warm if pluggd somewhere, so their power consumptions seems not to be nice.
If someone found a running key, please offer which it is.

drt1245, 2023-11-17:
On [the forum](https://forum.openwrt.org/t/usb-is-disconnected-when-wan-is-up/117886/1), someone pointed out that disabling WAN caused the USB to start working properly.

From this, I was able to figure out that the WAN LED pin assignment is wrong.
Running `echo 0 > /sys/class/leds/green:wan/brightness` causes the USB port to start working, and does not affect the WAN LED.

BaaGreene, 2024-01-28:
I'm in over my head here, but I will try this, and hopefully be helpful.

I wrong gpio pins are consistent with what I see, exept that I never tested wps.

I looked at the linked commit and see that lan3 and lan4 are both GPIO_ACTIVE_LOW. However, on my EA3700 V2, I get the following results:
`echo 0 > /sys/class/leds/green:lan3/brightness` turns LED3 ON.
`echo 1 > /sys/class/leds/green:lan3/brightness` turns LED3 OFF.
`echo 0 > /sys/class/leds/green:lan4/brightness` turns LED4 OFF.
`echo 0 > /sys/class/leds/green:lan4/brightness` turns LED4 ON.

This would lead me to believe that LED3 is GPIO_ACTIVE_LOW and LED4 is GPIO_ACTIVE_HIGH.

If I should post this elsewhere, please let me know.

brada4, 2024-01-28:
You can tag that commit author wrt to your findings.

BaaGreene, 2024-01-29:
Done. Thanks,