openwrt/openwrt #11291: ntp-server leave unsynced when equipped w/ RTC DS1307

EKNYESE, 2022-11-19:
With actual openwrt 22.03.0 and 22.03.2 and equipped with an RTC module DS1307 as described here
[http://kernelreloaded.com/adding-rtc-to-openwrt-devices-part-2/](http://kernelreloaded.com/adding-rtc-to-openwrt-devices-part-2/)
it is not possible to receive a proper ntp:

Result from my connected raspberry:
    openhabian@openHABianRPi4:~ $ ntpdate -q 192.168.9.1
    19 Nov 19:18:35 ntpdate[2454]: no server suitable for synchronization found
    openhabian@openHABianRPi4:~ $ ntpdate -q GL-MT300N-V2
    19 Nov 19:18:57 ntpdate[2610]: no server suitable for synchronization found

checking ping:
    openhabian@openHABianRPi4:~ $ ping GL-MT300N-V2
    PING GL-MT300N-V2.lan (192.168.9.1) 56(84) bytes of data.
    64 bytes from GL-MT300N-V2.lan (192.168.9.1): icmp_seq=1 ttl=64 time=1.59 ms
    64 bytes from GL-MT300N-V2.lan (192.168.9.1): icmp_seq=2 ttl=64 time=1.58 ms
    64 bytes from GL-MT300N-V2.lan (192.168.9.1): icmp_seq=3 ttl=64 time=1.53 ms
    ^C
    --- GL-MT300N-V2.lan ping statistics ---
    3 packets transmitted, 3 received, 0% packet loss, time 5ms
    rtt min/avg/max/mdev = 1.533/1.569/1.592/0.052 ms

my Tasmota devices also receive no time-signal (console):
    00:00:37.450 CMD: ntpserver
    00:00:37.453 SRC: WebConsole from 192.168.9.224
    00:00:37.456 CMD: Grp 0, Cmd 'NTPSERVER', Idx 1, Len 0, Pld -99, Data ''
    00:00:37.459 RSL: RESULT = {"NtpServer1":"192.168.9.1","NtpServer2":"GL-MT300N-V2","NtpServer3":"192.168.9.1:123"}
    00:00:54.249 WIF: Checking connection...
    00:01:00.382 WIF: Sending Gratuitous ARP
    00:01:14.260 WIF: Checking connection...
    00:01:34.522 WIF: Checking connection...
    00:01:36.384 NTP: Sync time...
    00:01:37.409 NTP: No reply from 0.255.255.255
    00:01:55.429 WIF: Checking connection...
    00:02:00.391 WIF: Sending Gratuitous ARP
    00:02:15.447 WIF: Checking connection...
    00:02:35.424 WIF: Checking connection...
    00:02:36.403 NTP: Sync time...
    00:02:36.421 NTP: IP 192.168.9.1 unsynced

Windows seems to report no problems (c:/Windows/System32/w32tm.exe /stripchart /computer:192.168.9.1):
    192.168.9.1 wird verfolgt [192.168.9.1:123].
    Es ist 19.11.2022 20:57:59.
    20:57:59, d:+00.0026014s o:+04.6700534s  [                           |            *              ]
    20:58:01, d:+00.0030599s o:+04.6699316s  [                           |            *              ]
    20:58:03, d:+00.0061549s o:+04.6705323s  [                           |            *              ]

Port 123 for TCP and UDP is open.
Problem is only detected without internet-connection.
With running internet-connection (--> openwrt received ntp-signal as client), no "problem" detected.
The solution for this problem is urgent, since in my point of view there is NO possibility to serve ntp without internet/ on an island after blackout.

brada4, 2022-11-20:
You need to explicitly allow serving from "local" time source (other source types being (USB) GPS etc), busybox ntpd does not seem to do it.

Options are chrony and isc ntpd. Former is somewhat more safe as it allows to select on which IPs to run listeners on and which subnets to serve.

Serving local time consider temperature-dependency and battery-drain-dependency of chosen RTC.

EKNYESE, 2022-11-26:
Thank you for your response!
So I have de-installed busybox-ntpd and installed ntpd via opkg.
Unfortunately this "openwrt"-ntpd-package is not compatible to "original" ntpd. E.g. the standard config-file is not present any more.
The sample-config [https://github.com/openwrt/packages/blob/master/net/ntpd/README.md](https://github.com/openwrt/packages/blob/master/net/ntpd/README.md) is  n o t  working for my upper mentioned use-case.

--> **Does anybody have a hint how I can configure ntpd on openwrt to provide a local ntp-server** (if no internet connection is available)?

brada4, 2022-11-26:
https://github.com/openwrt/packages/blob/b062c3a001e245d7f70eda324807a046e5798c1e/net/ntpd/files/ntpd.init#L78

You need to create whatever.conf in that directory with
```
server 127.127.1.1 iburst
fudge 127.127.1.1 stratum 15 refid RTC
```
127.127 selects local timekeeping device - type1 aka RTC , instance 1 
Second sets lowest usable distance from reference time in place of highest - then while router boots on local clock - subordinate servers will not distribute time, and makes reference clock name more palatable.

EKNYESE, 2022-11-27:
Thank You much for your response!

So I included your upper mentioned lines into new file
`        /etc/ntpd.d/ntpd_server_config.conf`

After starting ntpd via
`        root@GL-MT300N-V2:~# ntpd`

I recognized the new entry
`        includefile /etc/ntpd.d/ntpd_server_config.conf`
in the (automatic generated) file `/var/etc/ntpd.conf`.

Unfortunatly these steps do not solve my problem - see output (taken >>30min. after starting ntpd):
> 	root@GL-MT300N-V2:~# ntpdate -q gl-mt300n-v2
> 	server 192.168.10.1, stratum **16**, offset -0.000015, delay 0.02609
> 	server 192.168.9.1, stratum **16**, offset -0.000035, delay 0.02605
> 	server fd70:6923:3fcf::1, stratum **16**, offset +0.000008, delay 0.02608
> 	server fd70:6923:3fcf:10::1, stratum **16**, offset +0.000005, delay 0.02606
> 	27 Nov 10:39:42 ntpdate[4729]: no server suitable for synchronization found

Do you have some further ideas? Is the address "127.127.1.**1**" correct ([https://www.linksysinfo.org/index.php?threads/how-to-set-up-ntp-timeserver-on-tomato-for-your-small-office.68515](https://www.linksysinfo.org/index.php?threads/how-to-set-up-ntp-timeserver-on-tomato-for-your-small-office.68515) highlighted "NOTE: **127.127.1.0** is a special IP address that ntpd interprets as a localhost- as- a- time-source")?

In the past days I have also tested some hints like
```
        server 127.127.1.0
        fudge 127.127.1.0 stratum 10
```
without any luck..

**THANK YOU FOR YOUR RESPONSE!!**

brada4, 2022-11-27:
Maybe server  ...  iburst prefer, then try ntpq  locally

EKNYESE, 2022-11-27:
I am sorry, what do You mean exactly?
I executed ntpq with following commands/results:

**ntpq> peer**
`No association ID's returned`

**ntpq> hostnames**
`hostnames being shown`

**ntpq> peers**
`No association ID's returned`

**ntpq> sysinfo**
```associd=0 status=c016 leap_alarm, sync_unspec, 1 event, restart,
system peer:        0.0.0.0:0
system peer mode:   unspec
leap indicator:     11
stratum:            16
log2 precision:     -17
root delay:         0.000
root dispersion:    0.000
reference ID:       .
reference time:     (no time)
system jitter:      0.000000
clock jitter:       0.008
clock wander:       0.000
broadcast delay:    -50.000
symm. auth. delay:  0.000
```

**ntpq> authinfo**
```time since reset:  9668
stored keys:       0
free keys:         0
key lookups:       0
keys not found:    0
uncached keys:     0
expired keys:      0
encryptions:       0
decryptions:       0
```

**ntpq> sysstats**
```uptime:                 9708
sysstats reset:         9708
packets received:       988
current version:        16
older version:          0
bad length or format:   0
authentication failed:  0
declined:               0
restricted:             0
rate limited:           0
KoD responses:          0
processed for time:     0
```

Do You have some further advise? I can not see any hints for the issue.

brada4, 2022-11-27:
Set stratum <=14, apparently ntpdate already increments it and does not take time, sorry for advicing wrong.
Correct config implant:
```
server 127.127.1.1 iburst
fudge 127.127.1.1 stratum 10 refid RTC

```

.1.0 -> /dev/rtc (this is symlink to next typically)
.1.1 -> /dev/rtc0
.1.2 -> /dev/rtc1

Changed `pool` to `server` in centos...
```
        refid      st t when poll reach   delay   offset  jitter
==============================================================================
 46.243.26.34    .GPS.            1 u    6   64    1   38.798    4.116   2.613
*127.127.1.1     .RTC.           10 l   15   64    1    0.000    0.000   0.001
```
And second and minute after start of it:
```
server 127.0.0.1, stratum 11, offset 0.000000, delay 0.02570
server 127.0.0.1, stratum 3, offset -0.000005, delay 0.02570
```

brada4, 2022-11-27:
Not a package maintainer, but it would be nice to have checkbox for your well described reasonable use case,

brada4, 2022-11-27:
chrony equivalent:
```
local stratum 10
```

EKNYESE, 2022-11-27:
Unfortunately, lowering the stratum to "10" did not have an effect:

Furthermore I recognized a strange behaviour: my upper statement

> Problem is only detected without internet-connection.
> With running internet-connection (--> openwrt received ntp-signal as client), no "problem" detected.

is not valid anymore:

**Behaviour when router started w/o internet:**

```
root@GL-MT300N-V2:~# ntpdate -q GL-MT300N-V2
server 192.168.10.1, stratum 16, offset -0.000012, delay 0.02608
server 192.168.9.1, stratum 16, offset -0.000067, delay 0.02611
server fd70:6923:3fcf::1, stratum 16, offset +0.000007, delay 0.02608
server fd70:6923:3fcf:10::1, stratum 16, offset +0.000003, delay 0.02606
27 Nov 18:22:35 ntpdate[3976]: no server suitable for synchronization found
```

**Behaviour when router started w/o internet and with re-established internet-connection (after boot):**

```
root@GL-MT300N-V2:~# ntpdate -q GL-MT300N-V2
server 192.168.10.1, stratum 16, offset -0.000010, delay 0.02608
server 192.168.9.1, stratum 16, offset -0.000034, delay 0.02603
server fd70:6923:3fcf::1, stratum 16, offset +0.000019, delay 0.02611
server fd70:6923:3fcf:10::1, stratum 16, offset +0.000001, delay 0.02606
27 Nov 18:25:45 ntpdate[4198]: no server suitable for synchronization found
```

**Behaviour when router started w/ internet:**

```
root@GL-MT300N-V2:~# ntpd
root@GL-MT300N-V2:~# ntpdate -q GL-MT300N-V2
27 Nov 18:39:51 ntpdate[3866]: no server suitable for synchronization found
```

It is very strange.
Hopefully You have further ideas - or I have to make a "factory reset" and give your alternative "chrony" a chance..

brada4, 2022-11-27:
stratum 15 makes the mess. You need to lower the value. Formally it is valid but ntpdate does artificial increment to KOD.

brada4, 2022-11-27:
i.e stratum 0 are the ultimate source devices like gps receivers, atomic clocks etc. Stratum 1 is connected to them, stratum 2 to those stratum 1 etc etc ...

EKNYESE, 2022-11-28:
Thanks for Your response.
The problem is, that - in the standard openwrt-package - ntpd seems not to be able to accept the .conf-file.

I have set the stratum to "0" - without any luck/ difference in system behaviour:

**ntpd
ntpdate -q GL-MT300N-V2**
```
server 192.168.9.1, stratum 16, offset -0.000047, delay 0.02615
server 192.168.10.1, stratum 16, offset -0.000038, delay 0.02605
server fd70:6923:3fcf::1, stratum 16, offset +0.000020, delay 0.02608
server fd70:6923:3fcf:10::1, stratum 16, offset +0.000012, delay 0.02608
28 Nov 06:12:39 ntpdate[3915]: no server suitable for synchronization found

```

It feels to me, that up to now I am the only person, who have testet/ validated the ntpd-openwrt-package with rtc only (w/o GPS/internet!).

Any further ideas, how to dive into the problem/analysis?
Your post shows, that you have GPS AND RTC connected to your openwrt. Is it possible for you to cross-check behaviour in your system w/o GPS?

brada4, 2022-11-28:
Please take anything else. zero is KoD - i.e. hint to client to buzz off. By bug or feature 15 lands in same place. (the stratum field is four bytes only in NTP packet)

EKNYESE, 2022-11-28:
Unfortunately changing the stratum to 0, 1, 4, 10 or 15 does not show any effect, it seems to be a bug/feature.
Is anybody able to do a cross-check with his own system? Or are there some further ideas how to investigate this problem?

**ntpd**
**ntpdate -q GL-MT300N-V2**
```
	server 192.168.9.1, stratum 16, offset -0.000047, delay 0.02615
	server 192.168.10.1, stratum 16, offset -0.000010, delay 0.02600
	server fd70:6923:3fcf::1, stratum 16, offset +0.000017, delay 0.02611
	server fd70:6923:3fcf:10::1, stratum 16, offset +0.000007, delay 0.02608
	28 Nov 08:09:33 ntpdate[3917]: no server suitable for synchronization found
```

**/etc/config/system**:

```
	config timeserver 'ntp'
		option enabled '1'
		option enable_server '1'
		list server '0.openwrt.pool.ntp.org'
		list server '1.openwrt.pool.ntp.org'
		list server '2.openwrt.pool.ntp.org'
		list server '3.openwrt.pool.ntp.org'
```

**/var/etc/ntpd.conf**:

```
	driftfile /var/lib/ntp/ntp.drift
	
	restrict default limited kod nomodify notrap nopeer
	restrict -6 default limited kod nomodify notrap nopeer
	restrict source noquery
	
	# No limits for local monitoring
	restrict 127.0.0.1
	restrict -6 ::1
	
	server 0.openwrt.pool.ntp.org iburst
	server 1.openwrt.pool.ntp.org iburst
	server 2.openwrt.pool.ntp.org iburst
	server 3.openwrt.pool.ntp.org iburst
	includefile /etc/ntpd.d/ntpd_server_config.conf
```

**/etc/ntpd.d/ntpd_server_config.conf**:

```
	server 127.127.1.1 iburst
	fudge 127.127.1.1 stratum 4 refid RTC
```

brada4, 2022-11-28:
You need to regenerate ntp.conf and restart ntpd (`/etc/init.d/ntpd restart`)
Configuration clearly says local clock should be served with stratum 4 within one second after start of process.

EKNYESE, 2022-11-28:
### Thank You very much, You solved this issue nearly.

I generated manually a file "/etc/ntp.conf" with following content and restarted ntpd as you recommended:

```
	driftfile /var/lib/ntp/ntp.drift
	
	restrict default limited kod nomodify notrap nopeer
	restrict -6 default limited kod nomodify notrap nopeer
	restrict source noquery
	
	# No limits for local monitoring
	restrict 127.0.0.1
	restrict -6 ::1
	
	server 0.openwrt.pool.ntp.org iburst
	server 1.openwrt.pool.ntp.org iburst
	server 2.openwrt.pool.ntp.org iburst
	server 3.openwrt.pool.ntp.org iburst
	
	server 127.127.1.1 iburst
	fudge 127.127.1.1 stratum 10 refid RTC
```

**boot w/o internet, w/o internet (wan connection) after boot: (--> everything is fine 👍)**

```
root@GL-MT300N-V2:~# ntpdate -q GL-MT300N-V2
	server 192.168.10.1, stratum 11, offset -0.000017, delay 0.02609
	server 192.168.9.1, stratum 11, offset -0.000041, delay 0.02605
	server fd70:6923:3fcf::1, stratum 11, offset +0.000008, delay 0.02608
	server fd70:6923:3fcf:10::1, stratum 11, offset +0.000004, delay 0.02606
	28 Nov 19:27:29 ntpdate[4269]: adjust time server 192.168.9.1 offset -0.000041 sec

root@GL-MT300N-V2:~# ntpq
ntpq>  sysinfo
	associd=0 status=0514 leap_none, sync_local, 1 event, freq_mode,
	system peer:        LOCAL(1)
	system peer mode:   client
	leap indicator:     00
	stratum:            11
	log2 precision:     -17
	root delay:         0.000
	root dispersion:    11.354
	reference ID:       127.127.1.1
	reference time:     e72f8a17.d60ae064  Mon, Nov 28 2022 19:27:19.836
	system jitter:      0.000000
	clock jitter:       0.008
	clock wander:       0.000
	broadcast delay:    -50.000
	symm. auth. delay:  0.000

```

**boot w/o internet, w/ internet (wan connection) after boot: (--> everything is fine 👍)**

```
root@GL-MT300N-V2:~# ntpdate -q GL-MT300N-V2
	server 192.168.10.1, stratum 3, offset -0.000012, delay 0.02609
	server 192.168.9.1, stratum 3, offset -0.000036, delay 0.02605
	server fd70:6923:3fcf::1, stratum 3, offset +0.000010, delay 0.02606
	server fd70:6923:3fcf:10::1, stratum 3, offset +0.000005, delay 0.02606
	28 Nov 19:32:31 ntpdate[4271]: adjust time server 192.168.9.1 offset -0.000036 sec

root@GL-MT300N-V2:~# ntpq
ntpq> sysinfo
	associd=0 status=0618 leap_none, sync_ntp, 1 event, no_sys_peer,
	system peer:        138.2.181.18:123
	system peer mode:   client
	leap indicator:     00
	stratum:            3
	log2 precision:     -17
	root delay:         47.736
	root dispersion:    146.044
	reference ID:       138.2.181.18
	reference time:     e72f8b47.8f8766b7  Mon, Nov 28 2022 19:32:23.560
	system jitter:      3.304191
	clock jitter:       8.091
	clock wander:       0.000
	broadcast delay:    -50.000
	symm. auth. delay:  0.000

```

### **Unfortunately with use of ntpd there is a new strange behaviour:**
**boot w/ internet, w/ internet (wan connection) after boot: (--> no running(?) ntp-server 👎)**

```
BusyBox v1.35.0 (2022-10-14 22:44:41 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 22.03.2, r19803-9a599fee93
 -----------------------------------------------------
root@GL-MT300N-V2:~# ntpd
root@GL-MT300N-V2:~# ntpdate -q GL-MT300N-V2
28 Nov 19:37:01 ntpdate[4047]: no server suitable for synchronization found
root@GL-MT300N-V2:~# ntpq
ntpq> sysinfo
ntpq: read: Connection refused
```
I waited several hours, but ntp-server seems **not** to start.

In my point of view you discovered at least one bug in openwrt, since there seems to be needed a manually generated file "/etc/ntp.conf" **and** the entries in /etc/ntpd.d/whatever.conf are not interpreted.

On the other hand there is now a big unsolved problem in case of
**boot w/ internet, w/ internet (wan connection) after boot:**

For my usecase (smarthome w/ normally well-functional internet-connection; but several blackouts with non-functional internet-connection afterwards for several days) it is still no sufficient solution.

### Could You please try to give me some further hints, how to fix the remaining ntp-server problem (w/ connected internet)?

brada4, 2022-11-28:
Some steps:

0) optional - rise your self-confidence, install OpenWRT x86 in a virtual machine, rtc works right away.

1) You can leave out all ntp servers so that only RTC is a time source.
`ntpq -n -c peer` will show time sources right after.

2) `/etc/init.d/ntpd stop ; /etc/init.d/ntpd start`
-> does it start?
Maybe kmod loading at boot comes too late so that RTC device is not yet up when ntpd starts?
You need to use scripts because they generate files from uci databases, create temporary users and directories etc

3) Make sure sysntpd, ntpdate or chrony are not enabled (just check web interface) - those will open /dev/rtc and interfere with ntpd.

4) If device file is not there.
`cd /dev/ ; mknod rtc0 c 252 0`

EKNYESE, 2022-11-29:
### Wonderful, now it works like a charme! 💯 

**In my case Step 3 (disabling of ntpdate) was the clue.**

### Once again: thank You very much! 🥇

closed 2022-11-29 by EKNYESE