openwrt/openwrt #11295: BPI-R3 - SFP1 (eth1): PPPoE - Timeout waiting for PADO packets using VLAN

hvandrie, 2022-11-20:
Snapshot 21301-21326

Can't set up PPPoE in VLAN 6.

```
Sun Nov 20 04:29:44 2022 daemon.debug pppd[9658]: Send PPPOE Discovery V1T1 PADI session 0x0 length 18
Sun Nov 20 04:29:44 2022 daemon.debug pppd[9658]:  dst ff:ff:ff:ff:ff:ff  src 72:47:be:ec:38:39
Sun Nov 20 04:29:44 2022 daemon.debug pppd[9658]:  [service-name] [host-uniq  ba 25 00 00] [PPP-max-payload  05 dc]
Sun Nov 20 04:29:49 2022 daemon.warn pppd[9658]: Timeout waiting for PADO packets
Sun Nov 20 04:29:49 2022 daemon.err pppd[9658]: Unable to complete PPPoE Discovery
Sun Nov 20 04:29:49 2022 daemon.info pppd[9658]: Exit.

```

TcpDump show request done on VLAN 6, but no replies.
```
04:29:08.845527 72:47:be:ec:38:39 (oui Unknown) > Broadcast, ethertype 802.1Q (0x8100), length 42: vlan 6, p 0, ethertype PPPoE D (0x8863), PPPoE PADI [Service-Name] [Host-Uniq 0xDE230000] [PPP-Max-Payload 0x05DC]
04:29:13.850665 72:47:be:ec:38:39 (oui Unknown) > Broadcast, ethertype 802.1Q (0x8100), length 42: vlan 6, p 0, ethertype PPPoE D (0x8863), PPPoE PADI [Service-Name] [Host-Uniq 0xDE230000] [PPP-Max-Payload 0x05DC]

```

Ethtool says 'link detected'
```
Settings for eth1:
        Supported ports: [ FIBRE ]
        Supported link modes:   1000baseX/Full
        Supported pause frame use: Symmetric Receive-only
        Supports auto-negotiation: Yes
        Supported FEC modes: Not reported
        Advertised link modes:  1000baseX/Full
        Advertised pause frame use: Symmetric Receive-only
        Advertised auto-negotiation: No
        Advertised FEC modes: Not reported
        Speed: 1000Mb/s
        Duplex: Full
        Port: FIBRE
        PHYAD: 0
        Transceiver: internal
        Auto-negotiation: off
        Current message level: 0x000000ff (255)
                               drv probe link timer ifdown ifup rx_err tx_err
        Link detected: yes
```

Same config for setting up on SFP as on ER-X-SFP running OpenWrt 21.02.3, r16554-1d4dea6d4f works w/o a problem.

hvandrie, 2022-11-24:
Most likely related to "VLAN offloading was broken in mtk_eth_soc driver".
https://forum.banana-pi.org/t/install-openwrt-snapshot-on-bpi-r3/14231/32

hvandrie, 2022-12-05:
Fixed since Nov 19th.
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=9b482ee22f3f509880d2420652392971eef3211d

Yet, still have issues with spontaneous PPPoE drops which don't recover until (cold) reboot.

Yudi, 2023-01-04:
> Yet, still have issues with spontaneous PPPoE drops which don't recover until (cold) reboot.

Might be related to #7826