openwrt/openwrt #11299: nat offloading breaks voip connection on wifi roaming.

odkrys, 2022-11-20:
Hello.

I am using openwrt snapshot version on iptime T5004 (mt7621).

The problem I noticed today is that when using sotftware or hardware nat offloading, if wifi roaming occurs, the voip connection drops.
It reconnects on skype, it disconnects on other voip apps I've tested.

I have currently turned offload off as a workaround for this problem, and report the problem.

schuettecarsten, 2022-11-21:
Maybe, this is related to #11277 and #11029

LGA1150, 2022-11-22:
> Maybe, this is related to #11277 and #11029 

While they look similar, they are not related at all.

The two issues you referenced are triggered by clients' roaming from one of the switch ports to a non-switch port such as WiFi. I've posted fixes to upstream netdev and later backported here, but the fixes got reverted because of a throughtput regression. I didn't touch any QoS configurations of the switch so it was probably a hardware bug.

OP's issue is different. The router does not have any non-switch ports so the ARL entries are updated as intended. It's the HNAT entries that are not updated when a client roams, so turning it off will avoid the issue.

schuettecarsten, 2022-11-22:
Ok, thank you for the explanation.

odkrys, 2024-08-10:
I tested it on 23.05.4 and confirmed it is resolved.

closed 2024-08-10 by odkrys