openwrt/openwrt #11301: Build image failed for Redmi AX6S : Bad file descriptor

trippleflux, 2022-11-20:
No matter what I've tried it seems I cannot make images for Redmi AX6S, failure at "**_bash: line 1: 8: Bad file descriptor_**", not sure which one is that line referring the Makefile into, full log : 
```
FIT description: ARM64 OpenWrt FIT (Flattened Image Tree)
Created:         Fri Nov 18 19:27:52 2022
 Image 0 (kernel-1)
  Description:  ARM64 OpenWrt Linux-5.15.79
  Created:      Fri Nov 18 19:27:52 2022
  Type:         Kernel Image
  Compression:  lzma compressed
  Data Size:    4390150 Bytes = 4287.26 KiB = 4.19 MiB
  Architecture: AArch64
  OS:           Linux
  Load Address: 0x44000000
  Entry Point:  0x44000000
  Hash algo:    crc32
  Hash value:   00c50b02
  Hash algo:    sha1
  Hash value:   0dd51ce3eb2ad276607f37537f8597ba23bc0cd0
 Image 1 (initrd-1)
  Description:  ARM64 OpenWrt xiaomi_redmi-router-ax6s initrd
  Created:      Fri Nov 18 19:27:52 2022
  Type:         RAMDisk Image
  Compression:  uncompressed
  Data Size:    19936416 Bytes = 19469.16 KiB = 19.01 MiB
  Architecture: AArch64
  OS:           Linux
  Load Address: unavailable
  Entry Point:  unavailable
  Hash algo:    crc32
  Hash value:   0dd9dd59
  Hash algo:    sha1
  Hash value:   26fa1a5df582abede688c3c722314fcfbddbfad0
 Image 2 (fdt-1)
  Description:  ARM64 OpenWrt xiaomi_redmi-router-ax6s device tree blob
  Created:      Fri Nov 18 19:27:52 2022
  Type:         Flat Device Tree
  Compression:  uncompressed
  Data Size:    29038 Bytes = 28.36 KiB = 0.03 MiB
  Architecture: AArch64
  Hash algo:    crc32
  Hash value:   2aab0edd
  Hash algo:    sha1
  Hash value:   9cd29a9030c5a37618b891e87eb4aabde134720e
 Default Configuration: 'config-1'
 Configuration 0 (config-1)
  Description:  OpenWrt xiaomi_redmi-router-ax6s
  Kernel:       kernel-1
  Init Ramdisk: initrd-1
  FDT:          fdt-1
cp /home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/tmp/custom-master-18-11-2022-r21326-bca663f20f-mediatek-mt7622-xiaomi_redmi-router-ax6s-initramfs-recovery.itb /home/username/works/AX6S/bin/targets/mediatek/mt7622/custom-master-18-11-2022-r21326-bca663f20f-mediatek-mt7622-xiaomi_redmi-router-ax6s-initramfs-recovery.itb
DEVICE_ID="xiaomi_redmi-router-ax6s" SOURCE_DATE_EPOCH=1668799672 FILE_NAME="custom-master-18-11-2022-r21326-bca663f20f-mediatek-mt7622-xiaomi_redmi-router-ax6s-initramfs-recovery.itb" FILE_DIR="/home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/tmp" FILE_TYPE="kernel" FILE_FILESYSTEM="initramfs" DEVICE_IMG_PREFIX="custom-master-18-11-2022-r21326-bca663f20f-mediatek-mt7622-xiaomi_redmi-router-ax6s" DEVICE_VENDOR="Xiaomi" DEVICE_MODEL="Redmi Router AX6S" DEVICE_VARIANT="" DEVICE_ALT0_VENDOR="Xiaomi" DEVICE_ALT0_MODEL="Router AX3200" DEVICE_ALT0_VARIANT="" DEVICE_ALT1_VENDOR="" DEVICE_ALT1_MODEL="" DEVICE_ALT1_VARIANT="" DEVICE_ALT2_VENDOR="" DEVICE_ALT2_MODEL="" DEVICE_ALT2_VARIANT="" DEVICE_TITLE="Xiaomi Redmi Router AX6S" DEVICE_PACKAGES="kmod-mt7915e" TARGET="mediatek" SUBTARGET="mt7622" VERSION_NUMBER="Master_18-11-2022" VERSION_CODE="r21326-bca663f20f" SUPPORTED_DEVICES="xiaomi,redmi-router-ax6s" /home/username/works/AX6S/scripts/json_add_image_info.py /home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/json_info_files/custom-master-18-11-2022-r21326-bca663f20f-mediatek-mt7622-xiaomi_redmi-router-ax6s-initramfs-recovery.itb.json
rm -f /home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/xiaomi_redmi-router-ax6s-kernel.bin
cp /home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/Image /home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/xiaomi_redmi-router-ax6s-kernel.bin
/home/username/works/AX6S/staging_dir/host/bin/lzma e /home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/xiaomi_redmi-router-ax6s-kernel.bin -lc1 -lp2 -pb2  /home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/xiaomi_redmi-router-ax6s-kernel.bin.new
/home/username/works/AX6S/scripts/mkits.sh -D xiaomi_redmi-router-ax6s -o /home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/xiaomi_redmi-router-ax6s-kernel.bin.its -k /home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/xiaomi_redmi-router-ax6s-kernel.bin -C lzma   -d /home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/image-mt7622-xiaomi-redmi-router-ax6s.dtb   -a 0x44000000 -e 0x44000000     -c "config-1" -A arm64 -v 5.15.79
PATH=/home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/linux-5.15.79/scripts/dtc:/home/username/works/AX6S/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/bin:/home/username/works/AX6S/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/bin:/home/username/works/AX6S/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/bin:/home/username/works/AX6S/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/bin:/home/username/works/AX6S/staging_dir/host/bin:/home/username/works/AX6S/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/bin:/home/username/works/AX6S/staging_dir/host/bin:/home/username/works/AX6S/staging_dir/host/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin mkimage  -f /home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/xiaomi_redmi-router-ax6s-kernel.bin.its /home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/xiaomi_redmi-router-ax6s-kernel.bin.new
/home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/xiaomi_redmi-router-ax6s-kernel.bin.its:17.11-19.6: Warning (unit_address_vs_reg): /images/kernel-1/hash@1: node has a unit name, but no reg or ranges property
/home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/xiaomi_redmi-router-ax6s-kernel.bin.its:20.11-22.6: Warning (unit_address_vs_reg): /images/kernel-1/hash@2: node has a unit name, but no reg or ranges property
/home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/xiaomi_redmi-router-ax6s-kernel.bin.its:34.11-36.6: Warning (unit_address_vs_reg): /images/fdt-1/hash@1: node has a unit name, but no reg or ranges property
/home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/xiaomi_redmi-router-ax6s-kernel.bin.its:37.11-39.6: Warning (unit_address_vs_reg): /images/fdt-1/hash@2: node has a unit name, but no reg or ranges property
FIT description: ARM64 OpenWrt FIT (Flattened Image Tree)
Created:         Fri Nov 18 19:27:52 2022
 Image 0 (kernel-1)
  Description:  ARM64 OpenWrt Linux-5.15.79
  Created:      Fri Nov 18 19:27:52 2022
  Type:         Kernel Image
  Compression:  lzma compressed
  Data Size:    4389625 Bytes = 4286.74 KiB = 4.19 MiB
  Architecture: AArch64
  OS:           Linux
[... 13 lines cut ...]
  Hash algo:    crc32
  Hash value:   2aab0edd
  Hash algo:    sha1
  Hash value:   9cd29a9030c5a37618b891e87eb4aabde134720e
 Default Configuration: 'config-1'
 Configuration 0 (config-1)
  Description:  OpenWrt xiaomi_redmi-router-ax6s
  Kernel:       kernel-1
  FDT:          fdt-1
bash: line 1: 8: Bad file descriptor
[ -f /home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/xiaomi_redmi-router-ax6s-kernel.bin -a -f /home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/root.squashfs ]
make[5]: *** [Makefile:31: /home/username/works/AX6S/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/tmp/custom-master-18-11-2022-r21326-bca663f20f-mediatek-mt7622-xiaomi_redmi-router-ax6s-squashfs-sysupgrade.bin] Error 1
make[5]: Leaving directory '/home/username/works/AX6S/target/linux/mediatek/image'
make[4]: *** [Makefile:17: install] Error 2
make[4]: Leaving directory '/home/username/works/AX6S/target/linux/mediatek'
make[3]: *** [Makefile:11: install] Error 2
make[3]: Leaving directory '/home/username/works/AX6S/target/linux'
time: target/linux/install#85.20#44.27#113.75
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/install] Error 1
make[2]: Leaving directory '/home/username/works/AX6S'
make[1]: *** [target/Makefile:24: /home/username/works/AX6S/staging_dir/target-aarch64_cortex-a53_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/username/works/AX6S'
make: *** [/home/username/works/AX6S/include/toplevel.mk:231: world] Error 2
```

musashino205, 2022-11-21:
The build system tried to display a message that the image generation failed on ```check-size```, probably because the size of kernel exceeded the limit, but it also failed. ```ERROR_MESSAGE``` implemetation seems to be broken.

- limit vs. kernel size

https://github.com/openwrt/openwrt/blob/0d375de10dac3160c65c264bb91a5137ef4c0817/target/linux/mediatek/image/mt7622.mk#L323

```
 Image 0 (kernel-1)
  Description:  ARM64 OpenWrt Linux-5.15.79
  Created:      Fri Nov 18 19:27:52 2022
  Type:         Kernel Image
  Compression:  lzma compressed
  Data Size:    4389625 Bytes = 4286.74 KiB = 4.19 MiB
```

- check-size of kernel

https://github.com/openwrt/openwrt/blob/0d375de10dac3160c65c264bb91a5137ef4c0817/include/image.mk#L560

- check-size

https://github.com/openwrt/openwrt/blob/0d375de10dac3160c65c264bb91a5137ef4c0817/include/image-commands.mk#L208-L215

trippleflux, 2022-11-21:
@musashino205 
Thanks, that's clear out my confusion.