openwrt/openwrt #11318: BPI-R3 bridge problem

vnoxxx, 2022-11-22:
After commit 9b482ee22f3f509880d2420652392971eef3211d (kernel: add more fixes for mtk_eth_soc), bridge is not working.

```
config device
	option name 'bridge0'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'

config interface 'lan'
	option device 'bridge0'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option zone 'lan'
```

I can ping 192.168.1.1, but DHCP is not working and I can´t connect using SSH or HTTPS…

If I remove 732-14-net-ethernet-mtk_eth_soc-drop-generic-vlan-rx-offloa.patch it works again.

Not sure if it's related, but I'm receiving these messages after "/etc/init.d/network reload" too, still trying to find the cause...

```
[49483.757250] page_pool_release_retry() stalled pool shutdown 20 inflight 1206 sec
[49544.013741] page_pool_release_retry() stalled pool shutdown 20 inflight 1266 sec
[49604.266293] page_pool_release_retry() stalled pool shutdown 20 inflight 1327 sec
[49664.523107] page_pool_release_retry() stalled pool shutdown 20 inflight 1387 sec
[49724.789378] page_pool_release_retry() stalled pool shutdown 20 inflight 1447 sec
[49785.034781] page_pool_release_retry() stalled pool shutdown 20 inflight 1508 sec
[49845.291468] page_pool_release_retry() stalled pool shutdown 20 inflight 1568 sec
```

dfateyev, 2022-11-22:
Just to note, these changes also affected [ramips](https://github.com/openwrt/openwrt/pull/11255#issuecomment-1324066616) and [mediatek](https://github.com/openwrt/openwrt/pull/11115#issuecomment-1322878322)

HZFrodo, 2022-11-23:
My Redmi AX6000 MT7986 has same problem with default network config after a clean upgrade.
iperf3 test from LAN to WAN has only ~8Mbps, and after a few minutes network is dead:
[ 2219.124396] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
[ 2219.130618] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[ 2219.150845] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
[ 2219.159233] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
[ 2224.084396] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
[ 2224.090624] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[ 2224.110849] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
[ 2224.119240] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
[ 2229.124397] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
[ 2229.130622] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[ 2229.150845] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
[ 2229.159249] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
[ 2234.084400] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
[ 2234.090631] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[ 2234.110884] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
[ 2234.119284] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
[ 2239.044395] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
[ 2239.050615] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[ 2239.070815] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
[ 2239.079214] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
[ 2244.084399] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
[ 2244.090628] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[ 2244.110836] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
[ 2244.119238] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
[ 2249.044395] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
[ 2249.050620] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[ 2249.070801] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
[ 2249.079189] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
[ 2254.084398] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
[ 2254.090626] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[ 2254.110839] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
[ 2254.119239] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx
[ 2259.124398] mtk_soc_eth 15100000.ethernet eth0: transmit timed out
[ 2259.130629] mtk_soc_eth 15100000.ethernet eth0: Link is Down
[ 2259.150860] mtk_soc_eth 15100000.ethernet eth0: configuring for fixed/2500base-x link mode
[ 2259.159263] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 2.5Gbps/Full - flow control rx/tx

nbd168, 2022-11-23:
Fixed in eaba63cddee44e0761cee1b0f82e3dd359e2dac1, sorry about that.

closed 2022-11-23 by nbd168

airend, 2022-11-23:
> Fixed in [eaba63c](https://github.com/openwrt/openwrt/commit/eaba63cddee44e0761cee1b0f82e3dd359e2dac1), sorry about that.

@nbd168, this patch doesn't fix multicast on mt7621 (broken since 9b482ee22f3f). I had to remove `target/linux/generic/pending-5.15/732-14-net-ethernet-mtk_eth_soc-drop-generic-vlan-rx-offloa.patch` entirely.

dfateyev, 2022-11-24:
> @nbd168, this patch doesn't fix multicast on mt7621 (broken since [9b482ee](https://github.com/openwrt/openwrt/commit/9b482ee22f3f509880d2420652392971eef3211d)). I had to remove `target/linux/generic/pending-5.15/732-14-net-ethernet-mtk_eth_soc-drop-generic-vlan-rx-offloa.patch` entirely.

@airend just wonder if your issue is resolved? I'm planning to flash some mt7621 boards with master snapshot, I don't use multicast but anyway.

airend, 2022-11-24:
> @airend just wonder if your issue is resolved? I'm planning to flash some mt7621 boards with master snapshot, I don't use multicast but anyway.

Unfortunately, nothing new merged to address it. I assumed it's a multicast packet issue when Chromecast stopped working, but might be deeper in bridging. I just build without that problematic patch introduced in 9b482ee22f3f509880d2420652392971eef3211d.

dfateyev, 2022-11-24:
> Unfortunately, nothing new merged to address it. I assumed it's a multicast packet issue when Chromecast stopped working, but might be deeper in bridging. I just build without that problematic patch [...]

If the regression is confirmed, I suggest to create a new issue, to keep it tracked.
I suppose that not much users utilize multicast with master on daily basis, so this problem if it's really present can be missed.

airend, 2022-11-25:
> If the regression is confirmed, I suggest to create a new issue, to keep it tracked. I suppose that not much users utilize multicast with master on daily basis, so this problem if it's really present can be missed.

I usually wait a little bit since @nbd168 tends to fix regressions fairly quickly. It's probably tough maintaining a single driver for several different HW generations of `mt753x`, plus adding new features. At any rate, he fixed it in 3dff6b53db1c568b957e7bde15eb5e049ea978ce.