openwrt/openwrt #11319: [22.03] [master] lzma-loader corrupted, soft-bricking devices

mpratt14, 2022-11-23:
https://patchwork.ozlabs.org/project/openwrt/patch/mailman.40844.1669077476.4154159.openwrt-devel@lists.openwrt.org/
I sent a fix to mailing list recently, but I wanted to expand on the issue
so hopefully this is taken care of faster.

commit a01d23e755ba46f41e667d558d82d4871d7f5450
although it's a very good idea, is causing side-effects and regressions
because of how the recipes for the lzma-loader are written

specifically the uImage file starts out with an append (>> redirection)
so if the file already exists, the new binary is placed on top of the lzma compressed binary
this leads to the lzma-loader having a uImage inside the lzma compressed uImage
and being twice the size, and not working

example:
`openwrt-22.03.2-ath79-generic-engenius_eap1200h-squashfs-sysupgrade.bin`
images for this device are actually a tar.gz
the lzma loader is the "uImage" file within the tar.gz image

extraction:
```
$ wget https://downloads.openwrt.org/releases/22.03.2/targets/ath79/generic/openwrt-22.03.2-ath79-generic-engenius_eap1200h-squashfs-sysupgrade.bin

$ gzip -dc openwrt*-ath79-generic-engenius_eap1200h-squashfs-sysupgrade.bin | tar -xvf -

$ dumpimage -o openwrt-ar71xx-generic-eap1200h-lzma.lzma openwrt-ar71xx-generic-eap1200h-uImage-lzma.bin

$ unlzma openwrt-ar71xx-generic-eap1200h-lzma.lzma
```

size difference:
(expected size is 0xD1C)

```
$ ls -lsa openwrt-ar71xx-generic-eap1200h-uImage-lzma.bin
8 -rw-r--r-- 1 pi pi 6737 Oct 14 18:44 openwrt-ar71xx-generic-eap1200h-uImage-lzma.bin

$ ls -lsa openwrt-ar71xx-generic-eap1200h-uImage-lzma.bin
4 -rw-r--r-- 1 pi pi 3329 Oct 13 15:16 openwrt-ar71xx-generic-eap1200h-uImage-lzma.bin
```

hexdump first couple lines after lzma decompression
`$ hexedit -l 16 openwrt-ar71xx-generic-eap1200h-lzma`

```
22.03.2 after decompression is another uImage

00000000   27 05 19 56  E5 3B DA 75  63 49 E6 59  00 00 0C DC  '..V.;.ucI.Y....
00000010   80 06 00 00  80 06 00 00  C2 07 0D 7F  05 05 02 03  ................
00000020   4D 49 50 53  20 4F 70 65  6E 57 72 74  20 4C 69 6E  MIPS OpenWrt Lin
00000030   75 78 2D 35  2E 31 30 2E  31 34 36 00  00 00 00 00  ux-5.10.146.....
00000040   6D 00 00 80  00 00 00 01  00 00 00 00  00 00 20 20  m.............
00000050   0E 00 08 43  32 05 54 97  4E 05 43 94  EB DC 70 06  ...C2.T.N.C...p.
00000060   B9 EA 83 3E  AF F2 49 A7  B9 D3 38 18  1E 05 48 CC  ...>..I...8...H.
00000070   AD FB 50 BB  FE EE 1E 65  41 C7 CE CC  97 4E 7E 72  ..P....eA....N~r
00000080   6C E5 E2 B7  2F 90 D7 9B  02 45 18 5E  B8 6E D9 71  l.../....E.^.n.q

21.02 after decompression is expected machine code

00000000   40 80 90 00  40 80 98 00  40 80 68 00  40 08 60 00  @...@...@.h.@.`.
00000010   3C 09 10 00  35 29 00 1F  01 09 40 25  39 08 00 1F  <...5)....@%9...
00000020   40 88 60 00  00 00 00 C0  40 08 80 00  24 09 FF F8  @.`.....@...$...
00000030   01 09 40 24  35 08 00 03  40 88 80 00  00 00 00 00  ..@$5...@.......
00000040   40 80 48 00  40 80 58 00  00 00 00 C0  3C 08 81 80  @.H.@.X.....<...
00000050   25 08 00 5C  04 11 00 01  00 00 00 00  03 E8 40 23  %..\..........@#
00000060   11 00 00 1C  00 00 00 00  3C 09 81 80  25 29 00 00  ........<...%)..
00000070   3C 0A 81 80  25 4A 17 00  01 09 40 21  8D 0B 00 00  <...%J....@!....
00000080   AD 2B 00 00  21 29 00 04  01 2A 08 2A  14 20 FF FB  .+..!)...*.*. ..

```

mpratt14, 2022-11-29:
@hauke @blocktrron @Ansuel @NeoRaider @aparcar @ynezz @dangowrt 

I don't mean to be rude or annoying, but can someone please look at this. approximately 20 or more boards in ath79 and ramips have images from master and 22.03.2 that will soft-brick the device

I believe the linked patch I'm presenting is the simplest fix, if not, please advise me

lynxis, 2022-12-01:
I'm trying to reproduce the issue.

mpratt14, 2022-12-01:
thanks for taking a look

keep in mind that it's not reproducible on first build on a new fresh buildroot, but happens on subsequent builds after that, even after `make clean`

like i said, one can simply inspect images on downloads.openwrt.com to see the issue, however I also reproduced it by building while trying to fix it.

lynxis, 2022-12-03:
```
- build with pm patch. make clean ; make -j8 ; generic ath79 -> building
  4053.36user 189.83system 47:40.48elapsed 148%CPU (0avgtext+0avgdata 230708maxresident)k
  0inputs+46719096outputs (0major+80099962minor)pagefaults 0swaps     
                                                                  
- build without pm patch. make clean ; make -j8 ; generic ath79
  6031.74user 235.30system 14:51.16elapsed 703%CPU (0avgtext+0avgdata 230516maxresident)k
  0inputs+46720256outputs (0major+80300923minor)pagefaults 0swaps

- build without broken patch. make clean ; make -j8 ; generic ath79
  6035.78user 235.34system 14:56.94elapsed 699%CPU (0avgtext+0avgdata 230552maxresident)k
  0inputs+46687104outputs (0major+79208292minor)pagefaults 0swaps
```

So 14 minutes against 45 minutes.

lynxis, 2022-12-03:
@mpratt14 is 
```
.NOTPARALLEL: $$(_COMPILE_TARGET)
```

really required?

mpratt14, 2022-12-03:
so is that really stopping many things from being parallel?

I have experienced build error just from the build of the lzma-loader uImage trying to append a file that was not made yet.

is there a better way to achieve forcing multiple things in COMPILE for each device to run in order?

lynxis, 2022-12-03:
At least building the loader in ath79 is safe. It's overwriting the PKG_BUILD_DIR to ensure the build is done in a different directory.

For sure we want to have correct images. If it takes 30 minutes longer, it's not nice, but ok for me. But I would try remove the NOTPARALLEL part and see how it goes.

lynxis, 2022-12-03:
without the NOTPARALLEL, it will break the build.

lynxis, 2022-12-03:
```
define Device/loader-okli-uimage                                                                                                       │
  LOADER_TYPE := bin                                                                                                                   │
  COMPILE := loader-$(1).bin loader-$(1).uImage                                                                                        │
  COMPILE/loader-$(1).bin := loader-okli-compile                                                                                       │
  COMPILE/loader-$(1).uImage := append-loader-okli $(1) | pad-to 64k | \                                                               │
        lzma | uImage lzma                                                                                                             │
endef                                                                                                                                  │
```

So append-loader-okli shouldn't use append and expect the other loader already exists. This is a different problem. It should rebuild it on it's own.

lynxis, 2022-12-03:
It is working now. But I'm still wondering why it is compiling the exact same target twice. Once in linux/compile, once in linux/install.

Can you try out https://github.com/lynxis/openwrt/tree/fiximage on a real device which uses the loader?

mpratt14, 2022-12-05:
if you put `loader-okli-compile` in the recipe for the `.uimage`, then the `.bin` recipe is not needed at all

however I prefer to fix this in the makefile, the recipes for `COMPILE` should be done in order

I understand the issue that `.NOTPARALLEL` is causing, by making all images build serially so im working on something else

mpratt14, 2022-12-10:
@lynxis have you been able to take a look at my new patch?

lynxis, 2022-12-12:
@mpratt14 I hope I can take a look in the next days. I know this is a high prio issue.
I also have to take a closer look on your patch.

But still my opinion on the COMPILE target is:
The COMPILE target is still broken if it depends on another COMPILE target. Also allowing it only in a specific order could break easily without knowing why.

mpratt14, 2022-12-12:
I would agree that the lzma-loader recipes and targets in ath79 and ramips etc. were not written well, but I think the makefiles should be able to handle it anyway.

However, I'm thinking there might be a reason for the way it was written, but not sure. I think using a COMPILE target for the bin allows the path and name to be built into the variables that are already available, which probably made it easier to write the recipes in <target>/image/Makefile

remember that forcing the order of COMPILE targets actually brings the behavior of the makefiles back to the way it was before the FORCE was added, that each COMPILE target was built in the order it was called in (the make `foreach` function that calls `Device/Build/compile` starting with the first word in COMPILE, etc.)

mpratt14, 2022-12-12:
I mean I *think* there is a reason that `append-loader-okli` exists as a separate recipe, or am I wrong and it's completely useless?

mpratt14, 2022-12-12:
ping @981213 if you're available to comment if there is any specific reason that lzma-loader `.bin` and `.uImage` are separate targets?

981213, 2022-12-12:
Oh. I was copy-pasting the .bin recipe to .uImage. They don't have to be separated on this particular device as only the uImage is needed.

mpratt14, 2022-12-13:
> Oh. I was copy-pasting the .bin recipe to .uImage. They don't have to be separated on this particular device as only the uImage is needed.

so I guess, just a funny coincidence that for every board in ath79 that needs lzma-loader in a similar way, it was copied the same way from that...

mpratt14, 2022-12-13:
@lynxis I decided to open a PR here to show both ways to fix it #11484 

even though the problem happens to be only in ath79 right now (I thought this affected ramips too but I was wrong), I think there still is value in improving the part of the makefiles that deals with `COMPILE` in image.mk

mpratt14, 2023-01-05:
@lynxis @hauke with release of 22.03.3 this can be closed

closed 2023-01-06 by ynezz