openwrt/openwrt #11326: kmod-usb-dwc2 is required for FRITZ!Box 7430

lustmue, 2022-11-23:
The default image for FRITZ!Box 7430 does not include support for the built in USB interface. @lmadam20, the author who added support for this device to OpenWrt, helped me to fix this issue by suggesting that I install kmod-usb-dwc2. I suggest that you include this module in the default image for FRITZ!Box 7430.

Spudz76, 2022-11-23:
Here is the quite simple patch that does what you requested.

[fritz7430_usb_dwc2.patch.txt](https://github.com/openwrt/openwrt/files/10079276/fritz7430_usb_dwc2.patch.txt)

lustmue, 2022-11-23:
Wow, that was quick. @adschm, you committed the initial support for this device. Do you want to review this related patch, too?

abajk, 2022-12-08:
@lustmue @Spudz76 One of you should open a new PR, to get the proposed changes merged.

closed 2022-12-11 by openwrt-bot

Spudz76, 2022-12-11:
PR has been merged into `master`

commit 9a1ab8aa (2022-12-11): lantiq: vr9: include usb driver for fritz 7430

Reported by user: missing driver for USB; add to image definition
https://github.com/openwrt/openwrt/issues/11326

Resolves: #11326

Acked-by: Aleksander Jan Bajkowski <olek2@wp.pl>
PR #11457 (2022-12-11): lantiq: vr9: include usb driver for fritz 7430
> Reported by user: missing driver for USB; add to image definition https://github.com/openwrt/openwrt/issues/11326
> 
> Resolves: #11326

See linked issue for more information.
commit 3b066a65 (2022-12-21): lantiq: vr9: include usb driver for fritz 7430

Reported by user: missing driver for USB; add to image definition
https://github.com/openwrt/openwrt/issues/11326

Resolves: #11326

Acked-by: Aleksander Jan Bajkowski <olek2@wp.pl>
(cherry picked from commit 9a1ab8aa0398f93360bf54d81bcd332cd413c03f)