openwrt/openwrt #11332: [Packages/perl] Problem building OpenWRT 19.07 packages on Debian

MattBystrin, 2022-11-24:
Hi ! I've faced a problem during the build. The error raised in perl package. 
I tried to do the same build on 9 and 10 Debian, with and without chroot. The thing is that the same build goes without errors on Arch and Mint, so I think that it is distro-related issue.

Here is my log:
~~~
builder@debian:~/openwrt-ipq50-rc$ make -j1 V=s package/feeds/packages/perl/compile
WARNING: Makefile 'package/feeds/packages/openssh/Makefile' has a dependency on 'libfido2', which does not exist
WARNING: Makefile 'package/feeds/packages/openssh/Makefile' has a build dependency on 'libfido2', which does not exist
make[1]: Entering directory '/home/builder/openwrt-ipq50-rc'
make[2]: Entering directory '/home/builder/openwrt-ipq50-rc/package/libs/toolchain'
echo "libc" >> /home/builder/openwrt-ipq50-rc/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi_R50/pkginfo/toolchain.default.install
echo "libgcc" >> /home/builder/openwrt-ipq50-rc/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi_R50/pkginfo/toolchain.default.install
echo "libpthread" >> /home/builder/openwrt-ipq50-rc/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi_R50/pkginfo/toolchain.default.install
echo "librt" >> /home/builder/openwrt-ipq50-rc/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi_R50/pkginfo/toolchain.default.install
make[2]: Leaving directory '/home/builder/openwrt-ipq50-rc/package/libs/toolchain'
time: package/libs/toolchain/compile#0.11#0.01#0.12
make[2]: Entering directory '/home/builder/openwrt-ipq50-rc/package/libs/zlib'
echo "zlib" >> /home/builder/openwrt-ipq50-rc/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi_R50/pkginfo/zlib.default.install
make[2]: Leaving directory '/home/builder/openwrt-ipq50-rc/package/libs/zlib'
time: package/libs/zlib/compile#0.11#0.03#0.14
make[2]: Entering directory '/home/builder/openwrt-ipq50-rc/feeds/packages/libs/libxml2'
make[2]: Leaving directory '/home/builder/openwrt-ipq50-rc/feeds/packages/libs/libxml2'
time: package/feeds/packages/libxml2/compile#0.14#0.03#0.15
make[2]: Entering directory '/home/builder/openwrt-ipq50-rc/package/libs/uclibc++'
echo "uclibcxx" >> /home/builder/openwrt-ipq50-rc/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi_R50/pkginfo/uclibc++.default.install
make[2]: Leaving directory '/home/builder/openwrt-ipq50-rc/package/libs/uclibc++'
time: package/libs/uclibc++/compile#0.11#0.01#0.12
make[2]: Entering directory '/home/builder/openwrt-ipq50-rc/feeds/packages/libs/db47'
make[2]: Leaving directory '/home/builder/openwrt-ipq50-rc/feeds/packages/libs/db47'
time: package/feeds/packages/db47/compile#0.12#0.03#0.13
make[2]: Entering directory '/home/builder/openwrt-ipq50-rc/package/libs/gettext-full'
make[2]: Leaving directory '/home/builder/openwrt-ipq50-rc/package/libs/gettext-full'
time: package/libs/gettext-full/host-compile#0.13#0.05#0.15
make[2]: Entering directory '/home/builder/openwrt-ipq50-rc/feeds/packages/libs/gdbm'
make[2]: Leaving directory '/home/builder/openwrt-ipq50-rc/feeds/packages/libs/gdbm'
time: package/feeds/packages/gdbm/compile#0.09#0.03#0.11
make[2]: Entering directory '/home/builder/openwrt-ipq50-rc/feeds/packages/lang/perl'
make[2]: Leaving directory '/home/builder/openwrt-ipq50-rc/feeds/packages/lang/perl'
time: package/feeds/packages/perl/host-compile#0.21#0.07#0.25
make[2]: Entering directory '/home/builder/openwrt-ipq50-rc/feeds/packages/lang/perl'
rm -f /home/builder/openwrt-ipq50-rc/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi_R50/perl/perl-5.28.1/.built
touch /home/builder/openwrt-ipq50-rc/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi_R50/perl/perl-5.28.1/.built_check
make  -C /home/builder/openwrt-ipq50-rc/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi_R50/perl/perl-5.28.1
make[3]: Entering directory '/home/builder/openwrt-ipq50-rc/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi_R50/perl/perl-5.28.1'
/bin/ln -s /home/builder/openwrt-ipq50-rc/staging_dir/hostpkg/usr/bin/generate_uudmap generate_uudmap
/bin/ln -s /home/builder/openwrt-ipq50-rc/staging_dir/hostpkg/usr/bin/perl miniperl
LD_LIBRARY_PATH=/home/builder/openwrt-ipq50-rc/build_dir/target-arm_cortex-a7\+neon-vfpv4_musl_eabi_R50/perl/perl-5.28.1 ./miniperl -w -Ilib -I. -Idist/Exporter/lib -MExporter -e '<?>' || sh -c 'echo >&2 Failed to build miniperl.  Please run make minitest; exit 1'
Can't locate File/Glob.pm in @INC (you may need to install the File::Glob module) (@INC contains: lib . dist/Exporter/lib ./. ./lib/perl5/site_perl/5.28.1/x86_64-linux ./. ./lib/perl5/site_perl/5.28.1 ./. ./lib/perl5/5.28.1/x86_64-linux ./. ./lib/perl5/5.28.1) at -e line 1.
BEGIN failed--compilation aborted at -e line 1.
Failed to build miniperl. Please run make minitest
makefile:360: recipe for target 'lib/buildcustomize.pl' failed
make[3]: *** [lib/buildcustomize.pl] Error 1
make[3]: Leaving directory '/home/builder/openwrt-ipq50-rc/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi_R50/perl/perl-5.28.1'
Makefile:143: recipe for target '/home/builder/openwrt-ipq50-rc/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi_R50/perl/perl-5.28.1/.built' failed
make[2]: *** [/home/builder/openwrt-ipq50-rc/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi_R50/perl/perl-5.28.1/.built] Error 2
make[2]: Leaving directory '/home/builder/openwrt-ipq50-rc/feeds/packages/lang/perl'
time: package/feeds/packages/perl/compile#0.26#0.03#0.28
package/Makefile:111: recipe for target 'package/feeds/packages/perl/compile' failed
make[1]: *** [package/feeds/packages/perl/compile] Error 2
make[1]: Leaving directory '/home/builder/openwrt-ipq50-rc'
/home/builder/openwrt-ipq50-rc/include/toplevel.mk:225: recipe for target 'package/feeds/packages/perl/compile' failed
make: *** [package/feeds/packages/perl/compile] Error 2
~~~

As we can see perl have problems with the include path.

pkgadd, 2022-11-24:
openwrt-19.07 is EOL and out of support, not that it ever had ipq50xx support in the first place (which isn't even available in master, yet).

MattBystrin, 2022-11-25:
Figured out. Just needed to have installed curl on host. Yep )

closed 2022-11-25 by MattBystrin