openwrt/openwrt #11333: Flashing Cudy X6 / AX 1800 seems to fail

waja, 2022-11-24:
Hi,

as described already in [the forum](https://forum.openwrt.org/t/cudy-x6-ax-1800-how-to-flash-successfully/143631):

I recently bought a Cudy X6. From what I'm reading, 22.03 should support those devices. Looking into the commit message it should be at first flashed the vendor OpenWRT image, which can be obtained from their website. It seems there are different revisions in the wild, but it's possible to identify them.
As it seems I've a V1 (serial includes an 'EU1.0), I downloaded the X6 (V1) OpenWRT and uploaded the included openwrt-ramips-mt7621-cudy-x6-flash.bin which worked well.
As documented, I used NOT the option to keep settings and flashed openwrt-22.03.2-ramips-mt7621-cudy_x6-squashfs-sysupgrade.bin 1.

The systems rebootes and the device get stuck, the timeline after powering up:

0s: short all NIC-LED lighting
1s: short all NIC-LED lighting
4s: connected NIC-LED shows activity
8s: short all NIC-LED lighting, active NIC-LED turns off
followed by 2s nothing => starting again at 0s

I tried both, via LuCI and CLI, no difference. Did I miss something or did something wrong? Reading through the forum and google seems to indicate it SHOULD work but somehow I managed to get it not working. Any ideas?

In the forum was [pointed out](https://forum.openwrt.org/t/cudy-x6-ax-1800-how-to-flash-successfully/143631/2?u=waja), that the issue might be the dts config. Searching the web it seems to be possible to extract this from a running system via `fdtdump`?

So I did this on the router with the Cudy customized OpenWRT image installed:

```
root@OpenWrt:~# fdtdump /sys/firmware/fdt

**** fdtdump is a low-level debugging tool, not meant for general use.
**** If you want to decompile a dtb, you probably want
****     dtc -I dtb -O dts <filename>

/dts-v1/;
// magic:		0xd00dfeed
// totalsize:		0x293f (10559)
// off_dt_struct:	0x38
// off_dt_strings:	0x2614
// off_mem_rsvmap:	0x28
// version:		17
// last_comp_version:	16
// boot_cpuid_phys:	0x0
// size_dt_strings:	0x32b
// size_dt_struct:	0x25dc

/ {
    #address-cells = <0x00000001>;
    #size-cells = <0x00000001>;
    compatible = "cudy,x6", "mediatek,mt7621-soc";
    model = "CUDY X6";
    cpus {
        #address-cells = <0x00000001>;
        #size-cells = <0x00000000>;
        cpu@0 {
            device_type = "cpu";
            compatible = "mips,mips1004Kc";
            reg = <0x00000000>;
        };
        cpu@1 {
            device_type = "cpu";
            compatible = "mips,mips1004Kc";
            reg = <0x00000001>;
        };
    };
    cpuintc {
        #address-cells = <0x00000000>;
        #interrupt-cells = <0x00000001>;
        interrupt-controller;
        compatible = "mti,cpu-interrupt-controller";
    };
    aliases {
        serial0 = "/palmbus@1E000000/uartlite@c00";
        led-boot = "/leds/internet_blue";
        led-failsafe = "/leds/internet_blue";
[... 523 lines cut ...]
            label = "reset";
            gpios = <0x00000014 0x00000008 0x00000001>;
            linux,code = <0x00000198>;
        };
        wps {
            label = "wps";
            gpios = <0x00000014 0x00000007 0x00000001>;
            linux,code = <0x00000211>;
        };
    };
    leds {
        compatible = "gpio-leds";
        internet_blue {
            label = "blue:internet";
            gpios = <0x00000014 0x0000000f 0x00000001>;
        };
        internet_red {
            label = "red:internet";
            gpios = <0x00000014 0x00000010 0x00000001>;
        };
    };
};
```

Many thanks, Jan

waja, 2022-11-27:
Neither https://github.com/waja/openwrt-cudy-x6-firmware/blob/5e892565471cf9be64c994e70837ecc670247436/mt7621_cudy_x6_fix_dts_broken-flash-reset.patch nor https://github.com/waja/openwrt-cudy-x6-firmware/blob/5e892565471cf9be64c994e70837ecc670247436/mt7621_cudy_x6_fix_dts_m25p_fast-read.patch does fix the issue for me.

DragonBluep, 2022-12-01:
Please put the vendor OpenWrt bootlog here. I think this is a Flash compatibility problem.

Spudz76, 2022-12-01:
Really tough to guess what's broken without serial port console output.  I had the same thing on a Linksys EA7300v1 (with broken "failsafe" bank of OEM firmware) and couldn't proceed until I got console working, it was boot-looping just like yours seems to be doing (unable to mount rootfs / some other early init issue).

WIsh there was some way for it to spew early console messages to broadcast on the network (to be seen on a sniffer or something) then non-serial-capable units could still be somewhat debugged (unidrectional read-only).  Message-in-a-bottle mode? haha

waja, 2022-12-01:
> Please put the vendor OpenWrt bootlog here. I think this is a Flash compatibility problem.

@DragonBluep you mean the bootlog when booting into the Cudy provided OpenWRT Firmware?

DragonBluep, 2022-12-01:
yes

waja, 2022-12-01:
You can find the `logread` output in [logread.log(https://github.com/openwrt/openwrt/files/10130870/logread.log).

Thanks Jan

DragonBluep, 2022-12-01:
@waja This commit should fix it. https://github.com/openwrt/openwrt/pull/9465/commits/98be59ecf65106ba8068049664603f8c225bf840

DragonBluep, 2022-12-01:
@Spudz76 Maybe this is because you have flushed the firmware into the wrong partition. Please make sure you follow the steps. https://github.com/openwrt/openwrt/commit/ed087cba8a8e41f76f9487caa34eff926ea8a065
> you should flash the OEM image over itself. This will then boot the router from 'B' and allow you to flash OpenWRT without 
> problems.

waja, 2022-12-01:
@DragonBluep okay ... Pipelines for [openwrt-22.03](https://github.com/waja/openwrt-cudy-x6-firmware/actions/runs/3591906825/jobs/6046999682) and [master](https://github.com/waja/openwrt-cudy-x6-firmware/actions/runs/3591913199/jobs/6047014188) are running ... waiting for the binaries. :)

waja, 2022-12-01:
This is so fucking awesome? Release [1337](https://github.com/waja/openwrt-cudy-x6-firmware/releases/tag/2022.12.01-1337) fixes the issues. So https://github.com/openwrt/openwrt/commit/98be59ecf65106ba8068049664603f8c225bf840 is fixing my issue. It would be very nice if this hits `master` AND `openwrt-22.03` soon.
@DragonBluep thanks for pointing this out. Is there any issue/PR referencing those patches for keeping them track?

DragonBluep, 2022-12-01:
Just wait and be patient, it's already on the way. https://github.com/openwrt/openwrt/pull/9465

Spudz76, 2022-12-01:
> @Spudz76 Maybe this is because you have flashed the firmware into the wrong partition. Please make sure you follow the steps. [ed087cb](https://github.com/openwrt/openwrt/commit/ed087cba8a8e41f76f9487caa34eff926ea8a065) [50f727b](https://github.com/openwrt/openwrt/commit/50f727b7737d118f7d44986181e305af0624c41d)
> 
> > you should flash the OEM image over itself. This will then boot the router from 'B' and allow you to flash OpenWRT without
> > problems.

No, but thanks.  There is a huge region of bad NAND blocks right in the midst of the second bank.

waja, 2022-12-05:
I just want to report for completeness, that the system is reporting `MediaTek MT7621 ver:1 eco:3`, on the board is written `Rev1.3`.

yogo1212, 2023-01-06:
hi - just a notification that support for the en256qh256a was merged ;-)

Djfe, 2023-03-31:
I couldn't figure out whether this has been backported to Release 22. If not, then there is another chance for 22.03.04 now to PR this as a backport.
@yogo1212 do you know more?

yogo1212, 2023-03-31:
Generally, the wiki says [not to rely on device backports happening](
https://openwrt.org/docs/guide-developer/device-support-policies?s[]=backport&s[]=device#device_support_backports).

Maybe the flash chip support patch has a better chance on its own.
It affects the X6 which is part of the release.
You could remove the 5.15 from [the commit](https://github.com/openwrt/openwrt/commit/ed7c1a504eae6b220e01da330c62c47d299e2c5d) and try to submit that.