openwrt/openwrt #11336: RTL8811cu support?

mr-gito, 2022-11-24:
Any chance for adding RTL8811cu/RTL8821cu support ?
Bus 001 Device 008: ID 0bda:c811 Realtek 802.11ac NIC

https://github.com/brektrou/rtl8821CU.git

mincaeuro, 2023-01-07:
did you solve this ?

ThiloteE, 2023-01-07:
There is ongoing rtl8821cu support in upstream kernel. See here:

- https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?h=v6.2-rc2&id=aff5ffd718de23cb8603f2e229204670e2644334
- https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?h=v6.2-rc2&id=fba119cee1412f49c472f3913241d00f830761fd
- https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?h=v6.2-rc2&id=7e68dd7d07a28faa2e6574dd6b9dbd90cdeaae91

Once OpenWRT reaches kernel 6.2+ it should work here too. Maybe earlier, if those are backported. It has reached a point where anybody with willingness and knowledge to support these chipsets should probably put their time and effort into the upstream project, rather than out of kernel packages.

everschen, 2024-03-09:
RTL8821cu seems work but not stable, it will disconnect after several minutes on 6.1