openwrt/openwrt #11343: USB Wifi Adapter Cannot be added

gavinsauder, 2022-11-26:
Setting up openwrt as per: https://www.youtube.com/watch?v=jlHWnKVpygw

I have this usb adapter: https://www.amazon.com/gp/product/B01KWQAQ00/ref=ppx_od_dt_b_asin_title_s00?ie=UTF8&psc=1

When installing drivers for usb adapter I used:
```
opkg update

opkg install kmod-rt2800-lib kmod-rt2800-usb kmod-rt2x00-lib kmod-rt2x00-usb kmod-usb-core kmod-usb-uhci kmod-usb-ohci kmod-usb2 usbutils openvpn-openssl luci-app-openvpn nano
```
I can see my usb device if I try: lsusb

But when I do: ifconfig wlan1 up

I get: ifconfig: SIOCGIFFLAGS: No such device

How do I fix this? Do I not have the correct drivers? What driver should I use?

Thanks for your time.
Gavin

Mijzelf, 2022-11-26:
> What driver should I use?
Plug it in your favorite Linux lap/desktop, and look which modules it loads. If your Linux system doesn't recognize it, odds are OpenWrt won't either.

brada4, 2022-11-27:
Probably rt2800-usb-firmware is needed, or even some more specific firmware file extracted from linux-firmware. rt2800 seems correct driver,, but please show lsusb output to be sure. Also there should be something in `dmesg` starting at connecting USB device down to (not yet) initializing wifi phy.

gavinsauder, 2022-11-28:
@brada4 this is my lsusb output:

Bus 002 Device 001: ID 1d6b:0003 Linux 5.10.146 xhci-hcd xHCI Host Controller
Bus 001 Device 003: ID 148f:7601 MediaTek 802.11 n WLAN
Bus 001 Device 002: ID 2109:3431  USB2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux 5.10.146 xhci-hcd xHCI Host Controller

and this is what I see at the bottom for dmesg after connecting wifi adapter:

[ 3558.560210] usb 1-1.1: new high-speed USB device number 3 using xhci_hcd
[ 3558.701350] usb 1-1.1: New USB device found, idVendor=148f, idProduct=7601, bcdDevice= 0.00
[ 3558.709713] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3558.717082] usb 1-1.1: Product: 802.11 n WLAN
[ 3558.721485] usb 1-1.1: Manufacturer: MediaTek
[ 3558.725874] usb 1-1.1: SerialNumber: 1.0

brada4, 2022-11-28:
Should work with installed kmod-rt2800-usb 
Typically USB core and port driver is installed by default in device kernel. You certainly do not need USB 1.0 OHCI driver.

Please try re-plugging after installing firmware package. 

On desktop distro-s it works, 100MB firmware is usually installed there by default and it works as stated here:
https://linux-hardware.org/index.php?id=usb:148f-5370

Different card, but 3 lines from ASIC to minstrel are absent if removing firmware files.
```
[464576.883060] usb 1-1: new high-speed USB device number 3 using ehci-pci
[464577.058960] usb 1-1: New USB device found, idVendor=148f, idProduct=761a, bcdDevice= 1.00
[464577.058974] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[464577.058980] usb 1-1: Product: WiFi
[464577.058984] usb 1-1: Manufacturer: MediaTek
[464577.058988] usb 1-1: SerialNumber: 1.0
[464577.979154] usb 1-1: reset high-speed USB device number 3 using ehci-pci
[464578.142172] mt76x0u 1-1:1.0: ASIC revision: 76100002 MAC revision: 76502000
[464579.208673] mt76x0u 1-1:1.0: EEPROM ver:02 fae:01
[464579.255038] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[464579.256815] usbcore: registered new interface driver mt76x0u
```

dhewg, 2022-11-28:
Wait, what?
7601 is not 5370. 7601 is "Ralink MT7601U", which has a MediaTek chip. And kmod-rt2800-usb has nothing to do with that.
Searching for "148f 7601 openwrt" and the top two hits have all the relevant infos.
Try kmod-mt7601u and mt7601u-firmware. But according to those two hits, it doesn't seem to support AP mode?

brada4, 2022-11-28:
your usb wifi never starts radio. Yes, my is different, if yours ever shows minstrel radio fairness algorithm you can get to configuring it "up", without it no go.
1/ driver rt2800usb is correct
2/ firmware (based on assumption desktop distros always have it) is still missing.

you refer to random internet pages about my card - they are totaly off-beat, it works as ap, sta and mon, and most importantly within usb 2.0 power bracket, sadly only one of 2.4 or 5.0 at once.

gavinsauder, 2022-11-30:
What firmware should I be using? I think I need it to support AP mode.

brada4, 2022-11-30:
`rt2800-usb-firmware` kind of there is no other choice. Then check dmesg while plugging in USB NIC. Fastest is to check in place if AP mode statts via web, then examine sources, changelogs etc.

DragonBluep, 2022-12-01:
> What firmware should I be using? I think I need it to support AP mode.

As far as I know, MT7601U does not support AP mode. `kmod-mt7601u` and `mt7601u-firmware` are required packages.
https://wireless.wiki.kernel.org/en/users/drivers

brada4, 2022-12-01:
My network adapter is not related to yours
What error says - there is no interface named wlan1. Try to find the name:
Wireless:
`iwinfo`
`iw list`
```
valid interface combinations:
                 * #{ managed } <= 2048, #{ AP, mesh point } <= 8, #{ P2P-client, P2P-GO } <= 1,
                   total <= 2048, #channels <= 1, STA/AP BI must match
```
Any interface:
`ip link`
```
16: wlan2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP qlen 1000
```
`ifconfig -a`

Network driver/firmware info:
`ethtool -i eth0`

dhewg, 2022-12-01:
And since @gavinsauder asked about his and not yours we can finally stop talking about rt2000.
This needs to be closed and moved to the forum. And I want my time back.

981213, 2022-12-02:
kmod-mt7601u and mt7601u-firmware are the driver for your MT7601U dongle and it doesn't support AP mode. You'll be better off with a different USB wifi adapter if you really need the AP support.
This isn't an OpenWrt issue. Closed as requested. :)

closed 2022-12-02 by 981213