openwrt/openwrt #11369: Newest OpenWRT builds for TP-Link Archer C5 v4 -- Missing option in 'menuconfig'

MartinVonReichenberg, 2022-11-28:
Hello,
Is there any chance on getting the latest build of **OpenWRT** for TP-Link **Archer C5 v4** _(MediaTek MT7620A)_
?_?

```Firmware Version:3.16.0 0.9.1 v600c.0 Build 191107 Rel.15316n Hardware Version:Archer C5 v4 00000004```

[image]
As you can see, there is a missing option in `'menuconfig'`, the option for **C5 v4** is missing, but there is option for **C5 v1** . . . . .

981213, 2022-12-02:
Hi!
This device isn't supported in the mainline openwrt yet.
You can continue using the fork for this particular device if you want.

closed 2022-12-02 by 981213

namiltd, 2023-07-18:
Look at https://github.com/openwrt/openwrt/pull/4327

luizluca, 2023-11-09:
> Look at #4327

The best luck for c5v4 is to get supported using upstream DSA driver. For that, we need #12738. There is simply too much kernel  backports to make it work safely with older kernels.

@namiltd , I noticed you are already playing with 6.1.

namiltd, 2023-11-09:
[luizluca](https://github.com/luizluca) Yes, but without DSA at the moment. I'm also waiting for my patches to be merged to make kernel 6.1 work.