openwrt/openwrt #11370: 21.02.3 & 21.02.5 sysupgrade should cause wndr4300 reboot again & again.

wenewzhang, 2022-11-28:
How to replay the bug:
power off the wndr4300 router,  press reset button and then power on, hold reset button for 15 secs,
after power light turn green and flash, release the reset button,

tftp 192.168.1.1 
binary 
put 21.02.3 or 21.02.5 factory img,

after router power on,
from luci -> backup/flash firmware -> flash 21.02.3 or 21.02.5 sysupgrade.bin,

and then, the router should reboot again & again,

the same step on 21.02.2 works fine.

wenewzhang, 2022-11-29:
more test on 21.02.2:
factory image -> sysupgrade.bin (works) -> sysupgrade.bin ( not work)

on 21.02.1

factory image -> sysupgrade.bin (works) -> sysupgrade.bin ( works)

The different between v21.02.1 & v21.02.2,
i suppose that this dts cause new problem,
```
diff --git a/target/linux/ath79/dts/ar9344_ocedo_raccoon.dts b/target/linux/ath79/dts/ar9344_ocedo_raccoon.dts
index e7ef02054e..64805eed43 100644
--- a/target/linux/ath79/dts/ar9344_ocedo_raccoon.dts
+++ b/target/linux/ath79/dts/ar9344_ocedo_raccoon.dts
@@ -150,16 +150,27 @@
 
        phy0: ethernet-phy@0 {
                reg = <0>;
+               eee-broken-100tx;
+               eee-broken-1000t;
        };
 };
 
 &eth0 {
        status = "okay";
 
-       pll-data = <0x06000000 0x00000101 0x00001313>;
+       pll-data = <0x02000000 0x00000101 0x00001313>;
 
        mtd-mac-address = <&art 0x0>;
 
        phy-mode = "rgmii-id";
        phy-handle = <&phy0>;
+
+       gmac-config {
+               device = <&gmac>;
+               rgmii-gmac0 = <1>;
+               rxdv-delay = <3>;
+               rxd-delay = <3>;
+               txen-delay = <0>;
+               txd-delay = <0>;
+       };
 };

```

wenewzhang, 2022-11-29:
By the way, all the sysupgrade doesn't keep the configuration.

DragonBluep, 2022-11-29:
> --- a/target/linux/ath79/dts/ar9344_ocedo_raccoon.dts

These firmware are not universal. Please use the correct firmware of the wndr4300.

wenewzhang, 2022-11-29:
Finally, I figure out the problem is that the run openwrt os miss the tar command,
because the custom firmware has remove /bin/tar for some purpose, so, fresh new firmware cause this problem

closed 2022-11-29 by wenewzhang