openwrt/openwrt #11372: Poor download rate on 100baseTx-FD linked client [TP-Link Archer C7 v5]

Neher-Veem, 2022-11-28:
This issue is a partial clone of my comment in the referenced thread just to let OpenWRT dev team know that this problem is also affecting TP-Link Archer C7 v5 and probably many other devices when a Fast Ethernet client is connected to any port on the router's gigabit switch.

ISP line: 100mbps/30mbps
Cable Modem Technicolor, auto linking at 1gbps with wan port.
Testing device: Dell laptop with: Realtek Semiconductor Co., Ltd. RTL810xE PCI Express Fast Ethernet controller (rev 05)
Testing OS: Ubuntu 22.04, Linux mint 20.3 and 21, Fedora 37.

I tested every stable release from OpenWRT 18.06.0 to 22.03.2 and I reproduced the same behavior every time:
Download speed in speedtest.net seems to be limited to ~30mbps. Upload speed seems to be fine, but my ISP connection limit is 30mbps too so I can't tell if the problem is also affecting the upload.
I also could find that not all kind of traffic is limited, for example, http/https download work normal (Screenshot 02 and 03 with aria2c downloading an Ubuntu ISO after running speedtest)
I can confirm **there is no such issue with OEM TP-Link Firmware** (Screenshot 01).
I can confirm the issue **is also present on DD-WRT** even in the latest versions (but I forgot to take a screenshot).

[image]
[image]
[image]

_Originally posted by @Neher-Veem in https://github.com/openwrt/openwrt/issues/9202#issuecomment-1329228468_

brada4, 2022-12-01:
Please see #10790 - I have same problem, 5GHz wifi STA-STA is faster than VLAN gigabit to other VLAN gigabit via CPU port. Not sure in regards to my guesses back there.

brada4, 2022-12-04:
I have cleared both PAUSE bits and rebuilding 22.03.2 recovered speed over CPU port. i.e I get all subscribed 250Mbps over default NAT to speedtest/fast/waveform speed tests.
https://github.com/openwrt/openwrt/blob/a8f3c97ce83dc97cd7acb040a715ecd75d824161/target/linux/generic/files/drivers/net/phy/ar8216.c#L212
Just an observation with unpatched - having any 100Mbps port with any traffic generates storm of pause frames which unconditionally broadcasts to all ports, including on totally isolated VLAN bound to one single untagged port and configured UP without any IP.

Neher-Veem, 2023-08-05:
@brada4 Have you had any updates on this bug in these months? Im using stock firmware and I miss a lot of OpenWRT functionalities :(

brada4, 2023-08-05:
Enable fw soft offload, hard offload drops states for ,e.

Neher-Veem, 2023-08-05:
I already tried that, and sadly, it had no impact on TP-Link Archer C7 v5.

brada4, 2023-08-05:
Bought a new router, not that this is that bad.
1/ flow control disabled clients work ok
2/ flow control enabled clients get pause packet on any drop anywhere, one of them is 'system' port. It is not 100Mbps/12.5MBps exactly, but having full load on slow port too many pause gets broadcast.

Neher-Veem, 2023-08-05:
I don't understand why a stock firmware can handle the bandwidth perfectly
without any issues and OpenWRT just can't.

brada4, 2023-08-05:
It is switching ASIC programming, just some bitmask and other set of bitmasks per port. Not exactly the Linux running on SoC CPU.

danpawlik, 2023-08-05:
Do you have packet steering  enabled?

Neher-Veem, 2023-08-05:
I think I had tried that too and didn't work either, now my router has
stock firmware installed so I would have to install openwrt to check one
more time.

brada4, 2023-08-05:
One core no steering.

danpawlik, 2023-08-06:
> One core no steering.

As I see from https://openwrt.org/toh/hwdata/tp-link/tp-link_archer_c7_v5 - it has CPU Qualcomm Atheros QCA9563, which have 2 cores, or?

brada4, 2023-08-06:
The problem is a gigabit switch, not CPU(s),  that performs significantly worse than generic unmanaged 5-port gigabit switch.