openwrt/openwrt #11380: ARMv8 python3 compile fail

very20101, 2022-11-29:
package/feeds/packages/python3 compile fail

ARMv8 

ubuntu 22.04

make[3] -C feeds/packages/lang/python/python3 compile
 make[3] -C feeds/packages/net/miniupnpd compile
 make[3] -C feeds/packages/net/vpnbypass compile
 make[3] -C feeds/packages/net/nginx-util compile
 make[3] -C feeds/packages/utils/parted compile
 make[3] -C feeds/packages/net/krb5 compile
   ERROR: package/feeds/packages/python3 failed to build.

.......

/usr/bin/env bash ../python-package-install.sh "/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-install" "/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base" "$V_Py3Package_python3_base_filespec"
copying: '/usr/bin/python3.10'
copying: '/usr/lib/python3.10/encodings'
copying: '/usr/lib/python3.10/_collections_abc.py'
copying: '/usr/lib/python3.10/_sitebuiltins.py'
copying: '/usr/lib/python3.10/_sysconfigdata.py'
copying: '/usr/lib/python3.10/_weakrefset.py'
copying: '/usr/lib/python3.10/abc.py'
copying: '/usr/lib/python3.10/codecs.py'
copying: '/usr/lib/python3.10/genericpath.py'
copying: '/usr/lib/python3.10/io.py'
copying: '/usr/lib/python3.10/os.py'
copying: '/usr/lib/python3.10/posixpath.py'
copying: '/usr/lib/python3.10/site.py'
copying: '/usr/lib/python3.10/sysconfig.py'
copying: '/usr/lib/python3.10/stat.py'
find /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base -name '*.exe' -delete
cd "." && CC="aarch64-openwrt-linux-musl-gcc" CCSHARED="aarch64-openwrt-linux-musl-gcc -DPIC -fPIC" CXX="aarch64-openwrt-linux-musl-g++" LD="aarch64-openwrt-linux-musl-gcc" LDSHARED="aarch64-openwrt-linux-musl-gcc -shared" CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7=Python-3.10.7 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CPPFLAGS="-I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/include -I/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include/python3.10" LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-11.3.0_musl/lib -znow -zrelro -lpython3.10" _PYTHON_HOST_PLATFORM="linux-aarch64" __PYVENV_LAUNCHER__="/usr/bin/python3.10" PYTHONPATH="/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/python3.10:/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl//usr/lib/python3.10/site-packages://usr/lib/python3.10/site-packages" PYTHONDONTWRITEBYTECODE=1 PYTHONOPTIMIZE="" _python_sysroot="/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl" _python_prefix="/usr" _python_exec_prefix="/usr" 
PYTHONHASHSEED="1669557968"
/workdir/openwrt/staging_dir/hostpkg/bin/python3.10  -m compileall -r "20" -b -d '/' /workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base
Listing '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base'...
Listing '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/CONTROL'...
Listing '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr'...
Listing '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/bin'...
Listing '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib'...
Listing '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/_collections_abc.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/_sitebuiltins.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/_sysconfigdata.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/_weakrefset.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/abc.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/codecs.py'...
Listing '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/__init__.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/aliases.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/ascii.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/base64_codec.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/big5.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/big5hkscs.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/bz2_codec.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/charmap.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/cp037.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/cp1006.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/cp1026.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/cp1125.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/cp1140.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/cp1250.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/cp1251.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/cp1252.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/cp1253.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/cp1254.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/cp1255.py'...
Compiling '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/Python-3.10.7/ipkg-aarch64_cortex-a53/python3-base/usr/lib/python3.10/encodings/cp1256.py'...
[... cut ...]
time: package/feeds/packages/python3/compile#0.63#0.14#0.73
    ERROR: package/feeds/packages/python3 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/python3/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

jow-, 2022-11-29:
> /workdir/openwrt/staging_dir/host/bin/sed: cannot rename .../ipkg-aarch64_cortex-a53/python3-base/usr/bin/sedYOsNqu: Not a directory

Make sure you're not running out of disk space or inodes on your build partition.

zyxmon, 2022-12-06:
https://github.com/Entware/Entware/commit/3bebaea7854c824efb7896bf045873fce44b12f4

very20101, 2022-12-09:
> [Entware/Entware@3bebaea](https://github.com/Entware/Entware/commit/3bebaea7854c824efb7896bf045873fce44b12f4)

Thank you! it works.

closed 2022-12-09 by very20101