openwrt/openwrt #11385: Mt7621.dtsi error while building v21.02.3 target unifi 6 lite

tricksol, 2022-11-30:
Target = Unifi 6 Lite
Version = v21.02
Tag = v21.02.3

When building for this device on v21.02 it is now failing to build with these errors. v22.03 and relevant tags are fine. What more information would you like.

/dts/mt7621.dtsi:478.16-483.5: ERROR (phandle_references): /ethernet@1e100000/mac@1: Reference to non-existent node or label "macaddr_factory_e000"

make[5]: *** [Makefile:207:openwrt-21.02/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_gnubee_gb-pc2.dtb] Error2

neheb, 2022-11-30:
ping @arinc9

musashino205, 2022-11-30:
It seems to be caused by the commit https://github.com/openwrt/openwrt/commit/d604032c2a50fc6228bcaae78baee1048ff9af16 .
ramips/mt7621 in 21.02 branch hasn't been converted to NVMEM MAC address implementation and GB-PC2 doesn't have a node labeled as ```macaddr_factory_e000```, but the above commit adds ```&gmac1``` node with ```nvmem-cells = <&macaddr_factory_e000>;```.

arinc9, 2022-11-30:
@neheb @musashino205 Thanks for letting me know, I'll send a patch.

arinc9, 2022-11-30:
Patch out https://patchwork.ozlabs.org/project/openwrt/patch/20221130083100.9002-1-arinc.unal@arinc9.com/

tricksol, 2022-11-30:
Confirming that I can build again with this patch on v21.02.3 for target unifi 6 lite

Ansuel, 2022-12-31:
Fixed!

commit 76c9c2b1 (2022-12-31): rampis: fix Reference to non-existent node for GB-PC2

Fix cannot build: Reference to non-existent node or label
"macaddr_factory_e000" dtb compilation error.

The cherry-pick had to be reworked to use the old mtd-mac-address way as
openwrt-21.02 still wasn't migrated to nvmem implementation.

Fixes: d604032c2a50 ("ramips: fix GB-PC1 and GB-PC2 device support")
Fixes: #11654
Fixes: #11385

[ rework commit message, add more fixes tag ]
closed 2022-12-31 by Ansuel