openwrt/openwrt #11394: A wireless chip containing dual-band can only generate 5G configuration in OpenWrt, but cannot generate 2.4G configuration.

tangyf20142016, 2022-12-01:
There is only one WIFI chip on my router, but it contains dual-band(2.4G and 5G), but only 5G wireless configuration can be generated in OpenWRT. The wireless configuration of 2.4G cannot be generated. Because it is considered in /lib/wifi/mac80211.sh script: A PHY device can only correspond to a 2.4G or 5G band, and only get the information of the last frequency band in a PHY device through the "get_band_defaults" function, but there is only one PHY0 device in the /sys/class/ieee80211 of my router. The information of 2.4G and 5G can be obtained through the "iw phy phy0 info" command, as follows:
# iw phy phy0 info
Wiphy phy0
        Supported interface modes:
                 * AP
        Band 1:
                Capabilities: 
                Frequencies:
                        * 2412 MHz [1] (30.0 dBm)
                        * 2417 MHz [2] (30.0 dBm)
                        * 2422 MHz [3] (30.0 dBm)
                        * 2427 MHz [4] (30.0 dBm)
                        * 2432 MHz [5] (30.0 dBm)
                        * 2437 MHz [6] (30.0 dBm)
                        * 2442 MHz [7] (30.0 dBm)
                        * 2447 MHz [8] (30.0 dBm)
                        * 2452 MHz [9] (30.0 dBm)
                        * 2457 MHz [10] (30.0 dBm)
                        * 2462 MHz [11] (30.0 dBm)
                        * 2467 MHz [12] (30.0 dBm)
                        * 2472 MHz [13] (30.0 dBm)
                        * 2484 MHz [14] (30.0 dBm)
        Band 2:
                Capabilities: 
                VHT Capabilities:
                VHT RX MCS set:
                VHT RX highest supported: 0 Mbps
                VHT TX MCS set:
                VHT TX highest supported: 0 Mbps
                Frequencies:
                        * 5180 MHz [36] (30.0 dBm)
                        * 5200 MHz [40] (30.0 dBm)
                        * 5220 MHz [44] (30.0 dBm)
                        * 5240 MHz [48] (30.0 dBm)
                        * 5260 MHz [52] (30.0 dBm)
                        * 5280 MHz [56] (30.0 dBm)
                        * 5300 MHz [60] (30.0 dBm)
                        * 5320 MHz [64] (30.0 dBm)
                        * 5500 MHz [100] (30.0 dBm)
                        * 5520 MHz [104] (30.0 dBm)
                        * 5540 MHz [108] (30.0 dBm)
                        * 5560 MHz [112] (30.0 dBm)
                        * 5580 MHz [116] (30.0 dBm)
                        * 5600 MHz [120] (30.0 dBm)
                        * 5620 MHz [124] (30.0 dBm)
                        * 5640 MHz [128] (30.0 dBm)
                        * 5660 MHz [132] (30.0 dBm)
                        * 5680 MHz [136] (30.0 dBm)
                        * 5700 MHz [140] (30.0 dBm)
                        * 5720 MHz [144] (30.0 dBm)
                        * 5745 MHz [149] (30.0 dBm)
                        * 5765 MHz [153] (30.0 dBm)
                        * 5785 MHz [157] (30.0 dBm)
                        * 5805 MHz [161] (30.0 dBm)
                        * 5825 MHz [165] (30.0 dBm)
                        * 4920 MHz [184] (33.0 dBm)
                        * 4940 MHz [188] (33.0 dBm)
                        * 4960 MHz [192] (33.0 dBm)
                        * 4980 MHz [196] (33.0 dBm)

WiFi configuration:
# cat /etc/config/wireless 

config wifi-device 'radio0'
        option type 'mac80211'
        option macaddr '01:04:06:4e0:06:00'
        option channel '36'
        option band '5g'
        option htmode 'VHT80'
        option disabled '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'OpenWrt'
        option encryption 'none'

How can I generate configuration in OpenWrt when there is only one wireless chip but containing dual-band(2.4G and 5G)?
Thanks.

obsy, 2022-12-01:
One radio, one active band (frequency). Use 2.4GHz OR 5GHz.

jow-, 2022-12-01:
Use the forum or mailing lists for support questions.

closed 2022-12-01 by jow-

tangyf20142016, 2022-12-02:
> One radio, one active band (frequency). Use 2.4GHz OR 5GHz.

But I want to use 2.4G and 5G at the same time