openwrt/openwrt #11398: branch openwrt-22.03 build fail for raspberrypi 4b

ecrasy, 2022-12-02:
**Build Log:**
make[2]: Entering directory '/home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/scripts/config'
make[1]: Entering directory '/home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt'
make[2]: Entering directory '/home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/target/linux'
make[3]: Entering directory '/home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/target/linux/bcm27xx'
rm -f /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch[64](https://github.com/ecrasy/TestBuildOfficial/actions/runs/3588739640/jobs/6041073645#step:14:65)_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/localversion
/home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/scripts/kconfig.pl  + /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/target/linux/generic/config-5.10 /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/target/linux/bcm27xx/bcm2711/config-5.10 > /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/.config >> /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.target
/home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/scripts/package-metadata.pl kconfig /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/tmp/.packageinfo /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/.config 5.10 > /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.override
/home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/scripts/kconfig.pl 'm+' '+' /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.target /dev/null /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.override > /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.set
mv /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.set /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.old
grep -v INITRAMFS /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.old > /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.set
rm -rf /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/modules
cmp -s /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.set /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.prev || { cp /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.set /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config; cp /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.set /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.prev; }
export MAKEFLAGS= ; [ -d /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/user_headers ] || make -C /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156 KCFLAGS="-fmacro-prefix-map=/home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl=target-aarch64_cortex-a72_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Thu Dec  1 00:57:01 2022" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.10.156 CC="aarch64-openwrt-linux-musl-gcc" INSTALL_HDR_PATH=/home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/user_headers headers_install
grep '=[ym]' /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.config.set | LC_ALL=C sort | /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/staging_dir/host/bin/mkhash md5 > /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.vermagic
touch /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/.configured
rm -f /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/vmlinux /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156/System.map
make -C /home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-5.10.156 KCFLAGS="-fmacro-prefix-map=/home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl=target-aarch64_cortex-a72_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE= [... cut ...]
make[2]: Leaving directory '/home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/target/linux'
time: target/linux/compile#1.25#0.27#1.50
    ERROR: target/linux failed to build.
make[1]: *** [target/Makefile:30: target/linux/compile] Error 1
make[1]: Leaving directory '/home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt'
make: *** [/home/runner/work/TestBuildOfficial/TestBuildOfficial/workspace/openwrt/include/toplevel.mk:230: target/compile] Error 2
Error: Process completed with exit code 2.

stintel, 2022-12-14:
Interesting, this should have been fixed by 57634f472fbec519b5993d972056d6cc0a020875, which went in before the 5.10.156 kernel bump. Could you please let us know the exact version you are trying to build by posting the output of `scripts/getver.sh`?

dfateyev, 2022-12-14:
Also, 22.03 features 5.10.159 soon: https://github.com/openwrt/openwrt/pull/11516, you can check with it in advance

hauke, 2022-12-15:
This commit removed it again: 3564c22e46d5b025b174ee9b25291413a4621b21 We should probably fix it.

stintel, 2022-12-15:
Good catch. I fixed it in a different way in master afterwards, maybe better to backport that.

closed 2022-12-18 by ecrasy