openwrt/openwrt #11409: mt7921在22.03中开启ap异常

liuxichi, 2022-12-04:
mt7921在22.03开启ap用WIFI6 2.4和5g大概4米远距离有信号但无法传输，同样距离用WIFI5或WIFI4都没问题。请问这是什么原因?

981213, 2022-12-08:
麻烦开issue的时候换英文吧……这又不是个中国人主导的项目 你敲中文那些欧洲大佬们看不懂啊……

closed 2022-12-08 by 981213

liuxichi, 2022-12-08:
> 麻烦开issue的时候换英文吧……这又不是个中国人主导的项目 你敲中文那些欧洲大佬们看不懂啊……

给中国大佬看看就好😄，原因找到了是驱动。奇怪的是用最新的驱动在22.03.2下死活编译不过，只好用master了。有什么办法可以在22.03.2使用master中的mt76驱动呢？

OnlyCharacters, 2023-04-02:
@liuxichi 
I also have this problem with cf-953ax whose chipset is mt7921au.

The WIFI5 80MHz ap works fine. But when it comes to WIFI6, the AP only works in short distance, about 4~5m. If I go farther, it disconnects.
I notice that the rx rate of AP is low when I far away 4m from AP.
[image]
Have you solved this problem by using the master branch instead of 22.03 ?

我买了个 cf-952ax ，芯片为 mt7921au，也遇到这种情况。
WIFI5 情况下可用距离正常，但在 WIFI 6 情况下只能在 4~5 米内能用，走远了就不行了，热点的接受速率和带宽都很低。
用 master 分支编译能解决吗？

OnlyCharacters, 2023-04-02:
Everything works fine in master branch.

用 master 分支就好了。