openwrt/openwrt #11417: ath79 mtdblock5 too small in 22.03

odon-noda, 2022-12-04:
# Symptom
Lost 640 kB on / after upgrade from 21.02.3 to 22.03.2.
/dev/mtd5 size shrunk from 2048 kB to 1408 kB.

## Impact
 No space left on device during following
 \# opkg upgrade \`list-upgradable\` ... (libwolfssl5.5.3.ee39414e)

## Other terms
/dev/mtd5, mtdblock5, Flash layout

# Device
Same router
TP-Link Archer C6 v2 EU
Qualcomm Atheros QCA956X ver 1 rev 0
ath79/generic

# Steps to reproduce
```
21.02.3
# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/mtdblock5            2.0M    556.0K      1.5M  27% /overlay
overlayfs:/overlay        2.0M    556.0K      1.5M  27% /
```
```
22.03.2
# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/mtdblock5            1.4M    260.0K      1.1M  18% /overlay
overlayfs:/overlay        1.4M    260.0K      1.1M  18% /
```
# Analysis

| Table | **mtdblock5 size** | |
| ------- | ------ | ------ |
| Version | mtdblock5 | hex |
| 21.02.3 | 2048 | 0x800 |
| 22.03.0 | 1408 | 0x580 | 
| 22.03.2 | 1408 | 0x580 |
| 22.03 SNAPSHOT | 1536 | 0x600 |

```
-----------------------------------------------------
OpenWrt 21.02.3, r16554-1d4dea6d4f
-----------------------------------------------------
# cat /proc/partitions
major minor  #blocks  name

  31        0        128 mtdblock0
  31        1         64 mtdblock1
  31        2       7808 mtdblock2
  31        3       2001 mtdblock3
  31        4       5806 mtdblock4
  31        5       2048 mtdblock5
  31        6        128 mtdblock6
  31        7         64 mtdblock7
#
```
```
----------------------------------------------------
OpenWrt 22.03.0, r19685-512e76967f
-----------------------------------------------------
# cat  /proc/partitions
major minor  #blocks  name

  31        0        128 mtdblock0
  31        1         64 mtdblock1
  31        2       7808 mtdblock2
  31        3       2191 mtdblock3
  31        4       5616 mtdblock4
  31        5       1408 mtdblock5
  31        6        128 mtdblock6
  31        7         64 mtdblock7
#
```
```
-----------------------------------------------------
OpenWrt 22.03.2, r19803-9a599fee93
-----------------------------------------------------
# cat  /proc/partitions
major minor  #blocks  name

  31        0        128 mtdblock0
  31        1         64 mtdblock1
  31        2       7808 mtdblock2
  31        3       2191 mtdblock3
  31        4       5616 mtdblock4
  31        5       1408 mtdblock5
  31        6        128 mtdblock6
  31        7         64 mtdblock7
#
```
```
-----------------------------------------------------
OpenWrt SNAPSHOT, r21374-366bcffa0e
-----------------------------------------------------
# cat  /proc/partitions
major minor  #blocks  name

  31        0        128 mtdblock0
  31        1         64 mtdblock1
  31        2       7808 mtdblock2
  31        3       2440 mtdblock3
  31        4       5367 mtdblock4
  31        5       1536 mtdblock5
  31        6        128 mtdblock6
  31        7         64 mtdblock7
#
```

## Most likely reason: Flash layout mtdblock5 length has been shortened

odon-noda, 2022-12-04:
21.02.5 ate 64 kB from mtdblock5 compared to 21.02.3
```
-----------------------------------------------------
OpenWrt 21.02.5, r16688-fa9a932fdb
-----------------------------------------------------
# cat /proc/partitions
major minor  #blocks  name

  31        0        128 mtdblock0
  31        1         64 mtdblock1
  31        2       7808 mtdblock2
  31        3       2015 mtdblock3
  31        4       5792 mtdblock4
  31        5       1984 mtdblock5
  31        6        128 mtdblock6
  31        7         64 mtdblock7
# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/mtdblock5            1.9M    264.0K      1.7M  13% /overlay
overlayfs:/overlay        1.9M    264.0K      1.7M  13% /
#
```

| Table | **mtdblock5 size** | |
| ------- | ------ | ------ |
| Version | mtdblock5 | hex |
| 21.02.3 | 2048 | 0x800 |
| 21.02.5 | 1984 | 0x7C0 |
| 22.03.0 | 1408 | 0x580 | 
| 22.03.2 | 1408 | 0x580 |
| 22.03 SNAPSHOT | 1536 | 0x600 |

brada4, 2022-12-05:
https://openwrt.org/supported_devices/864_warning
With 22.03 you can reduce rootfs by removing some packages before compiling compact kernel, like no luci web but still up-to-date access point, but it is approaching end of the road, no less.

odon-noda, 2022-12-13:
Hi @brada4 
Thank you for you fast reaction, RCA and valuable advice.
I updated [Archer C6 V2 Flash Layout](https://openwrt.org/toh/tp-link/archer_c6_v2#flash_layout) and summarized this conclusion

# Root cause
Firmware size increases naturally with every new release.
[https://openwrt.org/supported_devices/864_warning#analysis_of_firmware_size_growth](https://openwrt.org/supported_devices/864_warning#analysis_of_firmware_size_growth)
Hence #11417 is not an issue, this is the feature.
# Solution

## Option 1 - standard image only
Use only master distribution (e.g. 22.03.2) without additional packages
- Pros
simple
luci GUI available
- Cons
mid-term "approaching end of the road"
only standard, limited functionality

## Option 2 - snapshot only
Use only snapshot distribution ```auc```
- Pros
simple
more space for additional packages
- Cons
no luci GUI 

## Option 3 - custom build
compiling compact image
- Pros
only required essential functions
long-term 
- Cons
demanding

## Option 4 - HW replacement
buy a new router
- Pros
...
- Cons
lost investment (flash ROM lifetime ~11 years)
higher power consumption (Archer C6 V2 has 3.06W)

brada4, 2022-12-13:
1.5 You can upgrade standard image components with auc.

odon-noda, 2022-12-13:
Oh @brada4 
Finally! I got it.
Option 1 = standard upgrade + auc is the right choose
```
BEFORE = 21.02.3
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 3.8M      3.8M         0 100% /rom
/dev/mtdblock5            2.0M      1.9M    144.0K  93% /overlay

AFTER = 22.03.2 + auc
root@a20:~# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                 4.3M      4.3M         0 100% /rom
/dev/mtdblock5            1.4M    268.0K      1.1M  19% /overlay
```

brada4, 2022-12-13:
it is even less scary - you dont need opkg upgrade anymore. auc server builds image from latest versions of packages for that release.