openwrt/openwrt #11459: WAN port frequently down on Linksys WRT1900AC v1 (Mamba)

e88z4, 2022-12-09:
I am currently having issue with Openwrt 22.03.2. The WAN port keeps disconnecting all the time with the following error in dmesg.

`[  616.744003] mv88e6085 f1072004.mdio-mii:00 wan: Link is Down
[  619.742993] mv88e6085 f1072004.mdio-mii:00 wan: Link is Up - 1Gbps/Full - flow control rx/tx
[  931.460832] mv88e6085 f1072004.mdio-mii:00 wan: Link is Down
[  934.373597] mv88e6085 f1072004.mdio-mii:00 wan: Link is Up - 1Gbps/Full - flow control rx/tx
[ 1121.443603] mv88e6085 f1072004.mdio-mii:00 wan: Link is Down
[ 1124.306126] mv88e6085 f1072004.mdio-mii:00 wan: Link is Up - 1Gbps/Full - flow control rx/tx`

`Fri Dec  9 16:05:52 2022 kern.info kernel: [ 1121.443603] mv88e6085 f1072004.mdio-mii:00 wan: Link is Down
Fri Dec  9 16:05:52 2022 daemon.notice netifd: wan (9313): udhcpc: received SIGTERM
Fri Dec  9 16:05:52 2022 daemon.notice netifd: wan (9313): udhcpc: unicasting a release of 198.52.165.221 to 7.127.36.137
Fri Dec  9 16:05:52 2022 daemon.notice netifd: wan (9313): udhcpc: sending release
Fri Dec  9 16:05:52 2022 daemon.notice netifd: wan (9313): udhcpc: entering released state
Fri Dec  9 16:05:52 2022 daemon.notice netifd: wan (9313): Command failed: ubus call network.interface notify_proto { "action": 0, "link-up": false, "keep": false, "interface": "wan" } (Permission denied)
Fri Dec  9 16:05:52 2022 daemon.notice netifd: Interface 'wan' is now down
Fri Dec  9 16:05:52 2022 daemon.warn dnsmasq[1]: no servers found in /tmp/resolv.conf.d/resolv.conf.auto, will retry
Fri Dec  9 16:05:52 2022 daemon.notice netifd: Interface 'wan6' is now down
Fri Dec  9 16:05:52 2022 user.notice ddns-scripts[9430]: duckdns: PID '9430' terminated by 'SIGTERM' at 2022-12-09 16:05
Fri Dec  9 16:05:54 2022 daemon.notice netifd: Network device 'wan' link is up
Fri Dec  9 16:05:54 2022 daemon.notice netifd: Interface 'wan' has link connectivity
Fri Dec  9 16:05:54 2022 daemon.notice netifd: Interface 'wan' is setting up now
Fri Dec  9 16:05:54 2022 daemon.notice netifd: Interface 'wan6' has link connectivity
Fri Dec  9 16:05:54 2022 daemon.notice netifd: Interface 'wan6' is setting up now
Fri Dec  9 16:05:54 2022 kern.info kernel: [ 1124.306126] mv88e6085 f1072004.mdio-mii:00 wan: Link is Up - 1Gbps/Full - flow control rx/tx
Fri Dec  9 16:05:55 2022 daemon.notice netifd: wan (9905): udhcpc: started, v1.35.0
Fri Dec  9 16:05:55 2022 daemon.notice netifd: wan (9905): udhcpc: broadcasting discover
Fri Dec  9 16:05:55 2022 daemon.notice netifd: wan (9905): udhcpc: broadcasting select for 198.52.165.221, server 7.127.36.137
Fri Dec  9 16:05:55 2022 daemon.notice netifd: wan (9905): udhcpc: lease of 198.52.165.221 obtained from 7.127.36.137, lease time 151547
Fri Dec  9 16:05:55 2022 daemon.notice netifd: Interface 'wan' is now up`

I connected my laptop directly to my cable modem, the connection is stable for hours. I also switched a few ethernet cables as a process of elimination. When I use the Linksys WRT1900AC router, the WAN keeps disconnected 24/7 every few minutes. Is this a bug in the current stable release?

e88z4, 2022-12-09:
Sorry for the log appearing in one line. I am not sure how to make the log appears in multiple lines. 

Just some background on this. I recently upgraded from 19.07 SNAPSHOT build. I didn't have any issue with this snapshot version.I doubt it is a hardware related issue.

Spudz76, 2022-12-09:
use triple-backticks instead of just one

\`\`\`
```
multi
line
stuff
```
\`\`\`

e88z4, 2022-12-10:
Hi @anomeome , 

I saw this particular pull request that may fix the issue.
https://github.com/openwrt/openwrt/pull/11399

I tried kernel 5.15.82 from the master branch. The issue persists. I called my ISP as well, technician came to check the line. The technician didn't find any issue with the signal. The modem web UI interface log didn't indicate any issue.

anomeome, 2022-12-11:
Not seeing any issue on _mamba_ with:

```
root@OpenWrt:/etc# cat openwrt_version
r21452-1e240f60a5
root@OpenWrt:/etc# uname -a
Linux OpenWrt 5.15.82 #0 SMP Sun Dec 11 17:32:53 2022 armv7l GNU/Linux
```
don't know if you used a buildbot snapshot but that PR was backed out due to buildbot image issue on bridge setup. Guessing this may be local to your particular setup / config though.

e88z4, 2022-12-12:
I noticed the pull request was reverted so I created my own local branch to force kernel 5.15.82. Issue persists with 22.03 SNAPSHOT that I built myself. 

I am currently back to 19.07.10. This old stable build is rock solid so far. The WAN interface has been stable with no disconnect more than 2 hours+. With 22.03.02, it won't even last 15 minutes before being up and down so frequently.

e88z4, 2022-12-12:
I wonder if this has anything to do with the new DSA architecture. The old stable build 19.07.10 is on swconfig instead.

e88z4, 2022-12-20:
I was to resolve the issue by disabling the energy efficient ethernet option using ethtool. After the option is disabled, the WAN interface is stable. I make it permanent by creating the following script in /etc/hotplug.d/iface/10-ethtool

```
[ ifup = "$ACTION" ] || exit 0
[ -n "$DEVICE" ] || exit 0

if [ "$DEVICE" == wan ] ; then
      ethtool --set-eee "$DEVICE" eee off
fi
```

closed 2022-12-20 by e88z4

JacobBrownAustin, 2025-04-07:
I first saw this problem years ago, and found a workaround by using a crappier cable with a lower rating.  But a week ago, it started happening again.  I'm glad I found this workaround by disabling EEE.  I wish I didn't have to disable energy efficiency to have stable network, as it sits in a closet that can get quite warm, but without this workaround, it's such a pain getting random network drops every few minutes.