openwrt/openwrt #11466: PHICOMM K3 wireless not working

ningnao, 2022-12-10:
Probably because of this commit [https://github.com/openwrt/openwrt/commit/6603748e0ca697f484adfc16a0d49d35ed36954d](url)
change the Interface name to "wlan0" or "wlan1" works fine

```
Sat Dec 10 17:49:25 2022 daemon.notice netifd: Wireless device 'radio1' is now up
Sat Dec 10 17:49:27 2022 kern.info kernel: [  478.725297] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
Sat Dec 10 17:49:27 2022 kern.info kernel: [  478.732087] br-lan: port 5(phy1-ap0) entered blocking state
Sat Dec 10 17:49:27 2022 kern.info kernel: [  478.737710] br-lan: port 5(phy1-ap0) entered forwarding state
Sat Dec 10 17:49:27 2022 daemon.notice netifd: Network device 'phy1-ap0' link is up
Sat Dec 10 17:49:27 2022 kern.err kernel: [  478.745089] ieee80211 phy1: brcmf_cfg80211_add_iface: iface validation failed: err=-16
Sat Dec 10 17:49:27 2022 daemon.err hostapd: Failed to create interface phy1-ap1: -16 (Resource busy)
Sat Dec 10 17:49:27 2022 daemon.err hostapd: Failed to add BSS (BSSID=2e:b2:1a:5c:8b:0f)
Sat Dec 10 17:49:27 2022 daemon.warn hostapd: phy1-ap1: Could not connect to kernel driver
Sat Dec 10 17:49:27 2022 daemon.warn hostapd: phy1-ap0: Could not connect to kernel driver
Sat Dec 10 17:49:27 2022 daemon.err hostapd: Interface initialization failed
Sat Dec 10 17:49:27 2022 daemon.notice hostapd: phy1-ap0: interface state HT_SCAN->DISABLED
Sat Dec 10 17:49:27 2022 daemon.notice hostapd: phy1-ap0: AP-DISABLED
Sat Dec 10 17:49:41 2022 daemon.notice hostapd: Remove interface 'phy1-ap0'
Sat Dec 10 17:49:41 2022 daemon.notice hostapd: phy1-ap0: interface state DISABLED->DISABLED
Sat Dec 10 17:49:41 2022 daemon.notice hostapd: phy1-ap1: AP-DISABLED
Sat Dec 10 17:49:41 2022 daemon.notice hostapd: phy1-ap1: CTRL-EVENT-TERMINATING
Sat Dec 10 17:49:41 2022 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sat Dec 10 17:49:41 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface phy1-ap1 wasn't started
Sat Dec 10 17:49:41 2022 daemon.notice hostapd: phy1-ap0: AP-DISABLED
Sat Dec 10 17:49:41 2022 daemon.notice hostapd: phy1-ap0: CTRL-EVENT-TERMINATING
Sat Dec 10 17:49:41 2022 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sat Dec 10 17:49:41 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface phy1-ap0 wasn't started
Sat Dec 10 17:49:41 2022 daemon.notice hostapd: nl80211: deinit ifname=phy1-ap0 disabled_11b_rates=0
Sat Dec 10 17:49:41 2022 kern.info kernel: [  492.425451] device phy1-ap0 left promiscuous mode
Sat Dec 10 17:49:41 2022 kern.info kernel: [  492.430335] br-lan: port 5(phy1-ap0) entered disabled state
Sat Dec 10 17:49:42 2022 daemon.notice netifd: Network device 'phy1-ap0' link is down
Sat Dec 10 17:49:42 2022 kern.err kernel: [  493.469983] ieee80211 phy1: brcmf_cfg80211_get_channel: chanspec failed (-52)
Sat Dec 10 17:49:42 2022 daemon.notice netifd: Wireless device 'radio1' is now down
Sat Dec 10 17:49:42 2022 daemon.notice netifd: radio1 (17661): command failed: No error information (-524)
Sat Dec 10 17:49:42 2022 daemon.notice netifd: radio1 (17661): command failed: I/O error (-5)
Sat Dec 10 17:49:42 2022 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy phy1-ap0) --> new PHY
Sat Dec 10 17:49:42 2022 kern.info kernel: [  493.903579] br-lan: port 5(phy1-ap0) entered blocking state
Sat Dec 10 17:49:42 2022 kern.info kernel: [  493.909286] br-lan: port 5(phy1-ap0) entered disabled state
Sat Dec 10 17:49:42 2022 kern.info kernel: [  493.915106] device phy1-ap0 entered promiscuous mode
Sat Dec 10 17:49:42 2022 kern.info kernel: [  493.920356] br-lan: port 5(phy1-ap0) entered blocking state
Sat Dec 10 17:49:42 2022 kern.info kernel: [  493.926037] br-lan: port 5(phy1-ap0) entered forwarding state
Sat Dec 10 17:49:42 2022 daemon.notice hostapd: phy1-ap0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sat Dec 10 17:49:42 2022 daemon.notice hostapd: phy1-ap0: interface state COUNTRY_UPDATE->HT_SCAN
Sat Dec 10 17:49:43 2022 kern.info kernel: [  494.465455] br-lan: port 5(phy1-ap0) entered disabled state
Sat Dec 10 17:49:43 2022 daemon.info dnsmasq[1]: read /etc/hosts - 4 addresses
Sat Dec 10 17:49:43 2022 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 addresses
Sat Dec 10 17:49:43 2022 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Sat Dec 10 17:49:43 2022 daemon.notice netifd: Wireless device 'radio1' is now up
```

ningnao, 2022-12-15:
@nbd168  
I revert tihs commit https://github.com/openwrt/openwrt/commit/6603748e0ca697f484adfc16a0d49d35ed36954d  
it works

hauke, 2022-12-17:
Did you check why this is failing?
````
Sat Dec 10 17:49:27 2022 daemon.notice netifd: Network device 'phy1-ap0' link is up
Sat Dec 10 17:49:27 2022 kern.err kernel: [  478.745089] ieee80211 phy1: brcmf_cfg80211_add_iface: iface validation failed: err=-16
Sat Dec 10 17:49:27 2022 daemon.err hostapd: Failed to create interface phy1-ap1: -16 (Resource busy)
````

`brcmf_cfg80211_add_iface()` is not directly checking the interface name.

ningnao, 2022-12-18:
@hauke 

Thanks for your reply
No matter how you set it up, it will always generate a wireless interface named wlan0

```
phy#0
	Interface phy0-ap0
		ifindex 16
		wdev 0x3
		addr 2c:b2:1a:5c:8b:0d
		ssid xxxxx
		type AP
		channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
		txpower 31.00 dBm
	Interface wlan0
		ifindex 9
		wdev 0x1
		addr 2c:b2:1a:5c:8b:0d
		type managed
		channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
```

painfull30, 2023-06-13:
For GL.iNet GL-X300B on openwrt22 i had no SSIDs time2time and only manually added patch [6603748] helped (https://github.com/openwrt/openwrt/commit/6603748e0ca697f484adfc16a0d49d35ed36954d)

predators46, 2023-06-30:
@hauke 

i have almost the same problem on openwrt 23.05.0-rc2 using rtl8189fs. i tried on openwrt 21 and 22 no problem

```
Thu Jun 29 17:18:23 2023 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Thu Jun 29 17:18:23 2023 kern.info kernel: [   23.776272] rtl8189fs mmc0:0001:1 phy0-ap0: renamed from wlan0
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769): Failed to create interface phy1-ap0
Thu Jun 29 17:18:23 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy phy1-ap0) --> new PHY
Thu Jun 29 17:18:23 2023 daemon.err hostapd: Could not read interface phy1-ap0 flags: No such device
Thu Jun 29 17:18:23 2023 daemon.notice hostapd: nl80211: Driver does not support authentication/association or connect commands
Thu Jun 29 17:18:23 2023 daemon.notice hostapd: nl80211: deinit ifname=phy1-ap0 disabled_11b_rates=0
Thu Jun 29 17:18:23 2023 daemon.err hostapd: Could not read interface phy1-ap0 flags: No such device
Thu Jun 29 17:18:23 2023 daemon.err hostapd: nl80211 driver initialization failed.
Thu Jun 29 17:18:23 2023 daemon.notice hostapd: phy1-ap0: CTRL-EVENT-TERMINATING
Thu Jun 29 17:18:23 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface phy1-ap0 wasn't started
Thu Jun 29 17:18:23 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy phy0-ap0) --> new PHY
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769): Command failed: ubus call hostapd config_add {"iface":"phy1-ap0", "config":"/var/run/hostapd-phy1.conf"} (Invalid argument)
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769): Usage: ubus [<options>] <command> [arguments...]
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769): Options:
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769):  -s <socket>:		Set the unix domain socket to connect to
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769):  -t <timeout>:		Set the timeout (in seconds) for a command to complete
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769):  -S:			Use simplified output (for scripts)
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769):  -v:			More verbose output
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769):  -m <type>:		(for monitor): include a specific message type
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769): 			(can be used more than once)
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769):  -M <r|t>		(for monitor): only capture received or transmitted traffic
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769):
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769): Commands:
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769):  - list [<path>]			List objects
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769):  - call <path> <method> [<message>]	Call an object method
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769):  - subscribe <path> [<path>...]	Subscribe to object(s) notifications
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769):  - listen [<path>...]			Listen for events
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769):  - send <type> [<message>]		Send an event
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769):  - wait_for <object> [<object>...]	Wait for multiple objects to appear on ubus
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769):  - monitor				Monitor ubus traffic
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769):
Thu Jun 29 17:18:23 2023 daemon.notice netifd: radio1 (2769): Device setup failed: HOSTAPD_START_FAILED
Thu Jun 29 17:18:23 2023 daemon.notice netifd: Wireless device 'radio1' set retry=0
Thu Jun 29 17:18:23 2023 daemon.crit netifd: Wireless device 'radio1' setup failed, retry=0
Thu Jun 29 17:18:23 2023 daemon.notice netifd: Wireless device 'radio1' is now down
```

predators46, 2023-06-30:
@nbd168 

this change is what breaks it

https://github.com/openwrt/openwrt/commit/6603748e0ca697f484adfc16a0d49d35ed36954d

ishanjain28, 2023-08-06:
Hi, This is also affecting 5Ghz Wi-Fi on Archer A6.

```
NAME="OpenWrt"
VERSION="22.03.4"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 22.03.4"
VERSION_ID="22.03.4"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r20123-38ccc47687"
OPENWRT_BOARD="ramips/mt7621"
OPENWRT_ARCH="mipsel_24kc"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 22.03.4 r20123-38ccc47687"

Sun Aug  6 22:10:25 2023 daemon.notice hostapd: Remove interface 'wlan1'
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: wlan1: interface state DISABLED->DISABLED
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: wlan1-3: AP-DISABLED
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: wlan1-3: CTRL-EVENT-TERMINATING
Sun Aug  6 22:10:25 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sun Aug  6 22:10:25 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan1-3 wasn't started
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: wlan1-2: AP-DISABLED
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: wlan1-2: CTRL-EVENT-TERMINATING
Sun Aug  6 22:10:25 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sun Aug  6 22:10:25 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan1-2 wasn't started
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: wlan1-1: AP-DISABLED
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: wlan1-1: CTRL-EVENT-TERMINATING
Sun Aug  6 22:10:25 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sun Aug  6 22:10:25 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan1-1 wasn't started
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: wlan1: AP-DISABLED
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: wlan1: CTRL-EVENT-TERMINATING
Sun Aug  6 22:10:25 2023 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sun Aug  6 22:10:25 2023 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan1 wasn't started
Sun Aug  6 22:10:25 2023 daemon.notice hostapd: nl80211: deinit ifname=wlan1 disabled_11b_rates=0
Sun Aug  6 22:10:25 2023 kern.info kernel: [735120.265048] device wlan1 left promiscuous mode
Sun Aug  6 22:10:25 2023 kern.info kernel: [735120.270247] br-lan: port 6(wlan1) entered disabled state
Sun Aug  6 22:10:25 2023 daemon.notice netifd: Wireless device 'radio1' is now down
Sun Aug  6 22:10:26 2023 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy1.conf (phy wlan1) --> new PHY
Sun Aug  6 22:10:27 2023 kern.info kernel: [735122.289052] br-lan: port 6(wlan1) entered blocking state
Sun Aug  6 22:10:27 2023 kern.info kernel: [735122.294546] br-lan: port 6(wlan1) entered disabled state
Sun Aug  6 22:10:27 2023 kern.info kernel: [735122.300743] device wlan1 entered promiscuous mode
Sun Aug  6 22:10:27 2023 daemon.notice hostapd: wlan1: interface state UNINITIALIZED->COUNTRY_UPDATE
Sun Aug  6 22:10:27 2023 daemon.notice hostapd: ACS: Automatic channel selection started, this may take a bit
Sun Aug  6 22:10:27 2023 daemon.notice hostapd: wlan1: interface state COUNTRY_UPDATE->ACS
Sun Aug  6 22:10:27 2023 daemon.notice hostapd: wlan1: ACS-STARTED
Sun Aug  6 22:10:28 2023 daemon.notice netifd: Wireless device 'radio1' is now up
Sun Aug  6 22:10:47 2023 daemon.notice hostapd: wlan1: ACS-COMPLETED freq=5260 channel=52
Sun Aug  6 22:10:47 2023 daemon.notice hostapd: wlan1: interface state ACS->HT_SCAN
Sun Aug  6 22:10:48 2023 daemon.notice hostapd: wlan1: interface state HT_SCAN->DFS
Sun Aug  6 22:10:48 2023 daemon.notice hostapd: wlan1: DFS-CAC-START freq=5260 chan=52 sec_chan=1, width=1, seg0=58, seg1=0, cac_time=60s
Sun Aug  6 22:10:48 2023 daemon.err hostapd: DFS start_dfs_cac() failed, -1
Sun Aug  6 22:10:48 2023 daemon.err hostapd: Interface initialization failed
Sun Aug  6 22:10:48 2023 daemon.notice hostapd: wlan1: interface state DFS->DISABLED
Sun Aug  6 22:10:48 2023 daemon.notice hostapd: wlan1: AP-DISABLED
```

ptpt52, 2023-10-08:
any progress?

predators46, 2023-10-19:
bad news for rtl8189fs after updating mac80211 in openwrt 23.05.0 rtl8189fs can no longer be used. changes in mac80211.sh must be made manually

Lenzelot, 2025-06-26:
After wondering why my 5 GHz interface wouldn't start (also the `daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied`) problem, giving it a different interface name using LuCi did the trick. Thanks OP!

Netgear Nighthawk X4S R7800 on OpenWrt 24.10.0