openwrt/openwrt #11468: Edgerouter `er` doesn't save settings on sysupgrade

532910, 2022-12-10:


stintel, 2022-12-14:
If possible, try the sysupgrade via serial console and attach the output from starting sysupgrade until boot finishes to this issue. If this isn't possible, post the output of logread after boot finishes.

532910, 2022-12-15:
I do sysupgrade via serial console.
As you can see `er` is not handled in https://github.com/openwrt/openwrt/blob/master/target/linux/octeon/base-files/lib/upgrade/platform.sh
Adding it to the list causes `sysupgrade.tgz` be placed on the kernel partition, but it's ignored at boot time:

```
--- /tmp/platform.sh	2022-12-15 16:22:44.598200283 +0300
+++ files/platform.sh	2022-12-07 03:12:44.793900708 +0300
@@ -27,13 +27,15 @@
 
 platform_copy_config() {
 	case "$(board_name)" in
-	erlite|\
-	ubnt,usg)
+	erlite)
 		platform_copy_config_helper /dev/sda1
 		;;
 	itus,shield-router)
 		platform_copy_config_helper /dev/mmcblk1p1
 		;;
+	er)
+		emmc_copy_config
+		;;
 	ubnt,edgerouter-4|\
 	ubnt,edgerouter-6p)
 		platform_copy_config_helper /dev/mmcblk0p1
@@ -87,13 +89,14 @@
 
 	[ -b "${rootfs}" ] || return 1
 	case "$board" in
-	er | \
+	er)
+		emmc_do_upgrade "$tar_file"
+		;;
 	ubnt,edgerouter-4 | \
 	ubnt,edgerouter-6p)
 		kernel=mmcblk0p1
 		;;
-	erlite|\
-	ubnt,usg)
+	erlite)
 		kernel=sda1
 		;;
 	itus,shield-router)
@@ -121,8 +124,7 @@
 	erlite | \
 	itus,shield-router | \
 	ubnt,edgerouter-4 | \
-	ubnt,edgerouter-6p | \
-	ubnt,usg)
+	ubnt,edgerouter-6p)
 		local kernel_length=$(tar xf $tar_file $board_dir/kernel -O | wc -c 2> /dev/null)
 		local rootfs_length=$(tar xf $tar_file $board_dir/root -O | wc -c 2> /dev/null)
 		[ "$kernel_length" = 0 -o "$rootfs_length" = 0 ] && {
```

closed 2023-10-21 by 532910