openwrt/openwrt #11469: Nanopi R1 gets no ip with squash and ext4 22.03.2 images

linxcow, 2022-12-11:
Device : Nanopi R1 512MB ram H3 no emmc 
the system boots and gives ip with an older version of friendlywrt and works thereafter.

with the same SD card and both images as below i never get ip 

- openwrt-22.03.2-53b24516c538-sunxi-cortexa7-friendlyarm_nanopi-r1-squashfs-sdcard.img.gz
- openwrt-22.03.2-sunxi-cortexa7-friendlyarm_nanopi-r1-ext4-sdcard.img.gz

Below is a debug dump using the serial uart running the ext4 version:

```
[  574.755587] usb 1
U-Boot SPL 2020.04-OpenWrt-r19803-9a599fee93 (Oct 14 2022 - 22:44:41 +0000)
DRAM: 512 MiB
Trying to boot from MMC1

U-Boot 2020.04-OpenWrt-r19803-9a599fee93 (Oct 14 2022 - 22:44:41 +0000) Allwinner Technology

CPU:   Allwinner H3 (SUN8I 1680)
Model: FriendlyARM NanoPi R1
DRAM:  512 MiB
MMC:   Device 'mmc@1c11000': seq 1 is in use by 'mmc@1c10000'
mmc@1c0f000: 0, mmc@1c10000: 2, mmc@1c11000: 1
Loading Environment from FAT... Card did not respond to voltage select!
In:    serial
Out:   serial
Err:   serial
Net:   phy interface7

Error: ethernet@1c30000 address not set.
No ethernet found.

starting USB...
Bus usb@1c1b000: USB EHCI 1.00
Bus usb@1c1b400: USB OHCI 1.0
Bus usb@1c1c000: USB EHCI 1.00
Bus usb@1c1c400: USB OHCI 1.0
Bus usb@1c1d000: USB EHCI 1.00
Bus usb@1c1d400: USB OHCI 1.0
scanning bus usb@1c1b000 for devices... 2 USB Device(s) found
scanning bus usb@1c1b400 for devices... 1 USB Device(s) found
scanning bus usb@1c1c000 for devices... 1 USB Device(s) found
scanning bus usb@1c1c400 for devices... 1 USB Device(s) found
scanning bus usb@1c1d000 for devices... 1 USB Device(s) found
scanning bus usb@1c1d400 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
377 bytes read in 2 ms (183.6 KiB/s)
## Executing script at 43100000
3359912 bytes read in 252 ms (12.7 MiB/s)
23066 bytes read in 5 ms (4.4 MiB/s)
## Booting kernel from Legacy Image at 42000000 ...
   Image Name:   ARM OpenWrt Linux-5.10.146
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    3359848 Bytes = 3.2 MiB
   Load Address: 40008000
   Entry Point:  40008000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 43000000
   Booting using the fdt blob at 0x43000000
   Loading Kernel Image
   Using Device Tree in place at 43000000, end 43008a19

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[... 206 lines cut ...]
[    1.781231] dwmac-sun8i 1c30000.ethernet: Current syscon value is not the default 50006 (expect 58000)
[    1.791283] mmcblk1: mmc1:aaaa SU16G 14.8 GiB 
[    1.798532] mmc2: queuing unknown CIS tuple 0x80 (7 bytes)
[    1.803888]  mmcblk1: p1 p2
[    1.810906] mmc2: queuing unknown CIS tuple 0x81 (9 bytes)
[    1.903980] dwmac-sun8i 1c30000.ethernet: No HW DMA feature register supported
[    1.911236] dwmac-sun8i 1c30000.ethernet: RX Checksum Offload Engine supported
[    1.918536] dwmac-sun8i 1c30000.ethernet: COE Type 2
[    1.923550] dwmac-sun8i 1c30000.ethernet: TX Checksum insertion supported
[    1.930354] dwmac-sun8i 1c30000.ethernet: Normal descriptors
[    1.936050] dwmac-sun8i 1c30000.ethernet: Chain mode enabled
[    1.941775] dwmac-sun8i 1c30000.ethernet: device MAC address 5e:3b:e5:c7:40:12
[    1.952960] dwmac-sun8i 1c30000.ethernet: Found internal PHY node
[    1.960149] dwmac-sun8i 1c30000.ethernet: Switch mux to internal PHY
[    1.966582] dwmac-sun8i 1c30000.ethernet: Powering internal PHY
[    1.974768] dwmac-sun8i 1c30000.ethernet: Switch mux to external PHY
[    1.989826] mmc2: new high speed SDIO card at address 0001
[    2.000948] cpufreq: cpufreq_online: CPU0: Running at unlisted initial frequency: 480000 KHz, changing to: 648000 KHz
[    2.022530] ALSA device list:
[    2.025561]   No soundcards found.
[    2.029413] Waiting for root device /dev/mmcblk0p2...
[   32.483504] vcc3v0: disabling
[   32.486485] vcc5v0: disabling

```

AiyionPrime, 2022-12-12:
@linxcow already found it, but for others coming along:
This looks like the second problem of #11104.

stintel, 2022-12-22:
Dupe of #10080.

closed 2022-12-22 by stintel