openwrt/openwrt #11477: wifi crash on mt7612

ptpt52, 2022-12-12:
test and crash found on mir3 and mir3g devices

```
<6>[   81.754880] br-lan: port 2(phy1-ap0) entered blocking state
<6>[   81.760613] br-lan: port 2(phy1-ap0) entered disabled state
<6>[   81.766816] device phy1-ap0 entered promiscuous mode
<4>[   82.617447] {natflow}:natflow_netdev_event(): catch NETDEV_UP event for dev=phy0-ap0(tso=0,0,hw_csum=0,0), add ingress hook
<6>[   82.708659] br-lan: port 3(phy0-ap0) entered blocking state
<6>[   82.714461] br-lan: port 3(phy0-ap0) entered disabled state
<6>[   82.720641] device phy0-ap0 entered promiscuous mode
<4>[   84.366528] {natflow}:natflow_netdev_event(): catch NETDEV_UP event for dev=phy0-sta0(tso=0,0,hw_csum=0,0), add ingress hook
<6>[   91.439272] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
<6>[   91.446369] br-lan: port 2(phy1-ap0) entered blocking state
<6>[   91.452134] br-lan: port 2(phy1-ap0) entered forwarding state
<6>[   93.051306] phy0-sta0: authenticate with cc:b1:82:7e:1e:d0
<6>[   93.465289] phy0-sta0: send auth to cc:b1:82:7e:1e:d0 (try 1/3)
<6>[   93.580971] phy0-sta0: send auth to cc:b1:82:7e:1e:d0 (try 2/3)
<6>[   93.611079] phy0-sta0: authenticated
<6>[   95.880578] phy0-sta0: associate with cc:b1:82:7e:1e:d0 (try 1/3)
<6>[   95.916080] phy0-sta0: RX AssocResp from cc:b1:82:7e:1e:d0 (capab=0x1511 status=0 aid=33)
<6>[   95.924793] phy0-sta0: associated
<7>[   95.928397] phy0-sta0: Limiting TX power to 23 (23 - 0) dBm as advertised by cc:b1:82:7e:1e:d0
<6>[   96.919916] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-sta0: link becomes ready
<1>[  104.662920] CPU 0 Unable to handle kernel paging request at virtual address 00000068, epc == 82b2e24c, ra == 82b3040c
<4>[  104.673859] Oops[#1]:
<4>[  104.676183] CPU: 0 PID: 1063 Comm: napi/phy0-3 Not tainted 5.15.82 #0
<4>[  104.682763] $ 0   : 00000000 808b0000 00000000 00000000
<4>[  104.688124] $ 4   : 82d304a0 8297fa4e 00000000 00026e6e
<4>[  104.693474] $ 8   : 0000013c 00000000 00000000 488454dc
<4>[  104.698825] $12   : 8297fa4e 00003bdd 00000000 00000000
<4>[  104.704177] $16   : 8400dc00 829e1d30 82d30000 8400dd80
<4>[  104.709529] $20   : 82d30000 00000000 829e1c60 82ad202a
<4>[  104.714880] $24   : 00000000 a4c8b494                  
<4>[  104.720233] $28   : 829e0000 829e1c08 829de300 82b3040c
<4>[  104.725584] Hi    : 0000001b
<4>[  104.728520] Lo    : 00000000
<4>[  104.731456] epc   : 82b2e24c ieee80211_sta_ps_transition+0x14c/0x708 [mac80211]
<4>[  104.739159] ra    : 82b3040c ieee80211_is_our_addr+0x304/0x2f80 [mac80211]
<4>[  104.746314] Status: 1100f403	KERNEL EXL IE 
<4>[  104.750598] Cause : 40800008 (ExcCode 02)
<4>[  104.754689] BadVA : 00000068
<4>[  104.757625] PrId  : 00019650 (MIPS 24KEc)
<4>[  104.761715] Modules linked in: rt2800soc rt2800mmio rt2800lib qcserial pppoe ppp_async option cdc_mbim wireguard usb_wwan sierra_net sierra rt2x00soc rt2x00mmio rt2x00lib rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic nft_fib_inet mt76x2e mt76x2_common mt76x02_lib mt76 mac80211 libchacha20poly1305 ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake r8152 poly1305_mips nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat
<4>[  104.762329]  nft_meta_bridge nft_masq nft_log nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c ipvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables crc_ccitt compat chacha_mips cdc_wdm br_netfilter asn1_decoder natflow natcap nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ntfs3 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit
<4>[  104.851969]  act_mirred act_gact xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfs msdos ip6_gre ip_gre gre ifb ip6_udp_tunnel udp_tunnel sit ip6_tunnel oid_registry tunnel6 tunnel4 ip_tunnel tun lockd sunrpc grace autofs4 dns_resolver nls_utf8 nls_iso8859_1 nls_cp437 sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 md4 hmac ecb des_generic libdes cmac uas usb_storage leds_gpio ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod scsi_common ehci_hcd gpio_button_hotplug vfat fat ext4 mbcache jbd2 exfat usbcore nls_base usb_common mii crc32c_generic
<4>[  105.028040] Process napi/phy0-3 (pid: 1063, threadinfo=69cc1eab, task=f1d4706c, tls=00000000)
<4>[  105.036754] Stack : 8400dd80 82d30000 00000000 82b3021c 829e1d30 829e1c54 00004288 8400dd80
<4>[  105.045323]         82d30000 82b3040c 63b2e58b 803ed050 82ba0978 82ad302a 00000002 829e1c60
<4>[  105.053889]         00000000 82ad3020 0000015d 8400d6c0 8400d6c0 00000001 54dc1178 b1cc4884
<4>[  105.062453]         d01e7e82 84008400 829e1d30 8400dd80 82ad202a 00000001 82d304a0 0000009e
<4>[  105.071018]         82cb5000 82b3410c 808e18a0 8048bbb8 8400dd80 8400dd80 806ebd28 8400dd80
<4>[  105.079584]         ...
<4>[  105.082092] Call Trace:
<4>[  105.084584] [<82b2e24c>] ieee80211_sta_ps_transition+0x14c/0x708 [mac80211]
<4>[  105.091963] 
<4>[  105.093481] Code: 8e220018  000318c0  00431021 <8c430068> 8c47006c  24660001  00c3182b  00671821  ac460068 
<4>[  105.103481] 
<4>[  105.105069] ---[ end trace 1ee952185e361121 ]---
```

base on the latest code in openwrt
no issue 4 days ago.

@nbd168

ptpt52, 2022-12-12:
trying to connect phone to AP, then it crash and reboot

Gingernut1978, 2022-12-12:
It's caused by the latest mac80211 update

stintel, 2022-12-12:
Anyone who experiences the crash building there own image? If so please run faddr2line from e.g. build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.158/scripts/faddr2line to learn the exact line in the code where it crashed.

Also there seem to be 2 different crashes reported in this bugreport.

ptpt52, 2022-12-13:
crash looks random
another log:
```
<6>[   47.820364] br-lan: port 2(phy1-ap0) entered blocking state
<6>[   47.826100] br-lan: port 2(phy1-ap0) entered disabled state
<6>[   47.832398] device phy1-ap0 entered promiscuous mode
<4>[   49.825068] {natflow}:natflow_netdev_event(): catch NETDEV_UP event for dev=phy0-ap0(tso=0,0,hw_csum=0,0), add ingress hook
<6>[   49.841178] br-lan: port 3(phy0-ap0) entered blocking state
<6>[   49.846922] br-lan: port 3(phy0-ap0) entered disabled state
<6>[   49.853039] device phy0-ap0 entered promiscuous mode
<6>[   49.858574] br-lan: port 3(phy0-ap0) entered blocking state
<6>[   49.864254] br-lan: port 3(phy0-ap0) entered forwarding state
<6>[   49.899778] br-lan: port 3(phy0-ap0) entered disabled state
<6>[   60.352022] IPv6: ADDRCONF(NETDEV_CHANGE): phy1-ap0: link becomes ready
<6>[   60.359305] br-lan: port 2(phy1-ap0) entered blocking state
<6>[   60.365040] br-lan: port 2(phy1-ap0) entered forwarding state
<6>[   63.386128] IPv6: ADDRCONF(NETDEV_CHANGE): phy0-ap0: link becomes ready
<6>[   63.393331] br-lan: port 3(phy0-ap0) entered blocking state
<6>[   63.399067] br-lan: port 3(phy0-ap0) entered forwarding state
<1>[ 1340.740794] CPU 2 Unable to handle kernel paging request at virtual address 0000005c, epc == 838307a4, ra == 83830f9c
<4>[ 1340.751455] Oops[#1]:
<4>[ 1340.753741] CPU: 2 PID: 1478 Comm: napi/phy1-9 Not tainted 5.15.82 #0
<4>[ 1340.760164] $ 0   : 00000000 00000001 8518a000 00000000
<4>[ 1340.765405] $ 4   : 839684c0 83be8c4e 00000000 000004d3
<4>[ 1340.770626] $ 8   : 00000008 00000000 00000000 3ddfe8fe
<4>[ 1340.775851] $12   : 83be8c4e 00000400 00000004 00000000
<4>[ 1340.781072] $16   : 82e2bd20 83bde000 83968000 83bde6c0
<4>[ 1340.786300] $20   : 83968000 82e2bc3e 00000000 8391b02a
<4>[ 1340.791575] $24   : 00000000 804b1614                  
<4>[ 1340.796809] $28   : 82e2a000 82e2bbf0 83acb380 83830f9c
<4>[ 1340.802038] Hi    : 00000089
<4>[ 1340.804910] Lo    : 00000000
<4>[ 1340.807773] epc   : 838307a4 ieee80211_sta_ps_transition+0x1588/0x19dc [mac80211]
<4>[ 1340.815431] ra    : 83830f9c ieee80211_is_our_addr+0x304/0x2fc4 [mac80211]
<4>[ 1340.822346] Status: 1100fc03	KERNEL EXL IE 
<4>[ 1340.826538] Cause : 40800008 (ExcCode 02)
<4>[ 1340.830530] BadVA : 0000005c
<4>[ 1340.833392] PrId  : 0001992f (MIPS 1004Kc)
<4>[ 1340.837467] Modules linked in: ksmbd qcserial pppoe ppp_async option cdc_mbim wireguard usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic pl2303 nft_fib_inet mt76x2e mt76x2_common mt76x02_lib mt7603e mt76 mac80211 libchacha20poly1305 ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_scalable tcp_bbr slhc sch_cake r8152 poly1305_mips nft_tproxy nft_socket nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_queue nft_objref nft_numgen nft_nat nft_meta_bridge nft_masq nft_log
<4>[ 1340.837974]  nft_limit nft_hash nft_fwd_netdev nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_dup_netdev nft_ct nft_counter nft_compat nft_chain_nat nfnetlink_queue nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_syslog nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_bridge ts_kmp nf_conntrack_amanda nf_conncount macvlan libcurve25519_generic libcrc32c ipvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ipheth ip_tables crc_ccitt compat chacha_mips cdc_wdm br_netfilter asn1_decoder natflow natcap nf_nat nf_conntrack ntfs3 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ledtrig_usbport xt_set
<4>[ 1340.924678]  ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink jool_siit jool nf_defrag_ipv6 nf_defrag_ipv4 jool_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfs msdos bonding ip6_gre ip_gre gre ifb nat46 ip6_udp_tunnel udp_tunnel sit ip6_tunnel oid_registry tunnel6 tunnel4 ip_tunnel udp_diag tcp_diag raw_diag inet_diag tun lockd sunrpc grace autofs4 dns_resolver nls_utf8 nls_iso8859_1 nls_cp437 sha512_generic sha256_generic libsha256 sha1_generic seqiv jitterentropy_rng drbg md5 md4 hmac ecb des_generic libdes cmac uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_mtk_hcd xhci_hcd ohci_platform ohci_hcd softdog fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod
<4>[ 1341.011519]  scsi_common ehci_hcd gpio_button_hotplug vfat fat ext4 mbcache jbd2 exfat usbcore nls_base usb_common mii crc32c_generic
[... cut ...]
<4>[ 1341.143292]         00000001 000004ac 82e2bd20 83bde6c0 8391b02a 00000001 839684c0 000000e4
<4>[ 1341.151653]         8518a000 83834cec 00000000 00000000 82e2bd80 814b4100 00019682 00004a18
<4>[ 1341.160013]         ...
<4>[ 1341.162478] Call Trace:
<4>[ 1341.164907] [<838307a4>] ieee80211_sta_ps_transition+0x1588/0x19dc [mac80211]
<4>[ 1341.172374] 
<4>[ 1341.173865] Code: 10400016  00000000  8e030018 <8c62005c> 24420001  ac62005c  0000000f  8e030024  8e020018 
<4>[ 1341.183619] 
<4>[ 1341.185519] ---[ end trace 7d3ff42580e11ded ]---
```

@stintel  I cannot address which line

stintel, 2022-12-13:
> @stintel I cannot address which line

Which object file did you try? Afaict it should be `build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/backports-6.1-rc8/net/mac80211/rx.o` but as I still didn't manage to capture a call trace locally I don't have a correct offset to try.

ptpt52, 2022-12-13:
looks like `rx->link_sta` is NULL so it crash when access to `rx->link_sta->rx_stats.syncp`

so it crash at line: `u64_stats_update_begin(&rx->link_sta->rx_stats.syncp);`

Spudz76, 2022-12-13:
Seems like maybe the `if` is wrong?

`if (rx->sta) {` ... accesses `rx->link_sta`

I'd change to `if (rx->link_sta) {` because there is really no reason to check existence of `rx->sta` if the block never touches it.

Either that or every other reference in the block should be `rx->sta` but I'd wager a single line typo rather than multiple lines.

nbd168, 2022-12-13:
From what I can tell, the `if (rx->sta) { ... rx->link_sta` pattern seems to be intentional. I think there are probably some cases where link_sta should get set but doesn't.

ptpt52, 2022-12-13:
It looks like the issue is only happening on the MT7612 chip. I see two cases, one on the Mir3g and one on the Mir3, which both have the MT7612 chip in common.

Spudz76, 2022-12-13:
Well there isn't even a `link_sta` member in the struct anymore so maybe that's how it's `null`

https://elixir.bootlin.com/linux/v5.15.82/source/net/mac80211/ieee80211_i.h#L202

ptpt52, 2022-12-13:
@Spudz76 you may look at the wrong code, it should be in the `backports-6.1-rc8`

Spudz76, 2022-12-13:
Yeah I see it's a new addition now, in the 6.1 tree.  So the new mac80211 is just the 6.1 stuff backported.

ptpt52, 2022-12-13:
It looks like the code at https://elixir.bootlin.com/linux/v6.1/source/net/mac80211/rx.c#L5062 has nothing to do with the link_sta variable, which is confusing to me.

Spudz76, 2022-12-13:
So then [this change](https://patchwork.kernel.org/project/linux-wireless/patch/20220902161143.028e767b20a4.I4133fd56517e260f147fce5a65f5a0b8bfa70d69@changeid/) seems really close.  I note that earlier in the file there is a check for both `sta` and `link_sta` (bail with continue if both aren't available) but then near the crash it only checks `sta`.  So then my next guess is make it `if (rx->sta && rx->link_sta) {` because apparently that earlier test either succeded somehow or does not protect the `ieee80211_deliver_skb` from garbage input.

nbd168, 2022-12-13:
I think checking for rx->link_sta in addition to checking for rx->sta is wrong. It may prevent the crash, but it's just papering over another bug.
If rx->sta is set and there is no active secondary link, rx->link_sta should have been initialized to &rx->sta->deflink

ptpt52, 2022-12-13:
```C
4784 static bool ieee80211_prepare_and_rx_handle(struct ieee80211_rx_data *rx,
4785                                             struct sk_buff *skb, bool consume)
4786 {
4787         struct ieee80211_local *local = rx->local;
4788         struct ieee80211_sub_if_data *sdata = rx->sdata;
4789         struct ieee80211_hdr *hdr = (void *)skb->data;
4790         struct link_sta_info *link_sta = NULL;
4791         struct ieee80211_link_data *link;
4792 
4793         rx->skb = skb;
4794 
4795         /* See if we can do fast-rx; if we have to copy we already lost,
4796          * so punt in that case. We should never have to deliver a data
4797          * frame to multiple interfaces anyway.
4798          *
4799          * We skip the ieee80211_accept_frame() call and do the necessary
4800          * checking inside ieee80211_invoke_fast_rx().
4801          */
4802         if (consume && rx->sta) {
4803                 struct ieee80211_fast_rx *fast_rx;
4804 
4805                 fast_rx = rcu_dereference(rx->sta->fast_rx);
4806                 if (fast_rx && ieee80211_invoke_fast_rx(rx, fast_rx))
4807                         return true;
4808         }
```

here call ieee80211_invoke_fast_rx() but never set `rx->link_sta`  ?

ptpt52, 2022-12-13:
call stack may be like:
```
ieee80211_prepare_and_rx_handle()
  ->ieee80211_invoke_fast_rx()
    ->__ieee80211_rx_h_amsdu()
      ->ieee80211_deliver_skb()
```

nbd168, 2022-12-13:
Here's a patch for testing: https://termbin.com/g47s

coudu, 2022-12-13:
> Here's a patch for testing: https://termbin.com/g47s

kmod-mac80211 rebuilded with the patch and no crash for now.

nbd168, 2022-12-14:
Fix is in OpenWrt master

closed 2022-12-14 by nbd168