openwrt/openwrt #11479: LED control broken in 7622 RT3200

andyleadbetter, 2022-12-12:
The gpio led control is only partially working on the Belkin RT3200.  I've setup a cron job to turn off the LEDs at night, and turn on in the morning by writing into sysfs GPIO entries.  The crontab is as follows
00 22 * * * echo none > /sys/devices/platform/gpio-leds/leds/inet:blue/trigger
00 22 * * * echo 0 > /sys/devices/platform/gpio-leds/leds/inet:blue/brightness
00 22 * * * echo none > /sys/devices/platform/gpio-leds/leds/power:blue/trigger
00 22 * * * echo 0 > /sys/devices/platform/gpio-leds/leds/power:blue/brightness
00 7 * * * echo netdev > /sys/devices/platform/gpio-leds/leds/inet:blue/trigger
00 7 * * * echo 1 > /sys/devices/platform/gpio-leds/leds/inet:blue/brightness
00 7 * * * echo 1 > /sys/devices/platform/gpio-leds/leds/inet:blue/rx
00 7 * * * echo 1 > /sys/devices/platform/gpio-leds/leds/inet:blue/tx
00 7 * * * echo none > /sys/devices/platform/gpio-leds/leds/power:blue/trigger
00 7 * * * echo 1 > /sys/devices/platform/gpio-leds/leds/power:blue/brightness

The LED's turn off correctly at 10pm.  But at 7am, they turn on but the triggers do not work, i.e. not responding to RX/TX events.

andyleadbetter, 2022-12-12:
Tried that, it seems that state is reset in the driver when certain writes occur.

Sending 1 to brightness for example turns on solid white
Sending netdev to trigger enables the RX/TX nodes
writing 1 to RX and 1 to TX has no effect

There does not seem to be a sequence of writes with waits, i.e. I setup a cron that run 7 00 7 01 7 02 etc, still no success.  Even when the config seems correct with 

trigger == netdev
brightness == 1
rx = 1
tx = 1

the LED's are lit solid.

Spudz76, 2022-12-12:
okay, good background investigation, seems actually broken then if the sysctls show the right content afterward

andyleadbetter, 2023-03-23:
Using the sysfs to disable LED's, and then running the LED startup script to re-enable worked.